nkolban/esp32-snippets #102: Order of #include <WiFi.h> and #include "BLEDevice.h"

pit001, 2017-09-29:
I tried to use Wifi and BT together. But I get a lot of compile and runtime problems. 

One was the order of  #include <WiFi.h> and #include "BLEDevice.h"

#include <WiFi.h>
#include "BLEDevice.h"
is working

#include "BLEDevice.h"
#include <WiFi.h>
gives a 'WiFiServer' does not name a type)

It drives me crazy. 

Working sketch:
#include <WiFi.h>
#include "BLEDevice.h"
WiFiServer server(80);
void setup() {
  // put your setup code here, to run once:

}
void loop() {
  // put your main code here, to run repeatedly:
}

NOT working sketch:
#include "BLEDevice.h"
#include <WiFi.h>
WiFiServer server(80);
void setup() {
  // put your setup code here, to run once:

}
void loop() {
  // put your main code here, to run repeatedly:
}

nkolban, 2017-09-29:
Weve been having issues getting WiFi and BLE working in base ESP-IDF ... however these have been runtime issues as opposed to compilation issues.  That said, the order of inclusion should not cause a compilation issues and will be investigated for resolution.

nkolban, 2017-09-30:
I'm having trouble recreating this issue:
I get 100% clean compile ... see attached screen shot:

[image]

pit001, 2017-09-30:
hmm, strange. I have to check this closer !

pit001, 2017-10-04:
If I delete the Wifi.h & .c in the folder \libraries\ESP32_BLE\src the order is equal ...

nkolban, 2017-10-04:
Sorry ... I don't understand, can you explain what you mean by "The order is equal"?  Have you found an enhancement to the description of the story that will help me understand it better.

pit001, 2017-10-04:
sorry: 
If I delete the Wifi.h & .c in the folder \libraries\ESP32_BLE\src I can place the two includes in a random order. 
I have no idea, but I also didn't understand the existence of the Wifi.x in your code. Mybee with your background about the C++ API you have an idea !?

nkolban, 2017-10-07:
Howdy my friend.  Still not following the idea in this puzzle.  I think you are saying that if you include WiFi.h and BLEDevice.h in one order in Arduino coding, we work but if we place them in our scripts in the other order we fail to compile.  I am unable to recreate this issue.

Perhaps we can wait a while to see what happens when Arduino support is restored after Espressif patches the inability to compile C++ with the latest toolchain and then the Arduino-ESP32 libraries pick up what ever is needed.

chegewara, 2017-10-07:
Maybe in arduino ide class WiFi is somehow shadowing standard arduino wifi class?

nkolban, 2017-10-08:
@chegewara My puzzle is that I ran up an Arduino environment and tried to recreate the issue ... but was unsuccessful.  The code compiled cleanly with the scenario that the original issue raiser found not to work.  I think I'm not properly understanding.  I don't think Mr @pit001 has English as his primary language and I'm being stymied with comprehension.  My gut reaction at the moment is to put all Arduino BLE C++ puzzles on hold until we can get stability and consistency in the environment.  There has been dramatic churn in the BLE APIs exposed in ESP-IDF and the tool chain required to build them over the last few weeks.  Personally, I would have been happy to say "The BLE C++ classes will only support released versions of ESP-IDF" which would currently be v2.1.  However the owners of the Arduino-ESP32 environment have changed their distributed environment to use the latest from ESP-IDF which is pre3.0 ... which has meant that I had to also rework BLE C++ to pre3.0 earlier than I would have hoped ... and in so doing, we find that the C++ environment doesn't work with the latest toolchain.

nkolban, 2017-10-29:
Closing for just now.  Re-open as needed.

closed 2017-10-29 by nkolban