nkolban/esp32-snippets #98: Feature Request: getCharacteristics() with return a map

pit001, 2017-09-29:
Enhance the getCharacteristics() to also return a pointer to the m_characteristic map like getServices()

nkolban, 2017-09-30:
The class "BLEClient" exposes a call called getServices() that returns a map keyed by UUID returning a BLERemoteService.  The goal will be to add a getCharacteristics() to BLERemoteService that returns a similar map containing the BLERemoteCharacteristic. entries.

The BLERemoteService::getCharacteristics() method has now been added.

Please review and close this issue when delighted.  Any issues, post back.

pit001, 2017-10-04:
HI. Thanks for implementing. 

What happens if I call getCharacteristics() ? -> I get a null pointer. 
So I first hav to call retrieveCharacteristics to fill the map ?! Than every time it's read and filled again !

So if have to do something like 

```
if (!m_haveCharacteristics) {
		retrieveCharacteristics();
}
```

In my solution I simple enhance the existing:
void BLERemoteService::getCharacteristics() {...}

to

std::map<std::string, BLERemoteCharacteristic*>* BLERemoteService::getCharacteristics() {
...
	return &m_characteristicMap;
}

nkolban, 2017-10-04:
Thank you sir.  That is indeed an excellent improvement and thanks for the detailed design and suggestion.  I have coded exactly as you suggested.  The code has now been changed and committed.  Hopefully we are now closer to the desired function.

pit001, 2017-10-04:
OK, works well !
thanks a lot
you should update your logstrings.. 
```
ESP_LOGD(LOG_TAG, ">> getCharacteristics() for service: %s", getUUID().toString().c_str());
ESP_LOGD(LOG_TAG, "<< getCharacteristics() for service: %s", getUUID().toString().c_str());
} // getCharacteristics
```
Peter

nkolban, 2017-10-04:
Thank you again.  Have updated the log strings which will be visible in the next commit/push.

When delighted with the completion of this issue, go ahead and close it or post back with additional requirements to be addressed.

closed 2017-10-04 by pit001