nkolban/esp32-snippets #99: Feature Request: Descriptors, special getDescriptors()

pit001, 2017-09-29:
Funtions to get the BEL GATT descriptors.

nkolban, 2017-09-29:
Can you elaborate on this one?  I had assumed that we already had the ability to get BLE GATT descriptors from a BLE Client?

pit001, 2017-10-06:
yes Sir ! My fail - I didn't recognize your library update. :-)
So, the first step is done. 

To be able to loop through the descriptors also here we need a getdescriptor**s**() with return value in the public of the class. 

In BLERemoteCharacteristic.cpp I simply modify the existing:
```
279: void BLERemoteCharacteristic::getDescriptors() {
...
327: }
```
to 
```
279: std::map<std::string, BLERemoteDescriptor *>* BLERemoteCharacteristic::getDescriptors() {
327:	return &m_descriptorMap;
328:} // getDescriptors
```

and movt it from 
`64: void              getDescriptors();`
to 
`52: std::map<std::string, BLERemoteDescriptor *>* getDescriptors(void);`

Peter

nkolban, 2017-10-06:
Thank you sir.  I like an issue where you have done all the work and all I need to do is copy/paste :-)

I have made the changes (I think/I hope).  If you could review and validate, that would be great.

Neil

pit001, 2017-10-06:
I do my best ! :)
You should move it to the public: section.
(seems that our line numbers does not match exactly)

and you have updated only 1 of 3 of the Strings ;)
	ESP_LOGD(LOG_TAG, "<< getDescriptors(): Found %d descriptors.", offset);
} // getDescriptors

Peter

pit001, 2017-10-06:
and: 
std::map<std::string, BLERemoteDescriptor *>* BLERemoteCharacteristic::getDescriptors() {
	return &m_descriptorMap;
} // getDescriptors

nkolban, 2017-10-06:
Thank you sir.  Lets see how this one feels ... new commit made.

nkolban, 2017-10-23:
Assuming resolved.  Don't hesitate to re-open as needed.

closed 2017-10-23 by nkolban