nkolban/esp32-snippets #104: move removeCharacteristics(); to public

pit001, 2017-09-30:
the map of Services and Characterisics keep (once filled) i.e. for the sensor TAG more than 10k of data. 
To allow the user to cleanup this if the data is not anymore move removeCharacteristics(); to public.

and add 
m_haveCharacteristics = false;
in BLERemoteService::removeCharacteristics();

nkolban, 2017-09-30:
I think we need to contemplate not just this memory utilization but the whole notion of BLEClient re-use.  Some initial considerations:

* Re-using a BLE Client to connect to a different BLE Server?
* Disconnecting a BLE Client and then re-connecting to the same BLE Server.
* Ability to "free up" resources from a BLE Client when we will never use it again.
* Choosing to keep resources associated with a BLE Client when it has only transiently disconnected from the BLE Server.

pit001, 2017-09-30:
It's a mutitasking system. It's a better implementation to keep big memory blocks only as long as they needed - not the full time, and only released one line before they reused in the next call.  The creates a continuous need of memory, blocking other tasks - i.e. the Webserver.

pit001, 2017-09-30:
// we start with Mem:125 kbyte of free mem
BLEDevice::init(""); // -35kbyte

BLEClient* pMyClient = BLEDevice::createClient(); // - 0,5k
// Mem: 89140 (before connect)
pMyClient->connect(address); // -4,5k
BLERemoteService* pMyRemoteService = pClient->getService(serviceUUID); // -11k
// what happens here?
//pseudo code
loop over services (for auto over m_servicesMap)
{
  read service data
  loop over characteristics (for auto over m_characteristicsMAP)
  {
     //read description
     //read handle
     //read properties
     //read valume
  }//end characteristic
 //deal with the results
 //i.e send it to a Server, deliver it to a client - often preparared as mqtt, json, JS, html, ...
 //each small part of  RAM is needed here, but we keep the never used Ch data, its 2K (depents on Ch size)
}//end service
disconnect // +9.7kbyte hippeyy
// Mem: 64688 (after disconnect) + 13239 in json = 77927 (89140 at start -> 11k unknown Ram )
in my example the json, containing the resulting tree strucure of the device, is additional 13,2 kbyte
connect to wifi -39kbyte
start webserver - 0.6kbyte

It's a multitasking system, the memory values are only a hint what happens. 
At the end the rest of memory is 25.2 Kbyte (started with 125k, walk throug BLE, start Websrver) (but that is NOT the theme of this issue)

The BLE object can't know if the data are needed anymore, but the app engineer knows it. So I'm happy to be able to clear the Characteristics mem with removeCharacteristics().

Peter

nkolban, 2017-09-30:
What if we provide an optional  parameter to disconnect().  For example:

BLEClient::disconnect(bool clean=false)

By default, this will do nothing new ... however if we call disconnect(true) then all the resources consumed/used during the connect() -> disconnect() period will be released.  It will then be invalid to use any pointers to BLEService, BLECharacteristic or BLEDescriptor that might have been saved by the programmer?

nkolban, 2017-10-14:
Closing for just now.  Re-open as needed.

closed 2017-10-14 by nkolban