nkolban/esp32-snippets #105: Semaphore is never given back when creating a new BLEServer

FulminatingGhost, 2017-09-30:
Removing the BLEDevice::m_pbleServer from the BLEServer constructor never lets the BLEDevice::gattServerEventHandler trigger the handleGATTServerEvent in the BLEServer and thus the new BLEServer never gives back the semaphore it took in registerApp().

This causes FreeRTOS to wait for the semaphore indefinitely.

nkolban, 2017-09-30:
Thank you sir.  Fixes are in and committed.  What this is showing me is that once again my testing / regression strategy is non-existent.  If these libraries catch on, we'll need to give some thought to a far more robust regression mechanism.  Let me create an issue to track that.  However, I think this bug is now fixed.  Please test and, when delighted, close this issue or comment back if it is still not working.

FulminatingGhost, 2017-09-30:
Thank you Mr Kolban! Indeed your fix resolved the issue.
I'm still in the learning phase regrding the esp32. These libraries really made things so much smoother to work with so I sure think that they'll catch on ;-)

closed 2017-09-30 by FulminatingGhost