nkolban/esp32-snippets #107: File Upload to Webserver

squonk11, 2017-10-03:
Dear Neil,
currently I am working on a project using the mongoose webserver. Unfortunately it turns out that it is quite cumbersome to write the html and javascript code due to the fact that I need to flip the SD card from the esp32 to my laptop and vice versa very frequently. So, I was asking myself if there is an easier way to do that. I tried to use the TFTP.cpp code but unfortunately this is only a simple ftp server which is not supported by e.g. filezilla.
Do you know a better solution for that?
Best regards

nkolban, 2017-10-03:
Very interesting that you ask about this ... Over the last few weeks a new component (prototype) has been added to the C++ utils (this repository).  Like many folks before me and like many folks after me, I decided to have a bash at writing my own HttpServer set of libraries.  This time, they were written to leverage C++ as opposed to C.  I am personally pleased with the results.   Having implemented an HttpServer, I then tackled WebSockets built on that .... and again I was pleased with the results.

From there, I wrote a class that uses WebSockets to receive incoming files over a web socket and write them to the file system.  The class is called "WebSocketFileTransfer" and is available in the current code base.  However, I have not yet documented it well nor have I had the opportunity to pilot it with interested consumers.  It might be that this is of use to you.  If interested, what I suggest we do is have a live chat via Skype (neil.kolban) or IRC (channel `#ESP32`) and see if we can't get a test running for you and see if it meets your needs and, if not, lets see what it is missing that would delight you.

squonk11, 2017-10-03:
Curretnly I am using mongoose.c with REST calls for reading data from the ESP32. It has the advantage that (maybe later) I could use MQTT also. If I should use your webserver library I need to rewrite that part of the code. Since this is a hobby project it would need some time to do it. So, thank you for offering your help via Skype but I think it will take some time until I will be ready for that.
Currently I am checking the solution of having the html and javascript files on a local drive on my PC and I modify the url of all REST calls to point to the ESP32. Unfortunately here I have the problem that my browser filters the REST calls and complains "Cross-Origin Request Blocked. (Reason: CORS header 'Access-Control-Allow-Origin' missing)". Now I am searching a solution for that. I don't want to modify the security settings of my browser. Maybe there is another solution available. I will check.

chegewara, 2017-10-03:
Add this header to response
`("access-control-allow-origin", "*")`

squonk11, 2017-10-03:
Where should I add it? Here:
`mg_printf(nc, "%s", "HTTP/1.1 200 OK\r\nTransfer-Encoding: chunked\r\n\r\n");` ?
Before or ater this?

nkolban, 2017-10-03:
As I remember, Mongoose is a low-level implementation of an HTTP server.  Are you programming to the lowest level Mongoose APIs or using a higher level wrapper such as 

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WebServer.h

If the latter, then I believe there is a class instance called HTTPResponse that encapsulates the response sent from your ESP32 application which has a method called "addHeader()" which could be used to add response headers to the response that will eventually be sent back to the browser.

squonk11, 2017-10-03:
yes, I am using the mongoose API directly. So, rewriting my webserver would be a bit more effort. First I would like to try if I am able at all to achieve my targets with this first step. Then I will decide if I proceed with this project and if I proceed I might change to your webserver code.

chegewara, 2017-10-03:
https://forum.mongoose-os.com/discussion/138/add-header-to-response

nkolban, 2017-10-03:
@squonk11 You might want to post the question to the Mongoose forum or to the Mongoose github.  This github issue forum is primarily for issues, enhancements and questions to the code found here.  If you have generic ESP32 questions, then the ESP32 forum here is a good place:

https://esp32.com/index.php

Or if you have queries about 3rd party software, maybe try the forums and communities related to those 3rd party packages?

squonk11, 2017-10-03:
Currently I am using successfully several of your cpp classes (WiFi, GPIO, Socket, System, Task, FreeRTOS, ...) and I appreciate the work your are doing.  I think I will also switch to one of your webserver classes (WebServer or HttpServer) if I continue my project. For the moment I am just checking the feasibility of the ESP32 plattform for my purposes.

nkolban, 2017-10-03:
No problems at all, and I'll be delighted to help in any way I can.  My previous post wasn't to try and *run you off*, on the contrary, I'm trying to get you the speediest and best response possible.  For code that I work upon, I hope to be "johnny on the spot" with fixes and responses.  But for code outside of my control, then forums and 3rd party web sites are the better alternatives.

If you decide to use one of the WebServer classes in this repository, my bias right now is the HttpServer.  However, I recommend that because it interests me more than WebServer class not because it is any more efficient or *better* in any obvious way ... only that I'm a big fan of C++ and a native C++ implementation appeals to my sense of purity more than a wrapped hybrid.

nkolban, 2017-10-23:
Closing for now.  Don't hesitate to re-open as needed.

closed 2017-10-23 by nkolban

burkulesomesh43, 2018-08-11:
hi,
I am using esp idf environment
I want to decode the string in url. mbedtls_base64_decode() function does not gives an error after compiling.   but when i see  decoded result in decode_pass variable, the processor gets restart again and again. 
char *getpass="cabin336%40efc";
unsigned char *decode_pass=NULL;
 size_t len;
 mbedtls_base64_decode(decode_pass, sizeof(decode_pass), &len, (unsigned char *) get_pass, strlen(get_pass));

chegewara, 2018-08-11:
Did you try this way:
`unsigned char *decode_pass[1000]={0};`
if this works then you need to initialize decode_pass with malloc.

burkulesomesh43, 2018-08-11:
I tried this. processor is not restarting now, but decode_pass variable does not contain decoded value after calling of this function
mbedtls_base64_decode(decode_pass, sizeof(decode_pass), &len, (unsigned char *) get_pass, strlen(get_pass));

chegewara, 2018-08-11:
Try to ask on forum or esp-idf repo.

burkulesomesh43, 2018-08-11:
ok. thank you.