nkolban/esp32-snippets #108: Can not compile test_rest.cpp

wiltOS, 2017-10-03:
Hello, I have tried to compile your example for using rest and Curl to post data, but I am receiving this error:

` error: 'RESTClient' was not declared in this scope
  RESTClient client;

error: 'RESTTimings' was not declared in this scope
  RESTTimings *timings = client.getTimings();

 error: 'timings' was not declared in this scope
  RESTTimings *timings = client.getTimings();

error: 'client' was not declared in this scope
  RESTTimings *timings = client.getTimings();
`

The file is in "cpp_utils/tests"

All files were installed to the components. Thank you for the help!

nkolban, 2017-10-04:
Howdy,
For some of the classes/packages in cpp_utils they leverage 3rd party packages/libraries that aren't for me to distribute.  Today, there are three such packages:

* curl
* mongoose
* u8g2

Because I don't want to force users who have no need for these packages to download and configure them, we elected to make them optional.  What that means is that you need to tell the build environment that they are present if you wish to make use of them.  To do this, run "make menuconfig", go to "Component Config" and then "C++ settings".  See the following screen shot.

[image]

Set the "LibCurl present" flag (and obviously make libcurl present) and then I think you will be good.

Have a review of this response and if all is working for you and you are delighted, close out the issue.  If there are still issues, respond back and we'll pick it up again.

wiltOS, 2017-10-04:
Hello, I have installed libcurl from here: https://curl.haxx.se/download.html
as well as your version of curl-esp32 here: https://github.com/nkolban/curl-esp32
by placing them into the components folder of the project and with the esp32-snippet fix you have suggested to use if you are having build problems.

 I have enabled libcurl in my options, performed "make app-clean" and I still have no luck.

I appreciate the response

nkolban, 2017-10-04:
In your project folder, you will find a directory called "build".  Remove that directory as it will be re-created when you re-compile with make.  I suspect that simply changing the "make menuconfig" option after a previous build may not have been enough for the new flag to be recognized.  Let's give that a try and see what happen next.

wiltOS, 2017-10-05:
I have removed the build folder and still the same errors. Currently in the components I only have "cpp_utils" and "curl" ( taken directly from github repository ). I am compiling only the test_rest.cpp file without any combination.

I have confirmed the RESTClient file was in the directory. Compiling the testCurl.c file in the fragments folder appears to work without error. The esp-idf is on the most recent version, and I am using the MSYS2 MINGW32 Shell.

I followed the directions as on your youtube video "ESP32 Technical Tutorials: Integrating with Lib CURL" but I do not use eclipse.

nkolban, 2017-10-05:
So far, I have performed all my development testing in a Linux environment ... it may be that is part of the issue.  In principle, what one need do is create a new ESP-IDF template project.  In that project, add a components directory and in the components directory have cpp_utils and curl.  In the project directory, one can then run "make menuconfig" and tell the environment that LibCurl is available.  Then when we run "make", your application code, the component for LibCurl and component for cpp_utils will compile and all should link correctly.

My next suggestion is to again remove the build directory and re-build all ... this time capture all the text that results from the build process and paste that into a "paste bin" and post the link to the results.  In a second pastebin also post the content of sdkconfig text file.  Once done, post those links here and we can take a look and see what we can spot.

wiltOS, 2017-10-05:
Following those steps, here is my paste: https://pastebin.com/4YV61eV8

I have tried removing the wrapping "#if defined(ESP_HAVE_CURL)" from RESTClient.h and the compilation did not send the error. It is possible that I did not install libcurl properly for mingw32, but I can not find any DLL on their website for it.

nkolban, 2017-10-05:
Thank you ... that was exactly what I needed ... we have found the problem.  Despite having coded the "make menuconfig" to enable/disable  LibCurl,  we didn't actually "honor" the setting.   We have made some changes that will now hopefully resolve the problem.  I also want to apologize for the hassle.  This was a stupid error and should have been caught.

Can I ask you to re-download the "cpp_utils", remove the project "build" directory and let us see if this resolved the issue.  Please report back once tested.

wiltOS, 2017-10-05:
This is definitely a step forward. I do not get the REST error anymore but am getting this new error:

> Z:/projects/rest_curl/build/curl\libcurl.a(mime.o):(.literal.curl_mime_filedata+0x10): undefined reference to `access'
> Z:/projects/rest_curl/build/curl\libcurl.a(mime.o): In function `curl_mime_filedata':
> Z:/projects/rest_curl/components/curl/lib/mime.c:611: undefined reference to `access'

Here is the complete log: https://pastebin.com/ztHVczny

nkolban, 2017-10-05:
Rather than just theorize that all works, I will build the test application myself and 100% validate that it at least builds on my system.  The "libcurl" library is a moving target ... and every now and again the good folks over at LibCurl insert some code that is perfectly fine on Linux (the target platform for LibCurl) which doesn't have an equivalent on ESP32.  I believe (but have not yet checked) that LibCurl has slipped in a call to the Linux system call named "access" (see: https://linux.die.net/man/2/access).  This system call deals with Linux process ids ... and their authority to read and write files.  In the ESP32 we don't have "Linux Processes" and hence this API (access) doesn't have an identical equivalent.  What I will do is attempt to build on my system which should result in the same error ... from there we see whether we can configure LibCurl to *not* use `access` and if so, we are good.  If we can't then we will need to either petition LibCurl to make `access` usage optional or petition ESP-IDF to add an `access` function that would satisfy a call to it.

wiltOS, 2017-10-05:
It could be very possible I did not install libcurl properly. I only copied the github repository into components and installed the esp32 make and config you have provided. The website did not specify a specific download to the libcurl library, so I assumed the libcurl library wqs included in the git.

Also, there was recently an update to the curl download on their website, it could be possible this has changed something

wiltOS, 2017-10-05:
I have checked the curl config and noticed you have set this property:

> /* cpu-machine-OS */
> #define OS "x86_64-pc-linux-gnu"

Perhaps this is the issue?

nkolban, 2017-10-07:
Problem recreated ... studying puzzle in detail now.

nkolban, 2017-10-07:
I now have a clean compile and a clean build.  Here is what has been done.

The latest LibCurl uses the Posix function called "access(2)" .. see:

https://linux.die.net/man/2/access

This function is not part of the ESP-IDF and, as such, at link time the compilation fails.  Thinking through the puzzle, I can't sensibly ask LibCurl to not use access().  It is a fundamental system call found in a Posix system and that is the target audience for LibCurl.  If we look inside LibCurl, it utilizes Posix all over the place.  The ESP-IDF provides an implementation of most of Posix which is why we get so far.  What I will do is make a request to ESP-IDF to add access().  However, in the interim, we have provided a "shim" for access which will satisfy our needs.  This shim is found in posix/posix_shims.c as part of the esp32-snippets repository.  Add that source file into your compilation and the error should resolve.

I compiled cleanly and, trying to learn from my previous mistakes, ran a curl test which worked as desired.

Have a go with the information presented in this post and let us know how you get on.  If there are questions, post them back.  If it ends up working for you and are delighted, close the issue when ready.

nkolban, 2017-10-07:
Enhancement request raised against ESP-IDF:

https://github.com/espressif/esp-idf/issues/1085

wiltOS, 2017-10-07:
That was it! I was able to compile and run without a single problem. Thank you for all the assistance!

closed 2017-10-07 by wiltOS

nkolban, 2017-10-07:
You are more than welcome my friend.  Thanks for using the issues technology and don't hesitate to create new ones as needed.

danielRM88, 2017-12-06:
Hi, I am facing the same problem and I need a bit of help @nkolban 
I am not an expert in C++ nor eclipse, the thing is I am not sure how to "Add that source file into your compilation". Could you please be more specific?
Any help would be really appreciated.
Thank you.

nkolban, 2017-12-06:
Howdy my friend.  How are you compiling your project today?  Let's start by understanding which of the many techniques for building (command line, Eclipse, Visual studio etc) you are using.  Once you respond to that, I can be more prescriptive.

At the highest level ... and the general ... you have some source code contained in source files that you have written.  These source files are making calls to library functions you did not write (eg. the code in the esp32-snippets repository).  I am thinking that when you compile your code, you are finding that a function called "access()" can not be found.  A workaround to this is to provide an implementation of that missing function yourself.  We have provided a sample which may be all that you need which is found here:

https://github.com/nkolban/esp32-snippets/tree/master/posix

Add the source files found there into your build, recompile and those source files should be compiled and linked with your app.  At that time, we should be able to resolve the missing "accept()" call.

danielRM88, 2017-12-06:
Hi, I am using eclipse in a mac OSX environment and I have created a C++ project. I have added everything needed for the project to compile including the curl library. I imported (added into main folder for now) the RESTClient.cpp and header file into my project with all its required dependencies.
So far I have encountered 2 problems:
1.- The option for "C++ settings" in "Component Config" in my menuconfig is not showing. The curl library is still included in my project and compiles fine. I managed to get around the ''**_RESTClient' was not declared in this scope_**' error by commenting out '**_#if defined(CONFIG_LIBCURL_PRESENT)_**' in the RESTClient.cpp file (Not sure if that may still be the cause of the problem).
2.- After commenting that line out I get to the point wiltOS was,  
[image]

As for the solution of including the posix_shims.c, what I tried was including it into the paths and symbols of the project, also tried adding the file into the main folder.

nkolban, 2017-12-06:
Let's review ... and again ... if my docs aren't good enough ... I need to fix that ... but only with help from folks like yourself.

The notion is that we have an ESP-IDF component that I call "cpp_utils".  This is a "component" in the ESP-IDF sense of the world and hence has to be installed as such in a user's components folder.  Once added, when you perform a project build, that should also build the source code in the cpp_utils component **and** add any extra entries to the "make menuconfig" environment.

Assuming you have a project called "XYZ" then in that project should be a folder called "components" and in "components" there should be a folder called "cpp_utils" which contains the source files from the "cpp_utils" directory of this repository.  Can you confirm?  If not, please describe what you have ... maybe providing an "ls -R" output?

danielRM88, 2017-12-06:
Oh I see, I misunderstood how it had to be implemented. Thank you for taking the time to explain it.
I just included the cpp_utils in my project like you said and the "C++ settings" appeared in menuconfig. 
And I was able to add the posix_shims to the '**_main_**' folder without any issues... Now my project is compiling just fine.
Thank you so much.

nkolban, 2017-12-06:
You are very welcome my friend.  If you can see ways of better describing this in the markdown documentation of any of our projects, please let us know.  Feel free to have a go at documenting it yourself should you be so inclined.