nkolban/esp32-snippets #110: Bootloop w Wifi + BLE

pit001, 2017-10-05:
Hi, not sure if this issue is allready contained in [#82](https://github.com/nkolban/esp32-snippets/issues/82). 
The simplified test code (below) results in a bootloop if line 
`BLEDevice::init(""); `
is active:

```
#include <WiFi.h>
#include "BLEDevice.h"

#include "c:/credentials.h"
//const char* ssid     = "yourssid";
//const char* password = "yourpasswd";

void setup()
{
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  BLEDevice::init(""); //<-affecting line
}

void loop(){}
```

Debug Output:

nkolban, 2017-10-05:
That's a great post/report.  Lots of great details and well described.  Unfortunately I'm still considering we are in Arduino limbo for just now.  The latest Arduino-ESP32 libraries use the `1.22.0-73-ge28a011` compiler which has a known breakage for C++ classes that use standard library.  See:

https://github.com/espressif/esp-idf/issues/1032

Until that is formally resolved by what ever change is necessary **and** the Arduino-ESP32 libraries are modified/re-released to pick up the change, there isn't a current environment available to us to sensibly perform Arduino work and testing.

SijmenSchoon, 2017-10-10:
Is there any known workaround for this issue? Sadly I depend on another bugfix in the Arduino repository fixing some HTTPS issues, which means I can't easily use an older version.

nkolban, 2017-10-10:
Progress is being made.  We have a "chain" of fixes needed ... the ESP-IDF is the foundation and the latest Git master seems to contain what we need.  The next level up is for the Arduino-ESP32 package to refresh itself with the latest ESP-IDF from Git master and a request has been opened for this just a day ago ... see:

https://github.com/espressif/arduino-esp32/issues/709

When that request/issue gets satisfied ... the final part will be testing the BLE libraries on Arduino and pushing a new Arduino BLE C++ library which should only take about a day.

nkolban, 2017-10-14:
The great news is that the Arduino ESP32 project has been updated.  Please re-build your base Arduino ESP32 environment.  The even better news is that the owner of that project has decided to include the BLE C++ classes as part of the distribution.  So it is now baked in.  Please re-test and let us now see where we stand.

SijmenSchoon, 2017-10-14:
Great! I'll test it later today

SijmenSchoon, 2017-10-14:
Sadly, the changes don't seem to have fixed the issue. I'm still getting the same problem as @pit001 had

nkolban, 2017-10-14:
I have managed to recreate the issue on my machine using the latest Arduino-ESP32 environment.  This is great ... because a problem re-created is a problem half solved.   Running the decoder against the stack trace, I find:

```
0x40089ae4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40089be3: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40083b43: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 350
0x40083ce0: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 207
```

Unfortunately, this appears to happen at the start of the program before any other activity is performed which had me stymied.   So I walked through the source code of the ESP-IDF found here:

```
https://github.com/espressif/esp-idf/blob/master/components/esp32/cpu_start.c
```

Working backwares ... Line 207 calls `start_cpu0()`.  Which takes us to the code start_cpu0_default (its an alias) starting at line 254.  We find the exception occurring at line 350 which has an extremely useful comment which says `/* Only get to here if not enough free heap to start scheduler */`.  Aha ... as scooby says "A clue ... a clue!!!".

However we have a new problem ... the stack size is governed by the configuration variable called `CONFIG_MAIN_TASK_STACK_SIZE` which is used when the cpu_start.c is compiled.  But cpu_start.c is supplied pre-compiled for us by the Arduino-ESP32 environment which means that we (as users) of the Arduino-ESP32 can't tweak anything to change the size.  I'm going to raise an issue against Arduino-ESP32 to see if we can't get it raised.

akshar001, 2017-10-26:
Hello Mr. Kolban thank you for your this great utility of yours . You have dug very much into esp32 i think. What i want to find out is actually esp and ble can be used together via esp idf core?
 Has anyone treid it?

akshar001, 2017-10-26:
Hello what i have found is something interesting. We can actually start the esp_bluedroid_init() and wifi.begin() together but when we i just only include one of the library file only include..it goes into same error.!

akshar001, 2017-10-26:
Sorry i might disturbing again...But thought one time i should share this..
I have deleted all the code from bleutils.h and tried it and all is working very fine.
I have also deleted generalutils.h and releted code..
i have successfully ran web server also.. I think that companies and manufraturer names might causing a problem if you can eliminate him your library can work like a heaven. 
Thanks

Cypher87, 2017-11-05:
I'm not sure if I should create a new issue, or add to this one. Anywho;
I can connect to my WiFi network, and send data. Then I scan for BLE devices and open a UDP connection to send information about the found BLE devices to my server. This works maybe the first 2 times, after that the WiFi connection state goes to WL_CONNECTION_LOST and I'm unable to connect to the WiFi network after that. Seems that BLE and WiFi does not work well together. Can we disable BT after the BLE scan is completed?

chegewara, 2017-11-05:
First of all, ble and wifi works togheter very well, just require some effort to setup settings good enough to avoid heap related isues etc. Its been proved with ESP32Explorer.
Second, yes, if you dont need to use BT anymore then you can disable it, and you can enable BT again if required.

My advise is to start with those settings:
- menuconfig->wifi-> change all settings as low as possible, then see if wifi network works stable,(this will give you about 30kB RAM)
- if you dont need to use classic BT just BLE (just realized that i switched of classic BT in last PR, so cant be used anyway) then add this line `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);` somewhere in your code, maybe as first line in `void app_main()`. This will release about 30kB

Then you can see how app will work and you can back

Cypher87, 2017-11-05:
I'm using the Arduino BLE package, with latest IDF. Not sure where to use menuconfig?

Anyway, I've added `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)` but that did not solve the issue.

My code: [https://pastebin.com/zsfBBrde](url)

My output:

> Connecting...
.......
Connected!
E (34999) BT: btc_search_callback  BLE observe complete. Num Resp 2
5
e6:33:09:67:68:15
Sending data

Where it's first connecting to my WiFi network. Then starts a scan and sends it using udp. "5" is the WiFi state, which is WL_CONNECTION_LOST.

chegewara, 2017-11-05:
@Cypher87 I will try to help you somehow, just i hope its nothing very important because i didnt sleep today and my brain isnt working too good. This is my output from your code(code works great):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0010,len:4

load:0x3fff0014,len:812

load:0x40078000,len:0

load:0x40078000,len:10164

entry 0x400789f8

Connecting...

[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 2 - STA_START

........[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 4 - STA_CONNECTED

[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 7 - STA_GOT_IP

.

Connected!

[D][BLEScan.cpp:178] start(): >> start(duration=3)

[D][BLEAdvertisedDevice.cpp:353] setRSSI(): - setRSSI(): rssi: -42

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 12, data: 554152542053657276696365

[D][BLEAdvertisedDevice.cpp:342] setName(): - setName(): name: UART Service

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x0a (ESP_BLE_AD_TYPE_TX_PWR), length: 1, data: eb

[D][BLEAdvertisedDevice.cpp:391] setTXPower(): - txPower: -21

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x12 (ESP_BLE_AD_TYPE_INT_RANGE), length: 4, data: 20004000

[D][BLEAdvertisedDevice.cpp:278] parseAdvertisement(): Unhandled type: adType: 18 - 0x12

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06

[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 12, data: 554152542053657276696365

[D][BLEAdvertisedDevice.cpp:342] setName(): - setName(): name: UART Service

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.

E (8946) BT: btc_search_callback  BLE observe complete. Num Resp 1

[D][BLEScan.cpp:205] start(): << start()

3

24:0a:c4:03:b1:c2

Sending data

3

Sending done

[D][BLEScan.cpp:178] start(): >> start(duration=3)

```

I dont know what wifi state is descibed by code "3", but as you can see there is message about sending data and about sending done. Is it output you are expecting?

chegewara, 2017-11-05:
To have clear situation would be better to start new issue

nkolban, 2017-11-05:
Let us indeed start a new issue for this problem and leave this one for the Boot loop issue when using BLE and WiFi.   We appear to have a report of this now happening in pure ESP-IDF (i.e. none of the code used in this project).  See:

https://github.com/espressif/esp-idf/issues/1198

nkolban, 2017-12-14:
Closing for just now

closed 2017-12-14 by nkolban

hanadjerbi1, 2018-08-14:
https://github.com/espressif/arduino-esp32/issues/1756