nkolban/esp32-snippets #112: Output of getManufacturerData is empty

pit001, 2017-10-06:
In my understanding the call to
`String manufacturer = advertisedDevice.getManufacturerData().c_str();`
should give me back the ManufacturerData as String.

**Estimated result:**
BLE Advertised Device found: Name: CC2650 SensorTag, Address: c4:be:84:71:39:84, manufacturer data: 0d00030000, serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb
ManufacturerData: [**0d00030000**]

**Code segment:** 
```
Serial.print("BLE Advertised Device found: ");Serial.println(advertisedDevice.toString().c_str());
String manufacturer = advertisedDevice.getManufacturerData().c_str();
Serial.print("ManufacturerData: [") + manufacturer + "]');
```

**Result:**
BLE Advertised Device found: Name: CC2650 SensorTag, Address: c4:be:84:71:39:84, manufacturer data: 0d00030000, serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb
ManufacturerData: []

in BLEAdvertisedDevice.cpp - where gets m_manufacturerData filled ?
Peter

nkolban, 2017-10-07:
I haven't tested but studying the story, it seems that you are trying to print the manufacturer data.  If the manufacturer data is:

0x0d 0x00 0x0x03 0x00 0x00

then none of those are in the ASCII printable/visible set.  The manufacturers data is not  necessarily an ASCII string.  In this case, it looks like 5 bytes of binary.  In C++, the std::string class can be used to contain a dynamic sequence of bytes such that std::string.data() returns a pointer to the data and std::string.length() returns the number of bytes available.

pit001, 2017-10-09:
Hi, you are right ! 
I've updated my code. Thx for investigation and hint. Can be closed !
Peter

closed 2017-10-09 by pit001

dominicklee, 2019-04-04:
> I haven't tested but studying the story, it seems that you are trying to print the manufacturer data. If the manufacturer data is:
>
> 0x0d 0x00 0x0x03 0x00 0x00
>
> then none of those are in the ASCII [...]

@nkolban Could you please share some example code to Serial print the manufacturer data with a key of 0x700 with a message of "hello" ?