nkolban/esp32-snippets #113: adding String list of GATT Descriptors to BLEUtils.c

pit001, 2017-10-06:
Howto:
get the list from https://www.bluetooth.com/specifications/gatt/descriptors

Characteristic Aggregate Format 	org.bluetooth.descriptor.gatt.characteristic_aggregate_format 	0x2905 	GCD
Characteristic Extended Properties 	org.bluetooth.descriptor.gatt.characteristic_extended_properties 	0x2900 	GCD
...

put it to [regex converter](https://regex101.com/)
reg exp: (.+) \t(.+) \t(.+) \tGCD
Sub: {\3,"\1"},

**and get** 

{0x2905,"Characteristic Aggregate Format"},
{0x2900,"Characteristic Extended Properties"},
...

**Add struct, head and end:**
typedef struct {
	uint32_t    assignedNumber;
	std::string name;
} gattdescriptor_t;

static const gattdescriptor_t g_descriptor_ids[] = {
{0x2905,"Characteristic Aggregate Format"},
{0x2900,"Characteristic Extended Properties"},
{0x2904,"Characteristic Presentation Format"},
{0x2901,"Characteristic User Description"},
{0x2902,"Client Characteristic Configuration"},
{0x290B,"Environmental Sensing Configuration"},
{0x290C,"Environmental Sensing Measurement"},
{0x290D,"Environmental Sensing Trigger Setting"},
{0x2907,"External Report Reference"},
{0x2909,"Number of Digitals"},
{0x2908,"Report Reference"},
{0x2903,"Server Characteristic Configuration"},
{0x290E,"Time Trigger Setting"},
{0x2906,"Valid Range"},
{0x290A,"Value Trigger Setting"},
{ 0, "" }
};

**make it available by function:**
std::string BLEUtils::gattDescriptorToString(uint32_t descriptorId) {
	gattdescriptor_t* p = (gattdescriptor_t *)g_descriptor_ids;
	while (p->name.length() > 0) {
		if (p->assignedNumber == descriptorId) {
			return p->name;
		}
		p++;
	}
	return "";
} // gattDescriptorToString

to public: in BLEUtils.h:
static std::string gattDescriptorToString(uint32_t descriptorId);

use it in code (not nice):
#include "BLEUtils.h"
...
String gattdesc_str = "00002901-0000-1000-8000-00805f9b34fb";
uint32_t gattdesc = strtol(gattdesc_str.substring(0, 8).c_str(), NULL, 16); //"2901"
Serial.println( String("Descriptor: ") + BLEUtils::gattDescriptorToString(gattdesc).c_str()); //Characteristic User Description

//Peter

nkolban, 2017-10-07:
Changes have been made.  A new function called
```
BLEUtils::gattDescriptorUUIDToString()
```

is now available.  Please review and if there are corrections needed, let us know.  If delighted, close the issue.

nkolban, 2017-10-23:
Assuming resolved.  Don't hesitate to re-open as needed.

closed 2017-10-23 by nkolban