nkolban/esp32-snippets #119: WebSocket.cpp Compilation Error

wiltOS, 2017-10-08:
For some reason WebSocket is giving some very strange errors. I have been trying to compile the bootwifi.c file located here: https://github.com/nkolban/esp32-snippets/tree/master/networking/bootwifi

I have compiled this before without a problem, but ever since we fixed the esp32 libcurl integration I have been getting an error. Here is the entire compilation:

https://pastebin.com/ezg9kVuz

Thank you for the help

nkolban, 2017-10-08:
I'm going to rebuild bootwifi to use the new HttpServer classes.

nkolban, 2017-10-09:
Coding 80% complete but now blocked waiting on the new ESP32 v3.0 C++ exception handling which should only be 2-3 weeks out.

nkolban, 2017-10-10:
The whole of the bootwifi code has been revamped.  It is now C++ based and no longer has Mongoose as a pre-req.  It is now 100% self contained.

nkolban, 2017-10-29:
Closing for just now.  Re-open as needed.

closed 2017-10-29 by nkolban