nkolban/esp32-snippets #122: BLE working on iPhone (LightBlue)

hsph, 2017-10-13:
Howdy! ;-) I want to send sensor values from a machine using an ESP32 and send it to an iPhone (7+) via BLE. At the moment I'm doing this with an Arduino and a HC-08 BT/BLE module. 
I already tried out your cpp BLE examples, but I cannot see the characteristics via LightBlue. I can only see the device and I can also connect it, but that's it.
Do you have any experience using an iPhone in combination with the ESP? 

Thanks in advance! 
[image]

nkolban, 2017-10-13:
Unfortunately my household is 100% android and I have no opportunity to test with an iPhone.  I also don't know what "LightBlue" is?  Is it an application or a product or something else?

My testing so far has been against real/physical BLE devices, Android nRF Connect application and Linux Bluetoothctl.  I'll be happy to partner with you at some mutually convenient time.  We might be able to hop on a skype session and see if there is something special/needed in iPhone land.  My ideal would be that you test with Linux Bluetoothctl and/or Android nRF Connect application and convince yourself that your ESP32 app runs well against those.   If it doesn't then we have a common environment against which to examine.

hsph, 2017-10-13:
I'll continue testing this for an hour or two, but then I have to drive home (currently I'm in our electronics lab at work). I don't think, that I have a lot of time this weekend, but I'll continue working on this next week. I'll let you know, if I find out something new. 
Unfortunately my dev computer doesn't have BLE. Maybe I can get an Android phone from a friend or colleague to test the connection.

hsph, 2017-10-13:
Short update: I got a Samsung tablet from a colleague and installed nRF Connect. I also flashed your 'SampleNotify' example. When I connect to the ESP32 with the nRF app I get the following screen: 
[image]

It seems like the characteristics for WRITE and NOTIFY are missing(?)

nkolban, 2017-10-13:
I think I see the puzzle ... I think when I read your original post I understood it to mean that you are seeing your advertised service but are missing characteristics in that service.   What I now understand is that you are not even seeing your advertised service (and hence not seeing ANY characteristics of that service).  It takes one time to get ones head around the concepts of devices, services, characteristics and descriptors :-)

I think we are suffering from the problem described here:

https://github.com/nkolban/esp32-snippets/issues/109

We appear to have a "workaround" which is to switch on the diagnostics in the BLE C++ classes.  Somehow (possibly a race condition) the classes work.  We switch off debugging, we seem to loose services.  Have a review of the problem (109) above and see if it makes sense.  We are banging our heads trying to understand what different switching on/off debugging makes ... and still have a debt to get 109 fixed.

clowrey, 2017-10-13:
Have you tried the nRf app(s) on iphone? I have used them with success with the ESP32 about a month ago.

nkolban, 2017-10-13:
@clowrey ... I'm assuming you are asking mr @hsph  as I personally don't have an iPhone for testing.

clowrey, 2017-10-13:
@nkolban yes I think it might help @hsph if he hasn't found that these nRF apps are also available in iOS app store and can be used instead of lightblue. Sorry should have been more clear. But the actual problem may very well be #109 as you mentioned above.

hsph, 2017-10-16:
So... I read through #109 and changed the compiler optimization level from -Og to -Os and now it works perfectly! I can even use the same characteristics with the same UUIDs as I did with the Arduino + HC-08 bluetooth chip. So the app on the iPhone can stay as it is.
Next step for me is to refactor my Arduino code to work with the ESP32. For that I'll try your Arduino libraries @nkolban . So, for me this issue is closed. Thanks for your help!

closed 2017-10-16 by hsph

bhavenm, 2018-04-30:
I know this issue is closed but I wanted to ask @hsph what did he specifically do to make it work. I cannot see the Bluetooth from esp32 on iPad but see it only on android devices.

I am not using any specific app. I am trying to look for Bluetooth devices on my iPad through the bluetooth settings section.

Some feedback will be very helpful.

chegewara, 2018-04-30:
Hi,
thats why i dont like Apple (just kidding, there i much more).
Try to play with those 2 functions:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L94-L102

iPhone and iPad is sensitive on a lot settings, try with intervals about 100ms

sanu-krishnan, 2018-05-01:
@bhavenm BLE devices won't be visible sometimes on Bluetooth search in smartphones.

Please use BLE scanner (android) or LightBlue(iOS) application, and search for Bluetooth devices using those apps

muktillc, 2018-05-03:
@sanu-krishnan. I am using the Lightblue and I am able to see the ESP32. I am just wondering that if I am developing my own iOS app, is there something special that I need to do to make it work to find and connect the esp32 ? 

@bhavenm what esp32 board are you using?

ifrew, 2018-05-05:
I have same issue. At first I couldn't even see the esp32 but that was because I was using an old version of BLE library without the fix for #109 which basically set the service id to be the primary service. Great find btw. Now I see it in light blue although the summary page says no services for the device but clicking on esp32 device shows the service and characteristics. I then played around changing the advertising max interval using the API @chegewara suggested , starting at 100ms and going up to 500ms in 50ms increments. That didn't work either. I'm assuming the compiler optimization work around mentioned earlier is not needed. On ardunio ide its already set to -Os anyway. Interestingly my Windows phone can connect now when previously it couldn't see esp32. I'll have a look at the apple development guide for ble to see if that sheds any light.

rabee05, 2020-11-22:
> Unfortunately my household is 100% android and I have no opportunity to test with an iPhone. I also don't know what "LightBlue" is? Is it an application or a product or something else?
>
> My testing [...]

Hi @nkolban, I am using Linux Bluetoothctl and trying get notification data based on your code 
https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_notify on ESP32.

everything woks fine except notify. The only way I can get data via notify by connecting the ESP32 to iphone via BLE scanner, then use Node App using Bluetoothctl under the hood. I have to do  this every time I turn off the ESP32. 

Thank you so much for your help.

Best,
RZ

chegewara, 2020-11-22:
You can add some code to esp32 example to make it works. This will be a workaround, just write value 0x01 or 0x02 to 0x2902 descriptor on characteristic that will send notify. I think it should works.

rabee05, 2020-11-23:
@chegewara  Thank you very much! you are absolutely correct. The main issue with descriptor 0x2902. after some investigations and tried to read descriptor value which was always 0, however when I connect BLE scanner via phone the value get chnaged to 1 and stay 1 until ESP32 unplugged or rebooted. I tried to send 0x01 or 0x02 to descriptor as you proposed using NOBLE 
  descriptors[0].writeValue(Buffer.from(['1'], 'hex'), (err) =>console.log(err));
  descriptors[0].writeValue(Buffer.from(['2'], 'hex'), (err) =>console.log(err));

but without any luck. would it possible to send the value  on connect in the example "BLE_notify "

`void loop() {
    // notify changed value
    if (deviceConnected) {
       pCharacteristic->setValue((uint8_t*)&value, 4);
       pCharacteristic->notify();
       Serial.println(value);
       value++;

        delay(1000); // bluetooth stack will go into congestion, if too many packets are sent, in 6 hours test i was able to go as low as 3ms
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        Serial.println("When connecting"); 
        delay(1000);
        oldDeviceConnected = deviceConnected;
    }
}`

@nkolban 
Thank you very much. 
RZ

chegewara, 2020-11-23:
You can set that descriptor value in setup or in onConnec/onDisconnect callbacks.