nkolban/esp32-snippets #126: Compile error with BLE_server example of BLE arduino

tomorrow56, 2017-10-15:
Using the arduino BLE_server.ino example from the library, the compile error occurred.

```
Arduino：1.8.2 (Windows 10), board: ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                                              ^

In file included from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                                                                                                        ^

D:\MyDocuments\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:131:118: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

    BLERemoteCharacteristic* pBLERemoteCharacteristic = pBLERemoteService->getCharacteristic(BLEUUID(evtParam->notify.char_id.uuid));

                                                                                                                      ^

In file included from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:133:111: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

     ESP_LOGE(LOG_TAG, "Could not find characteristic with UUID %s for notification", BLEUUID(evtParam->notify.char_id.uuid).toString().c_str());

                                                                                                               ^

D:\MyDocuments\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:196:55: error: no matching function for call to 'BLEUUID::BLEUUID(esp_gatt_id_t&)'

    BLEUUID uuid = BLEUUID(evtParam->search_res.srvc_id);

                                                       ^

In file included from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteCharacteristic.h:18:0,

                 from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:16,

                 from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.h:18,

                 from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note: candidate: BLEUUID::BLEUUID()

  BLEUUID();

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note:   candidate expects 0 arguments, 1 provided

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:25:2: note: candidate: BLEUUID::BLEUUID(esp_gatt_srvc_id_t)

  BLEUUID(esp_gatt_srvc_id_t srcvId);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:25:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:24:2: note: candidate: BLEUUID::BLEUUID(uint8_t*, size_t, bool)

  BLEUUID(uint8_t* pData, size_t size, bool msbFirst);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:24:2: note:   candidate expects 3 arguments, 1 provided

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:23:2: note: candidate: BLEUUID::BLEUUID(esp_bt_uuid_t)

  BLEUUID(esp_bt_uuid_t uuid);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:23:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_bt_uuid_t'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:22:2: note: candidate: BLEUUID::BLEUUID(uint32_t)

  BLEUUID(uint32_t uuid);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:22:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint32_t {aka unsigned int}'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:21:2: note: candidate: BLEUUID::BLEUUID(uint16_t)

  BLEUUID(uint16_t uuid);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:21:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint16_t {aka short unsigned int}'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:20:2: note: candidate: BLEUUID::BLEUUID(std::__cxx11::string)

  BLEUUID(std::string uuid);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:20:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(const BLEUUID&)

 class BLEUUID {

       ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'const BLEUUID&'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(BLEUUID&&)

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'BLEUUID&&'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:197:94: error: no matching function for call to 'BLERemoteService::BLERemoteService(esp_gatt_id_t&, BLEClient*)'

    BLERemoteService* pRemoteService = new BLERemoteService(evtParam->search_res.srvc_id, this);

                                                                                              ^

In file included from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.h:18:0,

                 from D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2: note: candidate: BLERemoteService::BLERemoteService(esp_gatt_srvc_id_t, BLEClient*)

  BLERemoteService(esp_gatt_srvc_id_t srvcId, BLEClient* pClient);

  ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note: candidate: BLERemoteService::BLERemoteService(const BLERemoteService&)

 class BLERemoteService {

       ^

D:\MyDocuments\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note:   candidate expects 1 argument, 2 provided

exit status 1
```
Regards,

tomorrow56, 2017-10-15:
This issue was solved by using latest version of "Arduino core for the ESP32" and "The library source for the ESP32 BLE support for Arduino".
Thanks!

closed 2017-10-15 by tomorrow56

nkolban, 2017-10-15:
That's great news ... thanks for reporting and thanks for closing.  Don't hesitate to create new issues as they come up.