nkolban/esp32-snippets #129: barfs on BLE connect

tedder, 2017-10-17:
code:
```bool connectToServer(BLEAddress pAddress, BLEUUID serviceUUID) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());
    
    BLEClient*  pClient = BLEDevice::createClient();
    Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");
[...]

BLEAddress addr = BLEAddress((std::string)"e7:5b:f9:bf:02:52");
connectToServer(addr, ...);
```

output:
```Forming a connection to e7:5b:f9:bf:02:52
 - Created client
E (1409) BT: btc_gattc_call_handler()
Guru Meditation Error of type LoadStoreAlignment occurred on core  1. Exception was unhandled.
(register dump follows)```

Obviously I'm failing to create the connection. I haven't been able to figure out why.

nkolban, 2017-10-17:
Howdy my friend ... lets try the following.

Lets try:

```
BLEAddress addr = BLEAddress("e7:5b:f9:bf:02:52");
```

instead of:

```
BLEAddress addr = BLEAddress((std::string)"e7:5b:f9:bf:02:52");
```

nkolban, 2017-10-17:
... and if that doesn't work, lets try the debug and stack decoding recipes found here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

tedder, 2017-10-17:
Hi nkolban, thanks for the quick and friendly response. A couple of things before I get to the workaround:

- the Exception Decoder won't run against my board (adafruit esp32) because something is identifying as "arduino-esp32" instead of "esp32". I added to a ticket over there about that.
- c_string vs std::string, no change. still died.

And the fix/workaround. I was reading #45, enabled debugging, and it works. The debug is below, if that helps. So it's clearly a deadlock of some sort. I'll start going further - adding my serviceUUID/characteristics code - but I'm guessing it will also work, that the debug workaround is sufficient.

```
 trying addr: e7:5b:f9:bf:02:52
Forming a connection to e7:5b:f9:bf:02:52
 - Created client
D (51674) BLEClient: >> connect(e7:5b:f9:bf:02:52)
D (51674) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffe7270), owner: <N/A> for connect
D (51675) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffe7270), owner: connect
E (51682) BT: btc_gattc_call_handler()
V (51682) FreeRTOS: Semaphore waiting: name: RegEvt (0x3ffe7270), owner: connect for connect
D (51686) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
D (51701) BLEUtils: GATT Event: ESP_GATTC_REG_EVT
D (51706) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]
V (51710) FreeRTOS: Semaphore released: name: RegEvt (0x3ffe7270), owner: connect
V (51710) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffe7270), owner: connect
D (51718) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffe72d0), owner: <N/A> for connect
D (51733) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffe72d0), owner: connect
V (51740) FreeRTOS: Semaphore waiting: name: OpenEvt (0x3ffe72d0), owner: connect for connect
E (81741) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0100
E (81742) BT: p_conn is NULL in bta_gattc_conn_cback

E (81743) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0100
E (81750) BT: p_conn is NULL in bta_gattc_conn_cback

D (81755) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
D (81762) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (81768) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: e7:5b:f9:bf:02:52]
D (81775) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
D (81783) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (81788) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: e7:5b:f9:bf:02:52]
D (81795) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
D (81803) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (81807) BLEUtils: [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: e7:5b:f9:bf:02:52, mtu: 0]
V (81816) FreeRTOS: Semaphore released: name: OpenEvt (0x3ffe72d0), owner: connect
V (81816) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffe72d0), owner: connect
D (81823) BLEClient: << connect(), rc=1
 - Connected to server
done trying addr: e7:5b:f9:bf:02:52
```

chegewara, 2017-10-17:
> the Exception Decoder won't run against my board (adafruit esp32) because something is identifying as "arduino-esp32" instead of "esp32". I added to a ticket over there about that.

In arduino->hardware folder is Arduino-esp32, rename it to esp32 and this will be fine.

nkolban, 2017-11-04:
Closing for just now.  Re-open as needed.

closed 2017-11-04 by nkolban