nkolban/esp32-snippets #130: HttpServer.cpp compilation error

prasad-dot-ws, 2017-10-17:
I'm trying to compile cpp_utils components, But it gives me this error,

CXX build/cpp_utils/HttpServer.o
/Users/Prasad/esp/https_request/components/cpp_utils/./HttpServer.cpp: In member function 'virtual void HttpServerTask::run(void*)':
/Users/Prasad/esp/https_request/components/cpp_utils/./HttpServer.cpp:195:25: error: exception handling disabled, use -fexceptions to enable
    catch(std::exception e) {

nkolban, 2017-10-17:
Howdy my friend.  The cpp_utils now requires the very latest ESP-IDF and the latest toolchain.  I suspect that you are running an older version of the ESP-IDF and toolchain?

The ESP-IDF I am using was pulled from here:

https://github.com/espressif/esp-idf

and the tool chain from 

http://esp-idf.readthedocs.io/en/latest/get-started/index.html#setup-toolchain

prasad-dot-ws, 2017-10-17:
Hi Sir! Thank you for your response. I was trying to update the xtensa toolchain and i stuck here, Would you mind helping me out there?

https://www.esp32.com/viewtopic.php?f=2&t=3362

nkolban, 2017-10-17:
Sorry ... I have no skills on Mac.  What I normally recommend is that folks work on Linux (my favorite is Ubuntu).  That is the platform that I believe that the majority of ESP32 programmers work on and where you will find the majority of support.  I think you can install and run Ubuntu using Virtual Box on your Mac.

joehui, 2017-10-22:
I had the same issue when trying the project.  I fixed it by running "make menuconfig", go into "compiler options" and enabled C++ Exception handling.

nkolban, 2017-10-22:
Thank you @joehui ... that's a great catch.  That looks like a new option.  First there was no exception handling, then it was added by default and now its optional.  Its all good as the ESP-IDF v3.0 is still pre-release.

chegewara, 2017-10-23:
https://github.com/espressif/esp-idf/commit/9c7477ef34ac8260c908b637fdc3c291a018425c

nkolban, 2017-11-09:
Closing for now.  Re-open as needed.

closed 2017-11-09 by nkolban