nkolban/esp32-snippets #131: BootWifi.boot() failing

andyrutter90, 2017-10-17:
BootWifi.boot() hangs (or crashes) in esp_wifi_init_internal().  I must be doing something stupid, but I can't think what it is. My main.cpp is below. The esp-idf-template runs OK when using the same sdkconfig.

#include "freertos/FreeRTOS.h"
#include "esp_system.h"
#include "driver/gpio.h"
#include "BootWiFi.h"

#define LED_PIN   GPIO_NUM_2

BootWiFi boot;

extern "C" void app_main(void)
{
    boot.boot();

    gpio_set_direction(LED_PIN, GPIO_MODE_OUTPUT);
    int level = 0;
    while (true) {
        gpio_set_level(LED_PIN, level);
        level = !level;
        vTaskDelay(300 / portTICK_PERIOD_MS);
    }
}

nkolban, 2017-10-18:
Howdy my friend.  GREAT find.  I was able to recreate form your sample just fine.  MANY thanks for that.  The problem is interesting.   If we look at your code, you have a global/static instance called `boot`.  What I mean by this is that you have created an instance of "boot" that is global to this file and static.  What that means is that the instance of the called (`BootWiFi`) is constructed when the C++ runtime is initialized.  This occurs **before** the FreeRTOS runtime is initialized.  The BootWiFi class requires the FreeRTOS runtime to be available because we create semaphores and other items during class instance construction.  If we change your code to:
```
...
extern "C" void app_main(void)
{
   BootWiFi boot;
   boot.boot();
   ...
```

all appears to work.  If need a global, then we can change the code to:

```
BootWifi *pBoot;
extern "C" void app_main(void)
{
   pBoot = new BootWifi();
   boot->boot();
   ...
```

Can you review this post and see if it makes sense and test it in your environment.  Is this new usage pattern satisfactory for you?  If not, it would be possible for me to reworks the internals of BootWiFi to accommodate late creation of FreeRTOS data on demand.

andyrutter90, 2017-10-18:
Hi. Thank you for the clear explanation and the change to a variable instantiated inside app_main() works fine. I thought I must be doing something simple but stupid! No need to make changes to the code, a comment in the README will suffice to stop others making the same mistake.

BTW, many thanks for the excellent collection of code.  My plans include some of the enhancements you list in the BootWiFi README, I will send the changes if any of it looks sensible.

closed 2017-10-18 by andyrutter90