nkolban/esp32-snippets #135: Memory leak?

chegewara, 2017-10-22:
Hi,
im trying to bite somehow ESP32_Explorer project, but its still lack of RAM. This is what ive found:
`1604 bytes (@ 0x3ffe2698) allocated CPU 0 ccount 0x035a0558 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe2cec) allocated CPU 0 ccount 0x035a4ccc caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe3340) allocated CPU 0 ccount 0x035a7e24 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe3994) allocated CPU 0 ccount 0x035aaf7c caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe4374) allocated CPU 0 ccount 0x035ae0d4 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe49c8) allocated CPU 0 ccount 0x035b1360 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe501c) allocated CPU 0 ccount 0x035b45ec caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe5670) allocated CPU 0 ccount 0x035b787c caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe5cc4) allocated CPU 0 ccount 0x035bafe8 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe6318) allocated CPU 0 ccount 0x035be274 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe696c) allocated CPU 0 ccount 0x035c1500 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe6fc0) allocated CPU 0 ccount 0x035c4790 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe7614) allocated CPU 0 ccount 0x035c7a1c caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe7c68) allocated CPU 0 ccount 0x035caca8 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe82bc) allocated CPU 0 ccount 0x035cdf34 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe8910) allocated CPU 0 ccount 0x035d11c4 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe8f64) allocated CPU 0 ccount 0x035d4450 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe95b8) allocated CPU 0 ccount 0x035d76dc caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffe9c0c) allocated CPU 0 ccount 0x035da968 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffea260) allocated CPU 0 ccount 0x035ddbf8 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffea8b4) allocated CPU 0 ccount 0x035e0e84 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffeaf08) allocated CPU 0 ccount 0x035e4110 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffeb55c) allocated CPU 0 ccount 0x035e739c caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffebbb0) allocated CPU 0 ccount 0x035ea62c caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
1604 bytes (@ 0x3ffec204) allocated CPU 0 ccount 0x035ed8b8 caller 0x400ec44a:0x400e2f3b:0x400d9226:0x400d9891:0x400d2c51:0x40145907:0x40142591:0x401061e0:0x401416cf:0x40000000
`

It looks like "some loop" in here: (after decoding)

> 0x400ec44a: wDev_Rxbuf_Init at ?? line ?
0x400e2f3b: ic_init at ?? line ?
0x400d9226: chip_post_init at ?? line ?
0x400d9891: wifi_init at ?? line ?
0x400d2c51: esp_wifi_init_internal at ?? line ?
0x40145907: esp_wifi_init at /home/esp32/esp/esp-idf/components/esp32/./wifi_init.c line 23
0x40142591: WiFi::WiFi() at /home/esp32/ESP32_Explorer/components/cpp_utils/./WiFi.cpp line 531
0x401061e0: WiFiTask::run(void*) at /home/esp32/ESP32_Explorer/main/./main.cpp line 43
0x401416cf: Task::runTask(void*) at /home/esp32/ESP32_Explorer/components/cpp_utils/./Task.cpp line 59

nkolban, 2017-10-23:
I haven't been able to recreate this issue yet.  I did find other problems with the latest builds ... specifically, we appear to run out of free RAM.

chegewara, 2017-10-23:
To recreate it you have to turn on heap memory debbuging in menuconfig and add few lines of code to get it print out. 

>extern "C"{
#include "esp_heap_trace.h"
}
 int app_main(void) {
	   ESP_ERROR_CHECK( heap_trace_init_standalone(trace_record, NUM_RECORDS) );
	ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) ); 
	task_webserver(nullptr);
	ESP_ERROR_CHECK( heap_trace_stop() );
	heap_trace_dump();
	return 0;
} 

http://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html

chegewara, 2017-10-23:
@nkolban Finally. I have managed to do some tweaks and run wifi with ble on this stack. 
The OP was asking about memory leak, well its not memory leak its just esp-idf allocating RX static buffers. Almost 25 buffers 1.6kB each, but its managable with menuconfig, and atm i have setup 5 static RX buffers and esp32_explorer is running, which means wifi and ble can run togheter on this stack.
Of course You have to apply those tweaks we were talking about on irc.
Have a nice play.

closed 2017-10-23 by chegewara