nkolban/esp32-snippets #136: BLE_client.ino and additional callbacks

reaper7, 2017-10-23:
BLE_client.ino example works for me, connects correctly with remote service and reads the notify from remote characteristic 👍 
After a few tests I found in uart log two interesting information:
- after scan complete (30 sec passed):
```
E (31433) BT: btc_search_callback  BLE observe complete. Num Resp 1
```

<br>

- when I switch off remote device (or connection lost) then I got:
```
E (85209) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
```

It is possible to create a local callbacks for these events? Would be very useful for me.

chegewara, 2017-10-23:
I think its implemented already:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L391

nkolban, 2017-10-23:
@reaper7  Howdy, can we clarify the function you are looking for.  There are callbacks and blocks.

For example, when we wish to scan for BLE devices, we get a list of found devices when the scan ends but we can also register a callback handler that is invoked each time a new device is seen.  That way we can preempt the rest of the scan since we may have found what we want earlier.   These are "scanning" callbacks.

The second story ... being notified when a client connection is lost/dropped.  Is that the other item you are talking about?

reaper7, 2017-10-23:
**First I try to connect to ble multimeter, so I start scan for 30 sec.**
now we have two way:
**1.** interesting device (multimeter) found by AdvertisedDeviceCallbacks->onResult
in this case scan is stopped by **advertisedDevice.getScan()->stop()** before 30 seconds,
I do not receive any event in the logs.
I'm happy because I can play with my multimeter 👍 

**2.** scan over declared 30 sec do not found my device (multimeter),
scan ended on declared timeout 30 sec with event:
```
E (31433) BT: btc_search_callback  BLE observe complete. Num Resp 1
```
in this case I need restart scan again (and again until multimeter not found)

For this reason (2) I would like to know when scan is finished through timeout (not getScan()->stop())
<br>
**Second, when I successfully connected to multimeter**
for various reasons (too big distance/multimeter switch off) active connection may be broken/interrupted
in this case I got event:
```
E (85209) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
```
and in this situation I have to start scanning from top of this message :)

nkolban, 2017-10-23:
Let's look at 1 puzzle at a time.  First the idea of scanning for a BLE server and finding it or not finding it.  We have the notion that for each BLE server we find, we invoke a callback to indicate that a new server is found.  If we find the server we are looking for, we can stop further searching.  If after the time interval we still haven't found the desired server, then the call to scan unblocks and we wake up.

What I think I am hearing you say is that you want an event that declares that the scanning has finished because we have scanned for the maximum period we wanted to scan for.

Lets ask some more questions:

1) If we want to just "keep on scanning", can't we just scan for 1,000,000 seconds?  That way we won't ever stop scanning until we find the server we want?

2) If we do want to scan in "chunks", perhaps scan for a while, pause, scan for a while, pause etc ... could we not have logic such as:

```
bool found = false;
while(!found) {
    scan(30);
}
```
The callback function would then set the found flag to be true when the target BLE server is found.

Neither of these seem to need involving a new callback function.

To help me, can you describe a purpose/pattern for getting a callback to say that the scan interval has passed that wouldn't be solved by the previous two thoughts?

reaper7, 2017-10-24:
@nkolban - I know that we have "7" solutions for solve my problem, that's mean we can find next "7" solutions for this. Your both methods is very good!
Currently I have bool flag inside AdvertisedDeviceCallbacks->onResult :)

But question is: if we have some built-in methods for get scan states like for e.g. "onEnd", "onFoundNewDev".
Looking on uart log and event "...btc_search_callback...", seems that somewhere exists, but do we have an api for catch it?

nkolban, 2017-10-25:
At the ESP-IDF API level, we get events that tell us when a scan result is found and when scanning completes.  I believe those are the only two events that relate to scan outcomes.  Today, when we get a new scan result, we call the callback onResult() if the scan is for a device we haven't seen before.  The completion event does not have a callback because the scan() method simply blocks until the scan completes.  Internally it is the scan complete event that causes us to unblock the scan() call.

The ESP-IDF provides APIs that provide an interface to BLE programming.  The C++ classes use those.  Under the covers BLE is implemented by a stack called "bluedroid" and it is messages from the very low level environment that you are seeing in the console log.  I have no idea what these messages mean as they originate from a level beneath the ESP-IDF.

reaper7, 2017-10-25:
Ok, tnx for clarification 👍 
I will keep my solutions.

closed 2017-10-25 by reaper7