nkolban/esp32-snippets #137: Supporting multiple connection on a BLEServer

ayavilevich, 2017-10-23:
Hi Neil,

I have started working with the library with the intention of supporting the ESP32 BLE in one of my projects. Currently it uses an AVR Arduino with HM-10 compatible modules.
One of my interests is supporting connecting more than one client to the server at the same time.

I looked at the code and the examples and it seems it isn't implemented in the c++ library. Looking at BLEServer I can see it has one connection id (m_connId). OTOH it has a m_connectedCount field.
What is the goal with regards to the library and supporting multiple connections? Is it planned? Is help needed?

Regards,
Arik.

nkolban, 2017-10-23:
Howdy, in the story, it isn't clear to me what role the ESP32 plays.  Is the ESP32 the server that will service multiple parallel connections from external BLE clients or do you want a single ESP32 client to be able to form multiple parallel connections to external BLE Servers?

ayavilevich, 2017-10-24:
Hi, in this scenario, the ESP32 is the server (peripheral) that needs more that one client (central) connected to it at the same time. For example, a temperature sensor with several devices connected to it and getting the sensor data via notifications.

The difficult part IMO is the design. Supporting such a scenario will make the API more complex. Without it the functionality will be partial. Perhaps it is better to have two classes, a BLEServer and a BLEMultiServer. Possibly related by inheritance...

You can see an API example that supports multiple clients (devices) per server in the Android API: https://developer.android.com/reference/android/bluetooth/BluetoothGattServer.html

What do you think?

nkolban, 2017-10-25:
Ok ... I think I understand ... you want an ESP32 running the BLE C++ classes as a server to be able to handle multiple in-coming client connections concurrently.  To be honest, I have never considered that so what I will need to do is setup some kind of test environment and attempt to connect two in parallel and see what happens.

akshar001, 2017-10-31:
Hello @ayavilevich i also need exactly same kind of thing so if you find anything please let us know.

ayavilevich, 2017-10-31:
@akshar001 AFAIK the c esp-idf API does allow multiple client connection on a server as well as connecting to multiple servers from a client. Any evidence that it is not already implemented in ESP32?

akshar001, 2017-10-31:
Hello @ayavilevich no there is no evidence for that because it might be already implemented..
I actually have succeeded in what we want to do..
Actually what happens is when esp gatt_server connected to client than it stops advertising..
so we have to start advertising everytime device get connects i have done that thanks to @nkolban library..we can set onconnect callback and in that function we can start advertising again..
I have only worked with 2 devices i don't know about more.. you can try and close this issue for now.

ayavilevich, 2017-10-31:
Hi @akshar001 , you are right about restarting advertising, however the issue is not resolved yet. The internals of the server class do not support several connections. If you would notify on a characteristic, only one client will get the notification. There might be other issues. Restarting advertising is only part of the solution.

akshar001, 2017-10-31:
@ayavilevich yes you are right also... then let's solve out the other things..atleast we both want a exact thing..

akshar001, 2017-10-31:
For sending notification to every client you have to edit the "BLECharacteristic.cpp" line number 526 from..
`for(int i = 0;i<getService()->getServer()->getConnectedCount();i++){

	esp_err_t errRc = ::esp_ble_gatts_send_indicate(
			getService()->getServer()->getGattsIf(),
			i,
			getHandle(), length, (uint8_t*)m_value.getValue().data(), false);
	printf("sent notification to::%d\n",i);

}`

basically we have to notify every client...

now it has may some errors after words but i have tried in 3 devices and it works fine..

RottenVanSuti, 2018-04-30:
hello @akshar001 What I understand is that you restart the advertising with onconnect callback function. Once the advertising is restarted, the previous device keeps paired?  Where is the onconnect callback function? I'm sorry if my questions are a little silly, but I'm no so expert yet. Thanks

akshar001, 2018-05-31:
By asking we grow so don't worry about it? But you found onconnectcallback thing? sorry for late reply!