nkolban/esp32-snippets #138: WiFi.scan hangs at ESP_ERROR_CHECK( esp_wifi_init(&cfg) );

rweller59, 2017-10-24:
Kolban, having a bit of a problem with the cpp_utils. When I scan for wifi points my code hangs at (about line 350 in wifi.cpp) ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
      wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
      ESP_LOGD(LOG_TAG, "wifi scan 1"); ***************** this gets logged
      ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
      ESP_LOGD(LOG_TAG, "wifi scan 2");***************** this does not logged

I wrote the following that is virtually the same (as far as I can tell) and it runs fine:

	// initialize the wifi event handler
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
	// configure, initialize and start the wifi driver
	wifi_init_config_t wifi_config = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&wifi_config));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_start());
	printf("initializing scan process...");
	// configure and run the scan process in blocking mode
	wifi_scan_config_t scan_config = { };
	printf("Start scanning..."); ***************** keeps running from here and returns available access points

with associated event handler:
// Empty event handler
       static esp_err_t event_handler(void *ctx, system_event_t *event)
       { return ESP_OK; }

Any thoughts on what I might be able to look at?

nkolban, 2017-10-25:
Howdy,
I had a go at recreating the problem but all seemed to work.  It may take us a bit to get it diagnosed.

Here is a FreeRTOS task that I used for testing:

```
#include <WiFi.h>
#include <FreeRTOS.h>
#include <string>
#include <stdio.h>

#include "sdkconfig.h"

void task_test_wifi_scan(void *ignore) {
	WiFi wifi;
	auto results = wifi.scan();
	for (auto it = results.begin(); it != results.end(); ++it) {
		printf("%s\n", it->toString().c_str());
	}
	FreeRTOS::deleteTask();
}
```

If the problems persist, we will want to switch on logging to the verbose level.  That will show us everything that is going on inside.  If you can create your smallest possible recreation of the issue and pastebin your own application, I can test that here.

I currently use the latest available ESP-IDF from the Git master branch.

rweller59, 2017-10-25:
Ok, I created a new source file, cut and pasted your task into it, ran it from main and it runs fine.

I then moved wifi declaration outside of function (like I had in my other code) and it hangs at the same spot as my code did previously. 

#include <WiFi.h>
#include <FreeRTOS.h>
#include <string>
#include <stdio.h>

#include "sdkconfig.h"

WiFi wifi;
void task_test_wifi_scan(void) {
	//WiFi wifi;
	auto results = wifi.scan();
	for (auto it = results.begin(); it != results.end(); ++it) {
		printf("%s\n", it->toString().c_str());
	}
	FreeRTOS::deleteTask();
}

nkolban, 2017-10-25:
Aha ... I think we have a winner.  See also this issue #131.

At present, we can't create a "static" WiFi as the WiFi constructor is run before the rest of the ESP-IDF is initialized.  We can only construct as WiFi instance after control is given to your app.

I have create issue #140 to change the implementation to allow static initalization.  In the meantime, see the workarounds in #131 (it uses BootWiFi but the principle is the same).  Apologies for the inconvenience ... and MANY thanks for taking the time to report the problem ... we'll get it fixed soon.

rweller59, 2017-10-25:
thanks for the quick response and, looks like a pretty simple workaround that I will implement later today.

I realize that this is off subject but since I have your attention... I'll be working through mqtt next and was going to start by reviewing your snippets. Any gotchas to be aware of implementing in eclipse c++?

nkolban, 2017-10-25:
Maybe have a review of this ...

https://www.youtube.com/watch?v=-ttiPfmrehU

If you run into troubles, post to the ESP32 forum found here:

https://esp32.com/index.php

and search through the history.  There are lots and lots of great historic posts.

rweller59, 2017-10-25:
making the global * worked out just fine.
I've added a couple of things to your my copy of your wifi.cpp code. Don't know if you're interested in this type of input - if not, let me know.
 - added around line 375 because rssi was not captured:  wifiAPRecord.m_rssi = list[i].rssi;
 - added around line 380 (just before free(list)) because I want to connect to the strongest signal:
	std::sort(apRecords.begin(), apRecords.end(), [](const WiFiAPRecord& lhs, const WiFiAPRecord& rhs)
			{ return lhs.m_rssi > rhs.m_rssi; });

nkolban, 2017-10-25:
These are great additions.  You are welcome to report them just like this or use Github and make the changes yourself.  Either are great and welcomed.  Many thanks.  I'll post back when the code is changed in the master.

nkolban, 2017-10-26:
Again ... @rweller59 Thank you for the code changes.  I have made and committed them.  Your knowledge of C++ is great ... I had to study the sort command to even begin to figure out the very start of your knowledge.

rweller59, 2017-10-26:
wish that were true. I credit my googling skills.

Here is another issue I'm struggling with... If you would like me to open new issues with this kind of stuff, let me know
When I scan for wifi ap's, no problem. When I connect to an ap after the scan, I error out and esp32 reboots. APconnect works find by itself;
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x40132d51
0x40132d51: _ZN4WiFi9connectAPERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESa
IcEEES7_b at C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/W
iFi.cpp:168 (discriminator 2)

void newScan(void) {
	auto results = ptr_wifi->scan();
	for (auto it = results.begin(); it != results.end(); ++it) {
		printf("%s\n", it->toString().c_str());
	}
	ESP_LOGD(TAG, "wifi scan complete");
}

void newConnect(void) {
	ptr_wifi->connectAP(ssid, password, true);
	ESP_LOGD(TAG, "wifi connect complete");
}

void newInit(void) {
	ptr_wifi = new WiFi;

	wifiEventHandler *eventHandler = new wifiEventHandler();
	ptr_wifi->setWifiEventHandler(eventHandler);

	newScan();
	newConnect();
...

nkolban, 2017-10-26:
Howdy @rweller59 

If I could indeed ask you to copy/paste this into a new issue that would be great.  Issues are cheap and allow me to track them independently, title them and tag them so I can know which ones require attention.  In addition, I find that other users search issues and if we can keep them focused on just one problem that would be great.  Please keep the problem reports coming and for my part I'll try and resolve them ASAP.

closed 2017-11-08 by rweller59