nkolban/esp32-snippets #139: How should I execute Sampleclient.cpp programs in Eclipse

vishnunaik, 2017-10-25:
Dear Sir, 
I followed your tutorial: ESP32 Technical Tutorials: Building a C++ Application. I build and execute the code which is presents under the esp32-snippets/skeletons/cpp/main_helloworld.cpp. 
In the same way, I try to execute the program which is under esp32-snippets/cpp_utils/tests/BLETests/sampleclient.cpp. The problem is  I am getting error as
" BLEDevice.h no such file or directory".
My question is it the same way to build this program as mentioned in tutorial: Building a C++ Application or any other methods I should follow?
Thank you.

nkolban, 2017-10-25:
In ESP-IDF there is the concept of "components".  A component (I might call it a library) is a set of source files that become available to your application.

The cpp_utils is an example of such a component.

Visit this web page:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils

and scroll down to the bottom where you will find the documentation.  There is a section called:

Compiling the C++ classes ... here is a direct link:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils#compiling-the-c-classes

Have a read at that and see if it helps.  Please let me know how you get on.

vishnunaik, 2017-10-26:
Dear sir,
Thank you for your replay and valuable time.
I followed the way which as mentioned in the link above. Created the components directory and compiled the program. 
The header files are compiled but some other errors are displayed I attached printscreen of my Eclipse window , which shows some errors and added components directory, sampleclient.cpp program.
please help with this
Attachment file:
[image]

Sir I wat buy your Kolban's book on ESP32 is it a hard copy or it can be available as soft copy also?
Thank you.

nkolban, 2017-10-27:
Howdy,
I'm sorry there is a problem.  We'll get to the bottom of it.  I'm around evenings and weekends.  If you need live support, ping me on IRC in chat room `#ESP32` as user kolban.  If needed, we can set up a skype based screen share.  I'm sure when we get to the bottom of it, it will be quite simple.

Two immediate questions, both equally important.

1) Can you get me a listing of what you get as output when you execute:
```
$ cd /home/geokno/esp/ESP-IDF/BLEClient
$ ls -lR
```

2) How are you running the build?
The reason I ask is that I have this suspicion that you are asking "Eclipse" to perform the compilation rather than using the "ESP-IDF" build system.  When I run a build in Eclipse, this is what I see:

[image]

All my thinking is based around the notion we will be running the ESP-IDF build system by running make as a process against the ESP-IDF supplied make files.

chegewara, 2017-10-28:
I just created ne ble example project from template, as usual. Then i copied BLEServerSample and main to it and when i tried to compie it under eclipse ive got same error. I have eclipse prepared.
Now weird thing. When i compiled it from cli and then i back to eclipse then error gone.

vishnunaik, 2017-10-28:
Dear @nkolban Sir, 

1. As you asked the listing of above-executed code it is attached in .docx format please find below
[List of files home_geokno_esp_ESP-IDF_BLEClient.docx](https://github.com/nkolban/esp32-snippets/files/1423991/List.of.files.home_geokno_esp_ESP-IDF_BLEClient.docx)

2. I am asking eclipse to compile the code by creating Build-Targets like: all, flash. when run all command to build the code I got those errors which were mentioned in my previous post.

Sir, please mention the date and time when we can connect through Skype according to your free time. I will set my timings as per your time zone.
thank you.

vishnunaik, 2017-10-28:
Dear @chegewara, 
how you executed the commands in cli please mention I will also try that method. 
What are those commands and how you build the program which method you followed?
Thank you.

chegewara, 2017-10-28:
First you have to follow those instructions to setup linux cli:
http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html
http://esp-idf.readthedocs.io/en/latest/get-started/add-idf_path-to-profile.html#add-idf-path-to-profile-linux-macos

but im guessing you have this done, since before you can compile any project with eclipse you should use: `make menuconfig ` to prepare sdkconfig. Then you can easy compile in terminal with:

- make
- make all
- make flash

All is described here
http://esp-idf.readthedocs.io/en/latest/get-started/index.html#build-and-flash

nkolban, 2017-10-28:
To correctly compile a program using Eclipse, it has always been my experience that one should use the ESP-IDF supplied "build system".  This means that you should be able to go to the root directory of your project in a Linux command line and simply type "make".  This should build all.  If this fails, then we won't succeed from invoking make within Eclipse.  Once you can compile from the command line, the next step will be to tell your project that you wish Eclipse to drive the external make system.

Here is what my builder settings look like in Eclipse:

[image]

vishnunaik, 2017-10-30:
Dear @nkolban sir,
I tried ESP-IDF build system. In the command window, I typed "make" in the root directory of my project it started to build the program after a while it given the errors as same as in the eclipse. 
[image]
Errors:
BLEUUID does not name a type
BLEAddress was not  declared in the scope 
pAddrress was not decladred in the scope. etc....

My builder settings in Eclipse as shown in the below figure:
[image]

chegewara, 2017-10-30:
@vishnunaik  One question, when you add esp32_snippets file to project, do you include files in componets folder or in some subfolder, ie. components/cpp_utils? Because in esp-idf v2.1 components could be placed directly in components folder, but now its required to place all files in subfolders. Also check please if there is component.mk file in this subfolder.

vishnunaik, 2017-10-30:
Dear @chegewara our includes are in subfolder i.e components/cpp_utils only. component.file is exist in this subfolder also.
I am writing here how I executed the code [ref:](https://www.youtube.com/watch?v=-ttiPfmrehU&index=8&list=PLB-czhEQLJbWMOl7Ew4QW1LpoUUE7QMOo)
1. We clone the esp-idf-template file to a project folder with a name of BLEClient.
2. Then we created a components directory and copied the cpp_utils file in side this directory.
3. Next we opened a command line in project directory and typed makemenuconfig to create build foldrer in the projectdirectory.
4. Then we imported the BLEClient to Eclipse.
5. In subfolder of the project directory there exist "main" under this we copied sampleclient.cpp program.
6. Next we attached includes files from esp32-snippets/eclipse/C_includes.xml files for looking header files.
7. Next we did some modification in C/C++ build, C/C++ General properties folder of BLEClient in Eclipse.
8. Then we created build-targets i.e: make,flash.
9. Run the make in eclipse then we got those above errors. 
10. Then we run "make" command in the command window in projectdirectory folder we got the same errors.
If any mistakes in above steps please mention.
thank you

chegewara, 2017-10-30:
@vishnunaik Everything looks fine, except 1 thing. I think i just recreated this issue, if not then sorry and will try again. Run menuconfig and turn on Bluetooth if its not turned on

nkolban, 2017-10-31:
@vishnunaik 
This one's a puzzle.  It seems everything you are doing is correct.  Would you be able to make some time to do a screen share with me by skype?  My id is "neil.kolban".  If so, we can share your screen and I can look at your environment.  I'm not at all seeing anything wrong from the stories youve written.

I live in US central time and work from 9-5pm but am available my evenings and weekends for ESP32 assistance.

vishnunaik, 2017-11-02:
Dear sir can we have skype screen share on 4th November at 11 am of your time? 
Thank you.

nkolban, 2017-11-02:
Perfect ... I will be at my desk at 11:00am on Saturday November 4th.  My time zone is US Central time

See also:
https://www.worldtimebuddy.com/

My timezone is also known as CDT

Check that you can ping me at "neil.kolban" on skype anytime.

vishnunaik, 2017-11-03:
Thank you sir for your time.

nkolban, 2017-11-04:
I believe you resolved the issue and as such, at this time, there is no need for me to be on the call?  Is that correct?

vishnunaik, 2017-11-04:
No no sir problem is not resolved, we can have skype screen share at the same time as 4th, November at 11 am of your time(CDT). If you are busy we can arrange this for some other time. 
As I told thank you in the previous comment for adjusting your time for this skype call.
Thank you.

chegewara, 2017-11-04:
Im lost, its still the issue that is on screenshot?

vishnunaik, 2017-11-04:
Dear @nkolban sir, I send you message regrading our skype call on your account "neil.kolban" please check it. If you are available please send message or If you are busy we can have the skype call other time.
Thank you sir.

nkolban, 2017-11-04:
Here is a summary of the problem and its resolution.

@vishnunaik and I had a Skype call that lasted about an hour.  By eyeballing, I couldn't see the cause of the problem.  As we dug deeper we created a brand new template project, brought in the cpp_utils and still the problem persisted.  The solution was easy in the end.  We had missed enabling Blue Tooth in the make menuconfig here:

[image]

Once done, the problems completely went away.  This of course is a requirement ...

If we look on page 1 of the documentation found here:

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

We see the instruction being present.

I think the problem is solved but will let @vishnunaik close out.

@vishnunaik ... if you have new problems, create new issues ... we like to keep one problem per issue.

chegewara, 2017-11-04:
You should have read thorough:
https://github.com/nkolban/esp32-snippets/issues/139#issuecomment-340372857

vishnunaik, 2017-11-06:
Thank you for your support @nkolban and @chegewara .

closed 2017-11-06 by vishnunaik