nkolban/esp32-snippets #141: WiFi+BLE together not working

akshar001, 2017-10-25:
Hello,
This is an issue that i found while i have tried that new ble arduino example in esp32.
Every example i have tried to work with.
What happens is when i try to connect wifi.begin after or before bluetooth connection has established.
it goes into reboot loop.
Would you just let me know that it is my side of problem. And if it how can i solve it or it's some esp error that needed some research.

nkolban, 2017-10-25:
Howdy my friend.  Unfortunately, this is a known problem ... see #110.  It appears that we are running out of available RAM when trying to run both BLE and WiFi together at the same time.  We are studying ways that we can reduce RAM usage in our libraries.  If the match is close enough for you, can I ask that you close out this issue and we'll track through #110?

closed 2017-10-26 by akshar001

hanadjerbi1, 2018-08-14:
https://github.com/espressif/arduino-esp32/issues/1756