nkolban/esp32-snippets #143: WiFi.cpp:  calling connectAP after wifi scan causes failure and reboot

rweller59, 2017-10-26:
When I scan for wifi ap's, no problem. When I connect to an ap after the scan, I error out and esp32 reboots. APconnect works find by itself.  This is the error received right after the connectAP call

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x40132d51
0x40132d51: _ZN4WiFi9connectAPERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESa
IcEEES7_b at C:...components/cpp_utils/WiFi.cpp:168 (discriminator 2)

void newScan(void) {
	auto results = ptr_wifi->scan();
	for (auto it = results.begin(); it != results.end(); ++it) {
		printf("%s\n", it->toString().c_str());
	}
	ESP_LOGD(TAG, "wifi scan complete");
}

void newConnect(void) {
	ptr_wifi->connectAP(ssid, password, true);
	ESP_LOGD(TAG, "wifi connect complete");
}

void newInit(void) {
	ptr_wifi = new WiFi;

	wifiEventHandler *eventHandler = new wifiEventHandler();
	ptr_wifi->setWifiEventHandler(eventHandler);

	newScan();
	newConnect();
...

nkolban, 2017-10-28:
Yikes.  Recreated and THANK YOU for reporting.  I now need to dig deep and see what's going on.  This is now my weekend project.  Should have news shortly.

nkolban, 2017-10-28:
Thankfully the fix was relatively easy to find and make.  New code has been uploaded and tested and I hope that now all will be well.  Please test at your end and let me know what you find.  Apologies for the problem in the first place.

rweller59, 2017-10-28:
Should I upload new version of all cpp_utils? I've made various changes to a number of files and not really organized at the moment to check all of my changes.

nkolban, 2017-10-28:
Do you mean download as opposed to upload in your previous post?

If all you need are the minimum changes, then wifi.cpp should be sufficient to download.

rweller59, 2017-10-28:
That seemed to do the trick. I appreciate you getting it fixed so quickly Neil.

Was the fix anything interesting and worth explaining? If not, thanks again.

rweller59, 2017-10-29:
Yes, I've been a bit unclear on how to use your wifieventhandler... Since you are creating in wifi.cpp, if I define a new wifieventhandler in my code, don't I end up with the same issue that you stated above?

nkolban, 2017-10-29:
Ouch ... your comments caused me to re-read and review all the code related to event handling and I did not like what I found.  It appears that I had two algorithms.  The first was the original algorithm where I had a member variable of WiFi that that was an instance of a WiFiEventHandler.  We registered that class against the ESP-IDF event handler and then used that for ever more.  I then realize that users might want to change the event handler after WiFi had been started, so I added an indirection ... the ESP-IDF called an event handler function that was constant which then looked up the event handler to call and then called it.  However, I was a dummy ... and only changed the code in 1/2 the locations.  That meant that some were using the original algorithm and some using the new algorithm.  I have now walked the code and believe I have corrected all.

The current story should be
1) When we start as an Access Point OR a Station OR Perform a wifi scan, then we register the static event handler IF it has not already been registered.  Once registered, that static event handler stays fixed for ever more.  When an event occurs, the static event handler now looks for the last registered custom/user event handler and invokes it.

rweller59, 2017-10-29:
I'm a bit unclear... I create an event handler in my function that essentially is called from the static handler? How would I define that in my function?

On another note, are you able to use c++11 libraries with the ESPIDF?

nkolban, 2017-10-29:
If you want to handle events yourself (and remember, the primary goal of the C++ classes is to make life simpler for you) then you can.  There is a class called "WiFiEventHandler".  It contains methods called:

* virtual esp_err_t apStaConnected()
* virtual esp_err_t apStaDisconnected()
* virtual esp_err_t apStart()
* virtual esp_err_t apStop()
* virtual esp_err_t staConnected()
* virtual esp_err_t staDisconnected()
* virtual esp_err_t staGotIp(system_event_sta_got_ip_t event_sta_got_ip)
* virtual esp_err_t staStart()
* virtual esp_err_t staStop()
* virtual esp_err_t wifiReady()

These are virtual methods meaning that default implementations already exist.  This means that you can over-ride just the ones you care about.  For example:

```
class MyHandler: public WiFiEventHandler {
   virtual esp_err_t apStop() {
      // Called here when the access point stops
      return ESP_OK;
   }
}

...
myWifi.setWifiEventHandler(new MyHandler());
```

I have been using the standard C++11 libraries in ESP-IDF (so far) without problem.

rweller59, 2017-10-29:
I was planning on using the wifiEventHandler, just wasnted quite sure how it played with the static event handler you just updated.

I've been having a few issues related to c++11 functions not being resolved - wasn't sure if that was my Eclipse set up but apparently a quite common issue. Wanted to make sure ESPIDF was ok with it before I spent too much time chasing down. thakns

chegewara, 2017-10-29:
Im not 100% sure but try to change options in `menuconfig->compiler options->optimization level->release` and turn on `enable C++ exceptions` to workaround issues with c++11

rweller59, 2017-10-29:
I don't see any C++ exceptions options. Seems like its an Eclipse issue - Eclipse basically doesn't recognize the library functions. There is a lot of info regarding issues with Eclipse and C++11. I've tried some without luck but since the ESP-IDF seems to be ok with 11, I'll keep digging.

nkolban, 2017-10-29:
Howdy,
I'm more than happy to try and assist.  My goal is that if a code user like yourself has troubles getting a task done then the onus is on me to document it better.  Since I also try and write a book on ESP32 programming, if it isn't clear there then I also have work to do.  Let me encourage you to keep posting questions ... for my part, I'll try and update my writings after research an answer.

So lets take everything at one puzzle at a time ... don't say "nothing is working" but instead take the time document what you have tried, what your expected outcome or desired outcome was and what is actually being found.  Somewhere in that mix we'll discover whether my documentation is wrong or whether I am simply not explaining it well enough and hence improve.

chegewara, 2017-10-29:
@rweller59 this is Compiler options menu with options setup like mentioned
[image]

rweller59, 2017-10-30:
I've been to the compiler options screen but only have the first 2 options showing.

Just to be clear, the issue I was referring to with c++11 occurred to me because I'm getting this error on this function: 'std::string to_string() cannot be resolved'. I thought perhaps this was related to c++11 version but now I'm thinking probably not since 11 has been out for so long and I assume that I'm using other functions in 11 and snippets is as well and I am not having any other compile issues.

So, while I wanted to use the to_string function it's easy to work around it so I made it a pretty low priority on my list of things to figure out.

rweller59, 2017-10-30:
My comment 'nothing is working' only refers to my attempt to resolve the to_string() function issue. Everything else is fine

chegewara, 2017-10-30:
If you dont see the last option, about C++ exceptions, then you dont use latest version of esp-idf, most likely v2.1 or some older pull of version 3.

rweller59, 2017-10-30:
My version (3) I got on Oct. 17. I'll get latest and let you know if menuconfig changes

rweller59, 2017-10-30:
new version and the c++ exceptions option is now available. Looks like some other new options to keep me busy as well. thanks

closed 2017-11-08 by rweller59