nkolban/esp32-snippets #144: BLE client fails to get serviceUUID

weizenbock, 2017-10-27:
First of all, thanks for sharing the library and all the contribution you made at arduino-esp32.

I'm testing BLE_server and BLE_client from your ESP32_BLE_Arduino repo with two of my ESP32 devices, one as a client, another as a server.

I copied the service uuid and characteristic uuid written on the server example and replaced the ones on the client example with them, so that the client can find the server.

On my serial monitor, I could fine a line
> BLE Advertised Device found: Name: MyESP32, Address: ...  

which would apparently came from
https://github.com/nkolban/ESP32_BLE_Arduino/blob/cb1ab4ea76af17d347cdadcc003d1fe53af12aa7/examples/BLE_client/BLE_client.ino#L74-L75

but doesn't seem to have reached at
https://github.com/nkolban/ESP32_BLE_Arduino/blob/cb1ab4ea76af17d347cdadcc003d1fe53af12aa7/examples/BLE_client/BLE_client.ino#L81

I looks like the if statement right before that returned false.

So I added some lines like these
> Serial.print("haveServiceUUID: "); Serial.println(advertisedDevice.haveServiceUUID());
> Serial.print("device's UUID: "); Serial.println(advertisedDevice.getServiceUUID().toString().c_str());

to somewhere around
https://github.com/nkolban/ESP32_BLE_Arduino/blob/cb1ab4ea76af17d347cdadcc003d1fe53af12aa7/examples/BLE_client/BLE_client.ino#L76

and this is what I got (I have replaced the <MAC_ADDRESS> part, it was correctly shown on the serial monitor)
> BLE Advertised Device found: Name: MyESP32, Address: <MAC_ADDRESS>, txPower: -21
> haveServiceUUID: 0
> device's UUID: \<NULL\>

I don't have much experience in BLE environment.  Maybe failing to get a service uuid is not a rare thing to happen in the first place.

Could you tell me if those two examples are still working at your environment and if so, is there anything I can try to get this working?

nkolban, 2017-10-27:
Hello my friend.  I'll be delighted to work with you to get these going.  There is lots we can try and do.  First of all there has been a lot of churn in this area recently so I want to be sure that we are all using the same code base.  The owners of the Arduino-ESP32 library now include the BLE-C++ code with their distribution.  This means that we should be exclusively using the distribution that comes with Arduino-ESP32 library.

I too have a box of ESP32 to play with.  Can I ask you to "pastebin" the two scripts (one that will be running on ESP32 number 1 and the other running on ESP32 number 2).  Add the links to the pastebin to this issue.  When I get them, I'll grab them and run up two ESP32s to try and recreate the same issue and then I'll be 100% on the same page as you.  I'll be around this evening and all weekend to work on it.  We'll use your two scripts as the common place against which we can talk.

If you could also pastebin the serial logs exactly as you see them from both ESP32s that would also be useful.

chegewara, 2017-10-28:
Im trying to run my old test BLEServer sample programs and they dont have any characteristics advertised. Something got changed lately in BLE stack?

nkolban, 2017-10-28:
@chegewara In ESP-IDF or in Arduino?

chegewara, 2017-10-28:
Plain esp-idf. I know we are working in this issue in Arduino, but if i can make BLEServer working with BLEClient under esp-idf then i can move to Arduino IDE to ckeck if it works. But now BLEServer dont advertise any characteristics under esp-idf.

chegewara, 2017-10-28:
Hi @weizenbock, with @nkolban coop ive been able to run SamleServer and SampleClient under esp-idf. Now, can you edit sdkconfig file and change those lines to match this:
```
CONFIG_OPTIMIZATION_LEVEL_DEBUG=
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y

```

nkolban, 2017-10-28:
See also problem #109 

I think I'm going to consider this issue  a duplicate of the primary (#109).  As such, tracking the resolution to #109 to conclusion.  Please post new findings on this issue in #109 as I think they will apply there too.  There is also a lot more detail on what we tried and where we are there.

weizenbock, 2017-10-28:
Thanks for the help.

I wasn't sure if you also wanted me to move to #109.  My issue looks different than that of @chegewara.  So I'll just keep using this thread.  Let me know if you want otherwise.

I actually can see the service uuid on my Android app (called BLE Scanner).
So the server looks working ok.

Here are the codes I'm using.

BLE_server
https://pastebin.com/3Cak8jsv

BLE_client
https://pastebin.com/2Dxr56Xz

Serial log
https://pastebin.com/qeTjuGwy

chegewara, 2017-10-28:
Your SampleServer is not advertising any services hence client cant find one. You have to add 
`pAdvertising->addServiceUUID(pService->getUUID());` after line 31 and before line 32

chegewara, 2017-10-28:
@weizenbock Did you shorten server name? Check with android app if you can see any advertised service. And it has to be advertised before you connect to server, not service seen after connecting

weizenbock, 2017-10-28:
I'm not sure what you meant by "shortening server name" but I haven't changed anything other than the line you gave me since I pasted it on the pastebin.

If you are talking about this line,
> BLEDevice::init("MyESP32");

I can find "MyESP32" on the list of my Android app and its "connect" button gets enabled.
However I cannot find any service uuid on the list.  I can only see it after I hit "connect" button.

chegewara, 2017-10-28:
Try with BLEDevice::init("MyESP"); or BLEDevice::init("ESP32");

chegewara, 2017-10-28:
You have to see advertised service before conecting to esp32 like on picture
[image]

weizenbock, 2017-10-28:
Yes that worked!!! @chegewara 
So is there a limit for the length of server name??

chegewara, 2017-10-28:
Yes, there is limit. Because you are trying to find and connect to a device with custom UUID, which is 128 bit long and has to be advertised, and bluetooth limitation to send advertising packet is 31 bytes the name is limited to about 5 characters. That depends if you are advertising more info or not.

nkolban, 2017-10-28:
Good call mr Weizenbock ... it turns out that your problem wasn't the same as #109 so your choice to keep posting here was 110% correct.

nkolban, 2017-10-28:
Love and Kudos to mr @chegewara  ... you are awesome!

chegewara, 2017-10-28:
Its not my "fault", you have made very good logging, easy to debug.

weizenbock, 2017-10-29:
Oh I see.  If I chose to use custom uuid then it will take so much space in the packet.  That makes sense.
Now I close the issue.  Thank you all!

closed 2017-10-29 by weizenbock

SindhujaNarra, 2018-02-06:
Hello @nkolban, 

I had faced the same issue as @weizenbock  when I was testing BLE_server and BLE_client in Arduino IDE with 2 ESP32 DevKitC devices (one as the server and another as the client). It worked only after I made changes to the BLE_server code as suggested by @chegewara i.e., to reduce the length of device name to 3 (5 characters device name didn't work for me) and adding "pAdvertising->addServiceUUID(pService->getUUID());" to the code. 

I was wondering if you could please update the code for BLE_server in ESP32_BLE_Arduino repo so that in future no one else faces this same issue.

Thanks

niyaziatilay, 2019-02-23:
You have created examples for the Arduino IDE. Your BLEClient instance does not connect to your BLEServer instance. Try with any esp32. Codes are not modified. So the one-to-one code in the samples.

First, an Esp32 is loaded with BLEServer.
Then another Esp32 was loaded with BLEClient.

The advertisedDevice.haveServiceUUID () function in the BLEClient instance returns false. However, a Service ID was assigned to the BLEServer instance. But BLEClient doesn't see it.

Have you reached a solution?

niyaziatilay, 2019-02-23:
Hello @nkolban 

You have created examples for the Arduino IDE. Your BLEClient instance does not connect to your BLEServer instance. Try with any esp32. Codes are not modified. So the one-to-one code in the samples.

First, an Esp32 is loaded with BLEServer.
Then another Esp32 was loaded with BLEClient.

The advertisedDevice.haveServiceUUID () function in the BLEClient instance returns false. However, a Service ID was assigned to the BLEServer instance. But BLEClient doesn't see it.

Have you reached a solution?

Heru-MS, 2019-03-06:
Hello @nkolban,

I'm just starting with ESP32 and BLE on Arduino-IDE, using your BLE-Client example to connect to an iSteady pro gimbal. Unfortuanatly the gimbals name is "Gimbal_14", and I run into the problems shown above. Is there any hope that there will be a solution?

Avtopolet, 2022-01-17:
Have same problem with lenght of server name. Im correct understand - there is no solution for this lib?