nkolban/esp32-snippets #146: Compilation error for all example sketches!

tusharvb19, 2017-10-28:
I am getting following error(sic).I downloaded the latest branch.
Need help.

In file included from C:\Users\Robo1\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEDevice.h:19:0,

                 from C:\Users\Robo1\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\examples\BLE_uart\BLE_uart.ino:22:

C:\Users\Robo1\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEUtils.h:49:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

                                                 ^

nkolban, 2017-10-28:
It is likely that what we have is a mis-match between the version of Arduino-ESP32 you have installed and the version expected by the BLE libraries.  Recently, the author of the Arduino-ESP32 package included the BLE classes as part of the distribution.  As such, my suggestion is to re-install the very latest Arduino-ESP32 package and remove the ESP32_BLE_Arduino-master library.  You should then find that things will work again and you will have BLE available without having to perform any additional steps.

databeredning, 2017-11-01:
I have installed a fresh copy of Atom and added PlatformIO and after adding your lib 0.4.6 I try to compile and get the same error as mentioned above. I'm not quite sure what you mean by re-install latest Arduino-ESP32 package. Please advise. Thank you for your work with documentation and your contribution to esp32 community.

nkolban, 2017-11-01:
The high level story is that on the ESP32 there is an underlying library called ESP-IDF (think of this loosely as the operating system that runs on the ESP32).  Espressif makes available releases and a "best so far".  Releases happen only a couple of times a year but the best so far changes very often.  Last month, a VERY major change was made to the ESP-IDF that dramatically changed how it worked and how to program against it.  Not only did the ESP-IDF change but so too did the "toolchain" (the tools) needed to build ESP32 applications.  Normally this wouldn't be too big a deal as these changes are made in "best so far" and not in a release (the current of which is 2.1).  However, the designers/planners/owners of the Arduino ESP32 library (i.e. Arduino running on ESP32) decided to rebase the Arduino ESP32 distribution on this very latest and greatest ESP-IDF release.  Since the changes made to ESP-IDF were dramatic, I decided I could either support the last ESP-IDF release (v2.1) or the latest and greatest (pre v3.0 and the one now used by Arduino).  I chose the latter.

This means that the BLE libraries will ONLY run under the very latest distribution of the Arduino ESP32 libraries found here:

https://github.com/espressif/arduino-esp32

Now ... this applies to you ... because you are using ESP32 and the Arduino libraries through a distribution that is NOT this:

https://github.com/espressif/arduino-esp32

Instead, we need to ask Platform IO "What release of the Arduino ESP32 libraries are you using?"

In addition, the owners of Arduino ESP32 now include the BLE libraries as part of that distribution ... so you should not need to download anything special to use BLE.

Please let me know if this makes sense and PLEASE question any words or phrases I used here to help me make it a better explanation for others.

databeredning, 2017-11-01:
Many thanks for your rapid answer! I'm familiar with esp-idf and xtensa toolchain as this is the way I've been developing for esp32 so far. I'm also aware of the newly released esp-idf as I have it installed in my developing environment and notice a lot of changes. The actual problem I have is to be able to implement a custom profile with 128bit UUID services ( 4 parallell services ). The low level ESP-IDF functions are as you mention in the documentation difficult to use so I was hoping to short cut by using Arduino environment and therefor I picked PlatformIO. I will try to setup an environment with known working versions of each part. Again, thank you!

databeredning, 2017-11-01:
Works like a charm with Arduino IDE 1.8.5 and https://github.com/espressif/arduino-esp32 (60b8b47).

chegewara, 2017-11-01:
> The actual problem I have is to be able to implement a custom profile with 128bit UUID services ( 4 parallell services ).

@databeredning If you wish i can try to help you to resolve this problem with esp-idf, just start new issue and we can try to work with it.

databeredning, 2017-11-01:
@chegewara I might do that. Thanks in advance.

HudsonWerks, 2017-11-09:
> I still have troubles compiling the example sketches. Same error as OP.
> I'v reinstalled the Arduino core for ESP32 as described on
> https://github.com/espressif/arduino-esp32
> I'v removed the old ESP32_BLE_Arduino-master library and relaunched
> Arduino IDE, but still not able to compile. Any suggestions?
> Should the old core for ESP32 have been removed before reinstalling and
> if, how?

I am having the same exact problem.

chegewara, 2017-11-10:
Here is my BLE library in zip file which is working. If you would like to try it just delete old and install new from arduino. Inside this zip file is also sdkconfig, its in case if replacing library just wont help.

https://ns1.diskstation.eu/fbsharing/2mNDdOwC

HudsonWerks, 2017-11-15:
@chegewara Thanks for the link. I may have waited too long to d/l, so it's no longer available at that URL.

chegewara, 2017-11-15:
https://ns1.diskstation.eu/fbsharing/xnBfKn5S
Try this one

chegewara, 2017-11-15:
@HudsonWerks Sorry, i did not check links. I can send you in mail if you wish

HudsonWerks, 2017-11-15:
Hmm. Still the same problem: "The page is not found."

Sorry for the trouble.

chegewara, 2017-11-15:
Again my bad, sorry. But this will works: (checked)

https://dsm.chege.eu.org/fbsharing/sGPYbpjq

HudsonWerks, 2017-11-15:
@chegewara Fabulous! Your build works! Would you mind sharing what the basic differences are between your builds and Kolban's? Thanks a million.

chegewara, 2017-11-15:
None, im trying to make PR but i did some freaking mess in my local files and i have some damn conflicts and cant push to my own repo. I will work hard to fix it and make PR asap

hgpt, 2017-11-15:
@chegewara Thank you! The library you've provided in the archive works like a charm!
I've tried everything with the official one posted by @nkolban (the one that's over here: https://github.com/nkolban/ESP32_BLE_Arduino):
- reinstalling arduino IDE
- reinstalling last build of official ESP32 Arduino Core
It didn't work! I was getting a huge list of errors everytime I was trying to compile any example.
What did you change regarding your library?

chegewara, 2017-11-15:
Like i said, its nothing more than this repository prepared to install under arduino ide exactly like in this instruction https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

abhayghatpande, 2017-11-15:
@nkolban sorry for the newbie question ... but how does one safely remove the ESP32_BLE_Arduino-master library? Thanks.

chegewara, 2017-11-15:
Find it in Arduino/libraries folder and just delete it.

abhayghatpande, 2017-11-15:
oh ok. so it's redundant if one uses nkolban's library?

chegewara, 2017-11-15:
Yes. Those are the same libraries but this one is more often updated and the other one is available from arduino library manager. My suggestion is to use this one, more effort to update it but worth it.

abhayghatpande, 2017-11-15:
ok thanks, will do!

abhayghatpande, 2017-11-16:
Sorry, but how can we change the platform code being used by PlatformIO?

nkolban, 2017-11-18:
Its a good question ... the story is as follows.  Originally, we wanted a BLE C++ library for the ESP32 and this repository was created and augmented for it.  Then a chap found that the library ran great under Arduino so we packaged a distribution of the library that would have been the user's responsibility to build and install.  Then we found that we could place it in its own Github repository and it would be available in the Arduino management libraries.  Finally, the author of the Arduino library for ESP32 decided to bundle the code in the core distribution.  In the mean time, code development on the ESP-IDF BLE continues.

Platform IO was not even considered and I have no idea how they package libraries.

If it were me ... and knowing what I know ... if I wanted to use the latest code, I would remove all the BLE code and grab the code from this repository and build a BLE library manually.  I'd review the following:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

abhayghatpande, 2017-11-20:
@nkolban thank you for the explanation. I will use the library as advised by you. For now, it might be best to lay off PlatformIO because they seem to package the ESP platform in their own way that does not include the latest updates to ESP-IDF.

RudyFiero, 2017-11-21:
I'm getting the same error "error: 'esp_gattc_service_elem_t' has not been declared"

I am using Arduino 1.8.5 and I used git to update to the version of ESP32-Arduino from 9 hours ago. 

In sdkconfig I have the following

```
#
# Debug Log Configuration
#
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE is not set
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG is not set
# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE is not set
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
# CONFIG_ARDUHAL_LOG_COLORS is not set
# CONFIG_AUTOCONNECT_WIFI is not set
# CONFIG_AWS_IOT_SDK is not set
CONFIG_BT_ENABLED=y
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BTC_TASK_STACK_SIZE=8192
# CONFIG_BLUEDROID_MEM_DEBUG is not set
CONFIG_CLASSIC_BT_ENABLED=y
CONFIG_GATTS_ENABLE=y
CONFIG_GATTC_ENABLE=y
CONFIG_BLE_SMP_ENABLE=y
# CONFIG_BT_STACK_NO_LOG is not set
CONFIG_BT_ACL_CONNECTIONS=4
CONFIG_BTDM_CONTROLLER_RUN_CPU=0
CONFIG_SMP_ENABLE=y
CONFIG_BT_RESERVE_DRAM=0x10000
```
Anyone know what I should do?

chegewara, 2017-11-21:
@RudyFiero Try this, it will help you to solve issue
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

RudyFiero, 2017-11-22:
1. Go to <ArduinoIDE>/hardware/espressif/esp32/libraries
2. Delete the directory called BLE

There was no directory called BLE. There is one called SimpleBLE. I deleted it and it made no difference.

3. Go to your <Arduino>/libraries folder
4. Extract the ESP32_BLE.zip file there

Did that. Started ArduinoIDE by opening BLE_uart, compiled and got this error.

```
C:\Users\wrpa\Documents\Arduino\libraries\ESP32_BLE\src/BLEUtils.h:49:49: error: 'esp_gattc_service_elem_t' has not been declared

static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);
```
Do I have the proper Arduino-ESP32 code? How would I know? I had followed the instructions on updating through git, didn't seem to have a problem, but I'm not very familiar with git yet.

chegewara, 2017-11-22:
Please check this folder: Arduino/libraries and delete ESP32_BLE

chegewara, 2017-11-22:
Also what is the output when you run this command:
`xtensa-esp32-elf-gcc -v`

RudyFiero, 2017-11-22:
After I deleted - Arduino/libraries and delete ESP32_BLE - I get
fatal error: BLEDevice.h: No such file or directory

> Also what is the output when you run this command:
> xtensa-esp32-elf-gcc -v

Windows cannot find 'xtensa-esp32-elf-gcc -v'. Make sure you typed the name correctly, and then try again.

I ran that from Run  window

chegewara, 2017-11-22:
Thats ok. Now when you deleted proper folder, you can install library from zip.
Do you have ESP32_BLE.zip file created from the latest esp32-snippets repository or would you like me to attach one here?

RudyFiero, 2017-11-22:
I would download it but I don't see anything under esp-snippets that looks like ESP32_BLE.zip What folder should it be under?

RudyFiero, 2017-11-22:
Or do I put the entire unzipped esp32-snippets into mt library?

chegewara, 2017-11-22:
This is how you can create zip file to use in arduino
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

RudyFiero, 2017-11-22:
That seemed to work. At least I was able to compile BLE_client without error. BLE_uart compiled and loaded, but is looping/rebooting.

@chegewara - I appreciate the help you have given me here.

chegewara, 2017-11-22:
This is ready to install in arduino ide zip file
[ESP32_BLE.zip](https://github.com/nkolban/esp32-snippets/files/1493573/ESP32_BLE.zip)

chegewara, 2017-11-22:
Can you switch debug log to verbose and then post here log output?
`Tools->Core debug level-> verbose`

abhayghatpande, 2017-11-22:
​Have you "installed" the library using the library manager? Sketch ->
Include Library -> Add .zip library menu.​

RudyFiero, 2017-11-22:
I deleted the ESP32_BLE that I had in my Arduino/libraries directory, then unzipped the ESP32_BLE.zip above. Now I am again getting 
error: 'esp_gattc_service_elem_t' has not been declared

I did delete ESP_BLE_Arduino when I replaced the ESP_BLE that was there with the above ESP_BLE.zip. I got the compile error, then replace (un-deleted) ESP_BLE_Arduino but it didn't change things. 

I am assuming that I should only have the one BLE library, ESP_BLE. But I get the same error. 
error: 'esp_gattc_service_elem_t' has not been declared

I'm going to delete it again, download it again, put in library, then try again.

RudyFiero, 2017-11-22:
@abhayghatpande  - Yes I did, at some point. but mostly I have unzipped it in the libraries directory and then started up ArduinoIDE

chegewara, 2017-11-22:
You have to install this library, not unzip
In arduino ide go to ```
Sketch ->
Include Library -> Add .zip library menu
```

RudyFiero, 2017-11-22:
I deleted all BLE libraries in my Arduino/library folder. I deleted the esp32-snippets as well. I then downloaded the ESP32_BLE.zip and copied it to the libraries folder. I then started the ArduinoIDE and did the include zip library of ESP32_BLE.zip.  I then loaded the example BLE_uart.ino through the examples tab. 

I then get this error.

```
In file included from C:\Users\wrpa\Documents\Arduino\libraries\ESP32_BLE\src/BLEDevice.h:20:0,

                 from C:\Users\wrpa\Documents\Arduino\libraries\ESP32_BLE\examples\Arduino\BLE_uart\BLE_uart.ino:22:

C:\Users\wrpa\Documents\Arduino\libraries\ESP32_BLE\src/BLEUtils.h:50:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

```

chegewara, 2017-11-22:
This means you have old version of esp-idf. Old mean is 2.0 or 2.1. 
http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html

abhayghatpande, 2017-11-22:
ESP-IDF needs to be updated most probably.

Abhay Ghatpande, Ph.D., MBA
Product Innovation | Market Strategy
CAN: +1-514-442-7661
USA: +1-650-265-7710

RudyFiero, 2017-11-22:
Yesterday I followed the instructions from 
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code

This doesn't work?

chegewara, 2017-11-22:
It is good instruction. Ive send you to esp-idf instalation, sorry. 
Can you tell us what is version when you do this step:
`Open C:/Users/[YOUR_USER_NAME]/Documents/Arduino/hardware/espressif/esp32/tools and double-click get.exe`
There should be some output similar to this, but with version `1.22.0-73`
https://raw.githubusercontent.com/espressif/arduino-esp32/master/docs/arduino-ide/win-screenshots/win-gui-4.png

RudyFiero, 2017-11-22:
System: Windows, Info: Windows-10-10.0.15063
Platform: i686-mingw32
Tool xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip already downloaded
Extracting xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip

RudyFiero, 2017-11-22:
3. Wait for git to pull any changes and close Git GUI

```
On the Fetching new changes from origin

Success
```
Does "Success" mean it is done, or just found something and is currently downloading?

RudyFiero, 2017-11-22:
I'm going through the instructions again and I am now at 

```
Next Steps
To carry on with development environment setup, proceed to section Get ESP-IDF.

http://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-started-get-esp-idf
```
Could this be my problem? It doesn't look familiar so I probably missed it. Or maybe I blocked it out of my memory as it makes me nervous. I really need to understand how git works. :(

nkolban, 2017-11-22:
I can attest that if your version of xtensa-esp32-elf-gcc -v is not 1.22.0.73 then you are back level.

The error messages you are getting are solidly consistent with you running an old version of the Arduino-ESP32 libraries and toolchain.

RudyFiero, 2017-11-22:
I kind of figured that. But how do I get the current version? I thought that git should have taken care of that. But as I have said before, I know little about git and how to use it. I just followed the update instructions provided.

nkolban, 2017-11-22:
I suspect that if you follow exactly all the instructions documented here it will work:

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

Maybe start from a clean slate (i.e. un-install the Arduino IDE and begin again).  For example, did you run "get.exe" as documented in the last step 4?

Maybe if we create a document describing all the steps that you followed and screen shots from each we could spot a potential error?

RudyFiero, 2017-11-22:
Yes I did run get.exe. About six times. That was how I got this.

```
System: Windows, Info: Windows-10-10.0.15063
 Platform: i686-mingw32
 Tool xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip already downloaded
 Extracting xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip
```
The instructions are simple and easy to follow. The screen shots in the updated process are exactly what I got. I went through the steps on Monday and again yesterday. With the same result.

When I get home tonight I probably do a reinstall.

abhayghatpande, 2017-11-22:
Clearly, the package pulled into the `dist` directory is incorrect. I don't
have a Windows machine to test unfortunately. Please verify the
`package/package.json` file if it has the _73 version of the toolchain. You
could also use the `get.py` script instead of the exe.

RudyFiero, 2017-11-22:
I will give that a try. Thanks.

RudyFiero, 2017-11-23:
> Please verify the
> `package/package.json` file if it has the _73 version of the toolchain.

In /Documents/Arduino/hardware/espressif/esp32/packages there is only one file called 
package_esp32_index.template.json

```
"tools": [
        {
          "name": "xtensa-esp32-elf-gcc",
          "version": "1.22.0-61-gab8375a-5.2.0",
          "systems": [
            {
              "host": "i686-mingw32",
              "url": "https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip",
              "archiveFileName": "xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip",
              "checksum": "SHA-256:c00dbdab8e8acc273d4aa319a74f7bb1d9496c843159823201d685359174168f",
              "size": "76346909"
            },
```

It does not have _73 version.

abhayghatpande, 2017-11-23:
That's the older version. Try a fresh install in a different directory. That version should be `https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-73-ge28a011-5.2.0.zip` <https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-73-ge28a011-5.2.0.zip%60>.

All the best. Definitely a weird error ...

chegewara, 2017-11-23:
This is possible explanation. You did have installed arduino ide before and older version of esp-idf stack for arduino. When you did delete or reinstall arduino you did not delete old Arduino folder from [USER]/Document and tere still is old version of esp-idf. 

When i first tried to work with arduino i thought i know how to install old version of toolchain, but i forgot i have had installed arduino ide before.

[5 more replies]

closed 2017-12-09 by nkolban