nkolban/esp32-snippets #149: Exception thrown when parsing ill-formed json

rweller59, 2017-10-29:
I had a poorly-formed json and exceptions were thrown. The parse function does not error out but an operation on the json object does, i.e., calling 
**std::string str = jsonobj.getString("name");** throws exception because 
cJSON *node = cJSON_GetObjectItem(m_node, name.c_str()); returns a nullptr and
return std::string(node->valuestring); throws exception because of null pointer

I started to add error checks like this...
std::string JsonObject::getString(std::string name) {
	cJSON *node = cJSON_GetObjectItem(m_node, name.c_str());
	if(node != nullptr) return std::string(node->valuestring);
	return std::string();
} // getString

But thought there might be a better method at the object level to determine valid json

nkolban, 2017-10-29:
Again, thank you.  Another excellent catch.  What we have now done is added a new method to JsonObject which is "isValid()".  This will tell us whether or not the JsonObject we have is usable or if an error was detected.  For example, here is an example:

```
JsonObject myObj = JSON::parseObject(myString);
if (myObj.isValid()) {
   // Safe to use the returned object.
}
```

rweller59, 2017-10-29:
simple enough. thanks

closed 2017-11-08 by rweller59