nkolban/esp32-snippets #150: Compile error

maxpromer, 2017-10-30:
Hello, i update Arduino-ESP32 today (30/10/2017) and i add it librarie to Documents\Arduino\libraries. i run BLE_server sketche and test compile but arduino show error.

```
C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                                              ^

In file included from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                                                                                                        ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:131:118: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

    BLERemoteCharacteristic* pBLERemoteCharacteristic = pBLERemoteService->getCharacteristic(BLEUUID(evtParam->notify.char_id.uuid));

                                                                                                                      ^

In file included from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:133:111: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

     ESP_LOGE(LOG_TAG, "Could not find characteristic with UUID %s for notification", BLEUUID(evtParam->notify.char_id.uuid).toString().c_str());

                                                                                                               ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:196:55: error: no matching function for call to 'BLEUUID::BLEUUID(esp_gatt_id_t&)'

    BLEUUID uuid = BLEUUID(evtParam->search_res.srvc_id);

                                                       ^

In file included from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteCharacteristic.h:18:0,

                 from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:16,

                 from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.h:18,

                 from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note: candidate: BLEUUID::BLEUUID()

  BLEUUID();

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note:   candidate expects 0 arguments, 1 provided

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:25:2: note: candidate: BLEUUID::BLEUUID(esp_gatt_srvc_id_t)

  BLEUUID(esp_gatt_srvc_id_t srcvId);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:25:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:24:2: note: candidate: BLEUUID::BLEUUID(uint8_t*, size_t, bool)

  BLEUUID(uint8_t* pData, size_t size, bool msbFirst);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:24:2: note:   candidate expects 3 arguments, 1 provided

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:23:2: note: candidate: BLEUUID::BLEUUID(esp_bt_uuid_t)

  BLEUUID(esp_bt_uuid_t uuid);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:23:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_bt_uuid_t'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:22:2: note: candidate: BLEUUID::BLEUUID(uint32_t)

  BLEUUID(uint32_t uuid);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:22:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint32_t {aka unsigned int}'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:21:2: note: candidate: BLEUUID::BLEUUID(uint16_t)

  BLEUUID(uint16_t uuid);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:21:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint16_t {aka short unsigned int}'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:20:2: note: candidate: BLEUUID::BLEUUID(std::__cxx11::string)

  BLEUUID(std::string uuid);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:20:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(const BLEUUID&)

 class BLEUUID {

       ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'const BLEUUID&'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(BLEUUID&&)

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'BLEUUID&&'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:197:94: error: no matching function for call to 'BLERemoteService::BLERemoteService(esp_gatt_id_t&, BLEClient*)'

    BLERemoteService* pRemoteService = new BLERemoteService(evtParam->search_res.srvc_id, this);

                                                                                              ^

In file included from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.h:18:0,

                 from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2: note: candidate: BLERemoteService::BLERemoteService(esp_gatt_srvc_id_t, BLEClient*)

  BLERemoteService(esp_gatt_srvc_id_t srvcId, BLEClient* pClient);

  ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note: candidate: BLERemoteService::BLERemoteService(const BLERemoteService&)

 class BLERemoteService {

       ^

C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note:   candidate expects 1 argument, 2 provided

Using library ESP32_BLE_Arduino at version 0.4.2 in folder: C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino 
exit status 1
Error compiling for board WEMOS LOLIN32.
```

how to fix it ?

Thank for reply. :-)

nkolban, 2017-10-31:
Howdy ... in the very latest Arduino ESP32 library, the owner has now packaged the BLE C++ classes as part of that.  As such, you should now no longer need this directory "C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino".

Remove that directory so that the only place you are picking up the BLE C++ classes will be coming from the Arduino ESP32 library which you just refreshed.   Try that and let me know how you get on.

HudsonWerks, 2017-11-08:
Hello,

Great work and libraries. Thank you very much for all your efforts on the ESP32 front. However, I've been trying to get your BTLE sketches working on Arduino IDE 1.8.5 using Sparkfun ESP32 Thing and haven't been successful.  I've installed the latest core libraries and files, as well as removed the dir that you suggest in this thread (C:\Users\CAH\Documents\Arduino\libraries\ESP32_BLE_Arduino). But I still get the following compile errors (e.g. BLE_notify.ino):

Arduino: 1.8.5 (Windows 10), Board: "SparkFun ESP32 Thing, 80MHz, 921600"

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                                              ^

In file included from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                                                                                                        ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:131:118: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

    BLERemoteCharacteristic* pBLERemoteCharacteristic = pBLERemoteService->getCharacteristic(BLEUUID(evtParam->notify.char_id.uuid));

                                                                                                                      ^

In file included from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:133:111: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

     ESP_LOGE(LOG_TAG, "Could not find characteristic with UUID %s for notification", BLEUUID(evtParam->notify.char_id.uuid).toString().c_str());

                                                                                                               ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:196:55: error: no matching function for call to 'BLEUUID::BLEUUID(esp_gatt_id_t&)'

    BLEUUID uuid = BLEUUID(evtParam->search_res.srvc_id);

                                                       ^

In file included from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteCharacteristic.h:18:0,

                 from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:16,

                 from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.h:18,

                 from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note: candidate: BLEUUID::BLEUUID()

  BLEUUID();

  ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLEUUID.h:26:2: note:   candidate expects 0 arguments, 1 provided

[... 67 lines cut ...]

                 from C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:14:

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2: note: candidate: BLERemoteService::BLERemoteService(esp_gatt_srvc_id_t, BLEClient*)

  BLERemoteService(esp_gatt_srvc_id_t srvcId, BLEClient* pClient);

  ^

[... cut ...]
C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note: candidate: BLERemoteService::BLERemoteService(const BLERemoteService&)

 class BLERemoteService {

       ^

C:\Users\CAH\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note:   candidate expects 1 argument, 2 provided

exit status 1
Error compiling for board SparkFun ESP32 Thing.

------
Any suggestions or guidance you could lend would be much appreciated.
Charles

HudsonWerks, 2017-11-08:
Hello,

Great work and libraries. Thank you very much for all your efforts on the ESP32 front. However, I've been trying to get your BTLE sketches working on Arduino IDE 1.8.5 using Sparkfun ESP32 Thing and haven't been successful.  I've installed the latest core libraries and files, as well as removed the dir that you suggest in this thread (C:\Users\NAME\Documents\Arduino\libraries\ESP32_BLE_Arduino). But I still get the following compile errors (e.g. BLE_notify.ino):

Arduino: 1.8.5 (Windows 10), Board: "SparkFun ESP32 Thing, 80MHz, 921600"

C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':

C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                                              ^

In file included from C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                                                                                                        ^

C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                       
C:\Users\NAME\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7: note:   candidate expects 1 argument, 2 provided

exit status 1
Error compiling for board SparkFun ESP32 Thing.

------
Any suggestions or guidance you could lend would be much appreciated.
Charles

nkolban, 2017-11-28:
@HudsonWerks Sorry for the delay.  Just noticed your question distinct from the original poster.  Is this a new issue?  If so, please create a unique Issue and we'll track it from there.

nkolban, 2017-12-14:
Closing for just now.

closed 2017-12-14 by nkolban