nkolban/esp32-snippets #151: Trying to run SampleClient.cpp within ESP-IDF environment

yangantang, 2017-10-30:
Hi,

I'm using the latest master of everything, cpp_utils, esp-idf, and toolchain. When I compile and run the SampleClient example, I hit different errors each time. These are 2 samples I managed to save. In fact on some lucky attempts, I was able to get it to work and change the characteristic value. On other attempts I get Guru meditation errors: LoadProhibited or errors like the snippets I have below.

```Shell
D (1646) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
D (1656) BLEUtils: [status: 0]
D (1656) Task: >> runTask: taskName=Task`
D (1666) BLEClient: >> connect(7e:16:85:d4:ca:7a)
D (1666) FreeRTOS: Semaphore taking: name: RegEvt (0x60090d62), owner: <N/A> for connect
D (1676) FreeRTOS: Semaphore taken:  name: ScanEnd (0x0), owner: start
D (1686) FreeRTOS: Semaphore taken:  name: RegEvt (0x60090d62), owner: connect
E (1686) BT: btc_gattc_call_handler()
D (1696) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_EVT
D (1706) BLEUtils: GATT Event: ESP_GATTC_REG_EVT
D (1706) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]
D (1726) BLEScan: << start()
D (1736) FreeRTOS: Semaphore taking: name: OpenEvt (0x0), owner: <N/A> for connect
D (1746) FreeRTOS: Semaphore taken:  name: OpenEvt (0x0), owner: connect
E (31746) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0100
E (31746) BT: p_conn is NULL in bta_gattc_conn_cback

D (31746) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
D (31756) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (31756) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: 7e:16:85:d4:ca:7a]
D (31766) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_OPEN_EVT
D (31776) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (31776) BLEUtils: [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: 7e:16:85:d4:ca:7a, mtu: 0]
D (31806) BLEClient: << connect(), rc=1
D (31806) BLEClient: >> getService: uuid: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (31806) BLEClient: >> getServices
D (31806) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x0), owner: <N/A> for getServices
D (31826) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x0), owner: getServices
```

```Shell
D (1366) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
D (1376) BLEUtils: [status: 0]
D (1376) Task: >> runTask: taskName=Task
D (1386) BLEClient: >> connect(7e:16:85:d4:ca:7a)
D (1386) FreeRTOS: Semaphore taking: name: RegEvt (0xa52074ca), owner: <N/A> for connect
D (1396) FreeRTOS: Semaphore taken:  name: RegEvt (0xa52074ca), owner: connect
E (1406) BT: btc_gattc_call_handler()
D (1406) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_EVT
D (1416) BLEUtils: GATT Event: ESP_GATTC_REG_EVT
D (1416) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]
D (1426) FreeRTOS: Semaphore taking: name: OpenEvt (0x0), owner: <N/A> for connect
D (1426) FreeRTOS: Semaphore taken:  name: OpenEvt (0x0), owner: connect
D (1446) BLEScan: << start()
D (1696) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CONNECT_EVT
D (1696) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT
D (1696) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: 7e:16:85:d4:ca:7a]
D (1706) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_OPEN_EVT
D (1706) BLEUtils: GATT Event: ESP_GATTC_OPEN_EVT
D (1716) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, remote_bda: 7e:16:85:d4:ca:7a, mtu: 23]
D (1726) BLEClient: << connect(), rc=1
D (1726) BLEClient: >> getService: uuid: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (1736) BLEClient: >> getServices
D (1736) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x58206c65), owner: <N/A> for getServices
D (1746) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x58206c65), owner: getServices
D (2486) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (2486) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (2486) BLEUtils: [conn_id: 0, start_handle: 1 0x01, end_handle: 3 0x03, srvc_id: uuid: 00001801-0000-1000-8000-00805f9b34fb, inst_id: 1
D (2506) BLERemoteService: >> BLERemoteService()
D (2506) BLERemoteService: << BLERemoteService()
D (2506) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (2516) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (2526) BLEUtils: [conn_id: 0, start_handle: 20 0x14, end_handle: 26 0x1a, srvc_id: uuid: 00001800-0000-1000-8000-00805f9b34fb, inst_id: 1
D (2536) BLERemoteService: >> BLERemoteService()
D (2536) BLERemoteService: << BLERemoteService()
D (2546) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (2556) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (2556) BLEUtils: [conn_id: 0, start_handle: 40 0x28, end_handle: 45 0x2d, srvc_id: uuid: 10000000-328e-0fbb-c642-1aa6699bdada, inst_id: 1
D (2576) BLERemoteService: >> BLERemoteService()
D (2576) BLERemoteService: << BLERemoteService()
D (2586) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (2586) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (2596) BLEUtils: [conn_id: 0, start_handle: 46 0x2e, end_handle: 48 0x30, srvc_id: uuid: badbadba-dbad-badb-adba-badbadbadbad, inst_id: 1
D (2606) BLERemoteService: >> BLERemoteService()
D (2616) BLERemoteService: << BLERemoteService()
D (2616) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
D (2626) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT
D (2626) BLEUtils: [conn_id: 0, start_handle: 49 0x31, end_handle: 65535 0xffff, srvc_id: uuid: 91bad492-b950-4226-aa2b-4ede9fa42f59, inst_id: 1
D (2646) BLERemoteService: >> BLERemoteService()
D (2646) BLERemoteService: << BLERemoteService()
D (2656) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_CMPL_EVT
D (2656) BLEUtils: GATT Event: ESP_GATTC_SEARCH_CMPL_EVT
D (2666) BLEUtils: [status: ESP_GATT_OK, conn_id: 0]
D (2676) BLEClient: << getServices
D (2676) BLEClient: << getService: found the service with uuid: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (2686) BLERemoteService: >> getCharacteristics() for service: 91bad492-b950-4226-aa2b-4ede9fa42f59
D (2696) BLERemoteService: Found a characteristic: Handle: 51, UUID: 0d563a58-196a-48ce-ace2-dfec78acc814
D (2706) BLERemoteCharacteristic: >> BLERemoteCharacteristic: handle: 51 0x51, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
D (2716) BLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0d563a58-196a-48ce-ace2-dfec78acc814
D (2726) BLERemoteCharacteristic: Found a descriptor: Handle: 52, UUID: 00002902-0000-1000-8000-00805f9b34fb
D (2736) BLERemoteCharacteristic: << retrieveDescriptors(): Found 1 descriptors.
D (2746) BLERemoteCharacteristic: << BLERemoteCharacteristic
D (2746) BLERemoteService: << getCharacteristics()
D (2756) BLERemoteCharacteristic: >> readValue(): uuid: 0d563a58-196a-48ce-ace2-dfec78acc814, handle: 51 0x33
D (2766) FreeRTOS: Semaphore taking: name: ReadCharEvt (0x63613837), owner: <N/A> for readValue
D (2766) FreeRTOS: Semaphore taken:  name: ReadCharEvt (0x63613837), owner: readValue
E (32776) BT: gatt_rsp_timeout disconnecting...
E (32776) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
D (32776) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
D (32786) BLEUtils: GATT Event: ESP_GATTC_READ_CHAR_EVT
D (32786) BLEUtils: [status: ESP_GATT_ERROR, conn_id: 0, handle: 51 0x33, value_len: 0]
D (32796) BLERemoteCharacteristic: << readValue(): length: 0
D (32796) SampleClient: The characteristic value was: 
D (32806) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
D (32816) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT
D (32816) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: 7e:16:85:d4:ca:7a]
D (32826) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CLOSE_EVT
D (32836) BLEUtils: GATT Event: ESP_GATTC_CLOSE_EVT
D (32836) BLEUtils: [status: Unknown, reason:ESP_GATT_CONN_TERMINATE_LOCAL_HOST, conn_id: 0]
D (32846) SampleClient: Setting the new value
D (32856) BLERemoteCharacteristic: >> writeValue(), length: 19
D (32856) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x0), owner: <N/A> for writeValue
D (32866) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x0), owner: writeValue
```
I have been trying to debug this for a while now and would be great if I can get some help! I don't have much experience but would love to learn. I hope it's just a stupid simple mistake on my end.

chegewara, 2017-10-30:
Lets try step by step. I just did some PR to this particular code and it did work to me, also i see you have made some changes. Thats fine. First chack if you have settings in menuconfig setup like here:
https://github.com/nkolban/esp32-snippets/issues/143#issuecomment-340295910

Then try to comment all changes youve made and see if this works. We will make it work somehow.

yangantang, 2017-10-30:
I've rebuilt everything from scratch and setup the config accordingly and when I ran, it boot loops continuously with this error and a different backtrace each time.

[https://pastebin.com/kRFYKLf3](url)

I'm using the nRF Connect app on a Pixel XL to test the connection.

chegewara, 2017-10-31:
@yangantang Sorry its taking so long, can you pastebin sdkconfig please?

yangantang, 2017-10-31:
No worries! Here you go, [https://pastebin.com/64H0VYTF](url).

chegewara, 2017-10-31:
Im pretty sure we are not working with the same toolchain, and your sdkconfig is somehow weird. There are options which, i think, not suppose to be.
This is your toolchain: 
`I (35) boot: ESP-IDF v3.0-dev-1057-g4048db3 2nd stage bootloader
`
and mine
`I (38) boot: ESP-IDF v3.0-dev-1028-g2c95a77 2nd stage bootloader`
This is from where i have toolchain:
http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html

I dont know where those lines come from: 
`CONFIG_BT_ENABLED=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1=
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4=
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BLUEDROID_PINNED_TO_CORE_0=y
CONFIG_BLUEDROID_PINNED_TO_CORE_1=
CONFIG_BLUEDROID_PINNED_TO_CORE=0`

chegewara, 2017-10-31:
Anyway, i cant test with external device, all my tests are against 2 devKitC boards. One is configured as server, with SampleServer and the other one is runing SampleClient. Programs work perfectly.

yangantang, 2017-10-31:
So I know for a fact that the extra configs come from an update released today. Also the toolchain I have there was downloaded today, maybe there are some minor updates.

I'm trying to replicate your setup to see if it works, but alas even my SampleServer doesn't function properly. I get an exception then the board reboots.
[https://pastebin.com/pJCcpS5T](url)

chegewara, 2017-10-31:
Okay, i have to update my toolchain first and i will take care of it.

chegewara, 2017-10-31:
@yangantang Can you provide link to latest toolchain?

yangantang, 2017-10-31:
I got it from the official website ([http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html](url)) or more specifically [https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-73-ge28a011-5.2.0.tar.gz](url). I think the only differences we have are that my esp-idf repo is of a newer version, updated sometime today.

chegewara, 2017-10-31:
Toolchain is the same as main, but you have different sdkconfig. Please try with this
https://pastebin.com/VZrux7Vr

Just make menuconfig first and edit what you need.

yangantang, 2017-10-31:
After copying your sdkconfig file it works when having 1 board as server and the other as client. Then I tried it on my phone which still doesn't work but I'm not getting random dumps and exits anymore!

But my project is to use the boards as clients and servers anyway.

Thank you so much!

chegewara, 2017-10-31:
I didnt check SampleWrite, it may be broke after my last commit. Let me see it.

chegewara, 2017-10-31:
Ok. You have to do 2 things to make this work. First you need to change serviceUUID and characteristicUUID in SampleWrite to match those from SampleClient. And now the trick, server from SampleWrite has to be somehow visible for clients who wants to write to our server. Lets make him visible. Add following line just before >> `	pAdvertising->start();`:
`	pAdvertising->addServiceUUID(pService->getUUID());
`

yangantang, 2017-10-31:
Thank you so much! Got it to work.

closed 2017-10-31 by yangantang