nkolban/esp32-snippets #153: Clarification using json.cpp

rweller59, 2017-10-31:
I'm a bit confused on how to use json class with nested arrays. The following seems to work but I'm not sure its is the correct method. And, is it necessary to delete JsonObject and JsonArray on function complete.
...create json
	JSON json;
	JsonObject obj = json.createObject();
	JsonArray j_array = json.createArray();
    	obj.setInt("name", name);
        for(int i=0; i<num_items; i++) {
        	JsonObject obj1 = json.createObject();
        	obj1.setString("ssid", network.wifi[i].ssid);
        	obj1.setString("pwd", network.wifi[i].pwd);
        	j_array.addObject(obj1);
        }
        obj.setArray("wifi", j_array);

...read json - couldn't  figure out how to do without using m_node
obj = json.parseObject(std::string(str));
    	j_array.m_node = cJSON_GetObjectItem(obj.m_node, "contact");
        for(int i=0; i<j_array.size(); i++) {
        	j_array.getObject(i).getString("name").copy(name, sizeof(name);
        }

nkolban, 2017-11-01:
For the first part of your code, this will generate a JSON object that, if serialized to a string would look like:

```
{
   "name": <name>,
   "wifi": [
      {
          "ssid: <ssid>,
          "pwd": <pwd>
      },
      ...
   ]
}
```

Is that would you expect?

I'm not sure what the notion is with the second part of the code ... what does the string look like that you are parsing?

Assuming it is also:

```
{
   "name": <name>,
   "wifi": [
      {
          "ssid: <ssid>,
          "pwd": <pwd>
      },
      ...
   ]
}
```

Then something like:

```
JsonObj obj = JSON::parseObject(str);
std::string name = obj.getString("name");
...???

```

Ahh ... there doesn' seem to get a `getArray()` method ... so we appear to be missing:

```
JsonObject::getArray()

```

Is that the question?

nkolban, 2017-11-01:
Working on the assumption that it was a getArray() function that was needed, I went ahead and added one.  We now have a method on a JsonObject called `getArray(<fieldName>)` that will return a JsonArray object from object with the given field name.

rweller59, 2017-11-01:
thanks Neil.
The first part of the code was working fine for me. I was mostly wondering about 'JsonObject obj1 = json.createObject();' inside the loop and concerned that I would create a memory issue at some point.

The 2nd part was also working and yes, I was also thinking that some sort of getArray was needed but thought perhaps I was just mis-understanding something....

I also added:
/**
 * @brief Convert the JSON object to an unformatted string.
 * @return A JSON string representation of the object.
 */
std::string JsonArray::toStringUnformatted() {
	char *data = cJSON_PrintUnformatted(m_node);
	std::string ret(data);
	free(data);
	return ret;
} // toStringUnformatted

-------------------
/**
 * @brief Convert the JSON object to an unformatted string.
 * @return A JSON string representation of the object.
 */
std::string JsonObject::toStringUnformatted() {
	char *data = cJSON_PrintUnformatted(m_node);
	std::string ret(data);
	free(data);
	return ret;
} // toStringUnformatted

nkolban, 2017-11-01:
When you create a JsonObject or JsonArray, you are responsible for deleting it.  When you parse a string and are given a JsonObject or JsonArray, you are responsible for deleting it.

The one exception is a JsonObject that you then add to another JsonObject through the parent's "setObject" method.   The JsonObject that you added to the parent now belongs to the parent and you must not delete it.  Instead, you eventually delete the parent.

Let me also study and see if there is a better way through an Object's destructor.   Do you have a preference for how you would like to see JsonObjects deleted at the end?

rweller59, 2017-11-01:
I don't really have a preference. At the moment I am deleting primary JsonObject and any JsonArray created. Tried deleting the child objects but that caused some issues as you pointed out. 

The primary thing for me here is the understanding so that I close everything correctly.

nkolban, 2017-11-01:
Let me take an additional action to see if we can't sensibly close a JSON Object if it goes out of scope without having to perform an explicit delete.  Until then, assume that all **top level** JsonObject instances need explicitly deleted.

nkolban, 2017-11-04:
I looked at it again and think that is all as it should be just now.  An explicitly deletion of a JsonObject or array is where I want to leave it.

rweller59, 2017-11-08:
works for me

rweller59, 2017-12-30:
Having some issues with json objects/arrays again. Case 1 works fine. Case 2 errors at end when I call  json.deleteObject(obj); I've got a variety of cases similar to either 1 or 2. Case type 2 always fails

I have a function that starts like this.
	JSON json;
	JsonObject obj = json.createObject();
	obj = json.parseObject(std::string(payload));
	if(!obj.isValid()) {
		json.deleteObject(obj);
		return false; }
	JsonArray j_array = json.createArray();
------------------------------------
(switch)
case 1:
  network = obj.getBoolean("network");
  ... similar statements
case  2:
  network = obj.getBoolean("network");
  j_array = obj.getArray("name");
        for(int i=0; i<j_array.size(); i++) {
        	type = j_array.getObject(i).getInt("type");
                .... similar statements
        }
------------------------------------
always ends like this
    json.deleteArray(j_array);
    json.deleteObject(obj);

nkolban, 2017-12-30:
Just a guess ... but imagine the following JSON:

```
{
  x: 1,
  y: "abc",
  z: [ "p", "q", "r"]
}
```

if we create a JSON object from this (`obj`) and then ask for the array within it:

```
j_array = obj.getArray("z");
```
the the array is not ours to delete.  The `obj` can be deleted, but not the array as that array is owned by the object.   The high level rule is that if we create an object or an array, then we must delete that object or array at the end.  However, in your code, in "case 2", you are coding:

```
j_array = obj.getArray("name");
```
which means that you are losing the previous reference to the array that you explicitly created:

```
JsonArray j_array = json.createArray();
```

I think I may be bold and say that what you have here is a logic/coding error?  You may be deleting an array that you _think_ you created but there appears to be a code path that, if I am following it correctly, you end up deleting an array that you merely took a reference to from within a containing object.

Please review and comment back.

rweller59, 2017-12-31:
got it. hopefully new code will resolve - funny that I can't delete an object that I created but I get the concept. Unfortunately can't test because of new bt security issue after espidf update but I'll close anyway

closed 2017-12-31 by rweller59

nkolban, 2017-12-31:
Please realize that there is no rush to close an issue.  If you are not satisfied or haven't had a chance to test, leave it open as long as you need or open it again as desired.

rweller59, 2017-12-31:
it was a pretty straightforward solution after your explanation so np - got ble functional again and the solution did work. thanks