nkolban/esp32-snippets #154: E (1428) BT: btc_gattc_call_handler()

SensorsIot, 2017-11-01:
My scenario: 
One ESP32 runs BLE_UART example
The second one runs BLE_Client (checking on name, not on UUID because I do not get the UUID)
Newest library, newest ESP32 build from yesterday. Built according to setup directions on GitHub

If I connect with my iPhone or Raspi to the UART ESP it starts to send notifications. I also see serviceUUID 
The client ESP32 shows this log:

Starting Arduino BLE Client application...
BLE Advertised Device found: Name: UART Service, Address: 24:0a:c4:12:ee:ee, txPower: -21
BLE Name: UART Service
Found our device!  address: Forming a connection to 24:0a:c4:12:ee:ee
 - Created client
E (1428) BT: btc_gattc_call_handler()
E (1550) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e
E (1550) BT: p_conn is NULL in bta_gattc_conn_cback

 - Connected to server

I get this error message (btc_gattc_call_handler()) with other devices like Polar H7, Xiaomi MiFloara

chegewara, 2017-11-01:
This is a little bit modified BLE_client from arduino test folder. Modifications are made to find the other esp32 by name, to connect with TX characteristic and to send message to RX characteristic. Only thing i have to check is why notifications are not working.

> /**
>  * A BLE client example that is rich in capabilities.
>  */
> 
> #include "BLEDevice.h"
> //#include "BLEScan.h"
> 
> // The remote service we wish to connect to.
> #define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
> #define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
> #define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
> 
> static BLEAddress *pServerAddress;
> static boolean doConnect = false;
> static boolean connected = false;
> static BLERemoteCharacteristic* pRemoteCharacteristic;
> static BLERemoteCharacteristic* pRemoteCharacteristic2;
> 
> static void notifyCallback(
>   BLERemoteCharacteristic* pBLERemoteCharacteristic,
>   uint8_t* pData,
>   size_t length,
>   bool isNotify) {
>     Serial.print("Notify callback for characteristic ");
>     Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
>     Serial.print(" of data length ");
>     Serial.println(length);
> }
> 
> bool connectToServer(BLEAddress pAddress) {
>     Serial.print("Forming a connection to ");
>     Serial.println(pAddress.toString().c_str());
>     
>     BLEClient*  pClient  = BLEDevice::createClient();
>     Serial.println(" - Created client");
> 
>     // Connect to the remove BLE Server.
>     pClient->connect(pAddress);
>     Serial.println(" - Connected to server");
> 
>     // Obtain a reference to the service we are after in the remote BLE server.
>     BLERemoteService* pRemoteService = pClient->getService(SERVICE_UUID);
>     if (pRemoteService == nullptr) {
>       Serial.print("Failed to find our service UUID: ");
> //      Serial.println(SERVICE_UUID.toString().c_str());
>       return false;
>     }
>     Serial.println(" - Found our service");
> 
> 
>     // Obtain a reference to the characteristic in the service of the remote BLE server.
>     pRemoteCharacteristic = pRemoteService->getCharacteristic(CHARACTERISTIC_UUID_TX);
>     pRemoteCharacteristic2 = pRemoteService->getCharacteristic(CHARACTERISTIC_UUID_RX);
>     if (pRemoteCharacteristic == nullptr) {
>       Serial.print("Failed to find our characteristic UUID: ");
>  //     Serial.println(CHARACTERISTIC_UUID_TX.toString().c_str());
>       return false;
>     }
>     Serial.println(" - Found our characteristic");
> 
>     // Read the value of the characteristic.
> 
>     pRemoteCharacteristic->registerForNotify(notifyCallback);
> }
> /**
>  * Scan for BLE servers and find the first one that advertises the service we are looking for.
>  */
> class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
>  /**
>    * Called for each advertising BLE server.
>    */
>   void onResult(BLEAdvertisedDevice advertisedDevice) {
>     Serial.print("BLE Advertised Device found: ");
>     Serial.println(advertisedDevice.toString().c_str());
> 
>     // We have found a device, let us now see if it contains the service we are looking for.
>     if (advertisedDevice.getName() == "UART Service") {
> 
>       // 
>       Serial.print("Found our device!  address: "); 
>       advertisedDevice.getScan()->stop();
> 
>       pServerAddress = new BLEAddress(advertisedDevice.getAddress());
>       doConnect = true;
> 
>     } // Found our server
>   } // onResult
> }; // MyAdvertisedDeviceCallbacks
> 
> 
> void setup() {
>   Serial.begin(115200);
>   Serial.println("Starting Arduino BLE Client application...");
>   BLEDevice::init("");
> 
>   // Retrieve a Scanner and set the callback we want to use to be informed when we
>   // have detected a new device.  Specify that we want active scanning and start the
>   // scan to run for 30 seconds.
>   BLEScan* pBLEScan = BLEDevice::getScan();
>   pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
>   pBLEScan->setActiveScan(true);
>   pBLEScan->start(30);
> } // End of setup.
> 
> 
> // This is the Arduino main loop function.
> void loop() {
> 
>   // If the flag "doConnect" is true then we have scanned for and found the desired
>   // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
>   // connected we set the connected flag to be true.
>   if (doConnect == true) {
>     if (connectToServer(*pServerAddress)) {
>       Serial.println("We are now connected to the BLE Server.");
>       connected = true;
>     } else {
>       Serial.println("We have failed to connect to the server; there is nothin more we will do.");
>     }
>     doConnect = false;
>   }
> 
>   // If we are connected to a peer BLE Server, update the characteristic each time we are reached
>   // with the current time since boot.
>   if (connected) {
>     String newValue = "Time since boot: " + String(millis()/1000);
>     Serial.println("Setting new characteristic value to \"" + newValue + "\"");
>     
>     // Set the characteristic's value to be the array of bytes that is actually a string.
>     pRemoteCharacteristic2->writeValue(newValue.c_str(), newValue.length());
>   }
>   
>   delay(1000); // Delay a second between loops.
> } // End of loop

chegewara, 2017-11-01:
After further investigation i see there is from time to time this error:
`BT: p_conn is NULL in bta_gattc_conn_cback`

nkolban, 2017-12-14:
Closing for just now

closed 2017-12-14 by nkolban

forhuan, 2018-01-30:
Does notifications(notify) working now ? @chegewara 

```
Forming a connection to a4:c1:38:98:78:a7
 - Created client
[D][BLEClient.cpp:75] connect(): >> connect(a4:c1:38:98:78:a7)
E (3259) BT: btc_gattc_call_handler()
...
[D][BLERemoteCharacteristic.cpp:472] registerForNotify(): >> registerForNotify(): Characteristic: uuid: 0000ffe1-0000-1000-8000-00805f9b34fb, handle: 19 0x13, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1532] dumpGattClientEvent(): [status: ESP_GATT_OK, handle: 19 0x13]
[D][BLERemoteCharacteristic.cpp:503] registerForNotify(): << registerForNotify()
We are now connected to the BLE Server.

```

chegewara, 2018-01-30:
I think it should have work. Do you have server device that is sending notifications and esp32 is not receiving it?

forhuan, 2018-02-04:
yes, I have a Body Fat Scale which is sending notifications , i try to connect it with esp32 and registered the notifyCallback function , but it doesn't work at all.
i try to connect the scale using the FastBLE apk on Android and the notify success.

chegewara, 2018-02-04:
ok then, can we open new issue on this? we will make it work togheter, also im available on irc channel #esp32