nkolban/esp32-snippets #156: BLERemotecharacteristic register for notifications issue

chegewara, 2017-11-01:
There is issue with semaphore in registerForNotify, handle is 0x0 and we cant register to server for notifications:

```
D (2277) BLERemoteCharacteristic: >> registerForNotify(): Characteristic: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, handle: 44 0x2c, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
D (2297) FreeRTOS: Semaphore taking: name: RegForNotifyEvt (0x0), owner: <N/A> for registerForNotify
D (2306) FreeRTOS: Semaphore taken:  name: RegForNotifyEvt (0x0), owner: registerForNotify
E (2314) BLERemoteCharacteristic: REGISTER
D (2319) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
V (2319) FreeRTOS: Semaphore waiting: name: RegForNotifyEvt (0x0), owner: registerForNotify for registerForNotify
D (2327) BLEUtils: GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
```

nkolban, 2017-11-02:
You make a great comment.   Here's the back story.  As you probably know, a semaphore is the ability for one task/thread to block waiting for an indication from another task/thread.  We have encapsulated the implementation of this in a C++ class called FreeRTOS::Semaphore.

Until about a month ago, the only way to work with semaphores was through FreeRTOS ... with APIs such as:

* xSemaphoreCreate
* xSemaphoreGive
* xSemaphoreTake

and all was as it should be.  However, the good folks at Espressif gave us a new option.  They implemented the library called "pthreads" which is a Posix threading API.  Included in that are APIs such as:

* pthread_mutex_lock
* pthread_mutex_unlock
* pthread_mutex_init

I was wondering if some of our mysterious bugs could have been caused by C++ libraries using pthreads and the cpp_utils libraries using FreeRTOS.  So we implemented a runtime option that gives us the option of using pthreads or FreeRTOS for the implementation of FreeRTOS:Semaphore.  When we use pthreads, there is no "handle" that can be logged to show us the value of the mutex variable.

I am now changing the implementation back to using FreeRTOS.

chegewara, 2017-11-02:
Ive seen that option, but here is the thing. With set to true and using pthreads still most Semaphores got handle but that one.

nkolban, 2017-11-02:
When using pthread, we never assign to the semaphore handle.  What that means is that the semaphore handle's value is "undefined" and that can mean that if we print a value that was undefined we will get an undefined value ... maybe 0 ... maybe something random.  The correct solution will be to not write a handle value if we are using pthreads.

closed 2017-11-02 by chegewara