nkolban/esp32-snippets #158: wifi.cpp scan hangs after declaring object instance

rweller59, 2017-11-01:
Could you please explain the following

This scenario DOES NOT work for me:
WiFi wifi_debug;
auto results = wifi_debug.scan();
... hangs on:  0x4011cf9c: WiFiEventHandler::eventHandler(void*, system_event_t*) at C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/Wi
FiEventHandler.cpp:26

This scenario DOES work for me:
WiFi *pwifi1;
pwifi1 = new WiFi;
auto results = pwifi1->scan();

nkolban, 2017-11-02:
I have tried to recreate and both scenarios worked for me.

Here is the task that I ran...

https://pastebin.com/YX52vgSK

The usual suspects apply ... 

1. Refresh your ESP-IDF master Git image to the latest (a new one was delivered this week)
2. Rebuild everything (if in doubt, remove the project ./build directory which will get rebuilt)

rweller59, 2017-11-02:
Installed latest espidf. Trying to re-compile - error in cpp_utils...
C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/BLEUtils.cpp:1290:52: error: 'struct esp_ble_gattc_cb_param_t::gattc_connect_evt_param' has no member named 'status'
     BLEUtils::gattStatusToString(evtParam->connect.status).c_str(),
                                                    ^
C:/esp/esp-idf/components/log/include/esp_log.h:242:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/BLEUtils.cpp:1306:55: error: 'struct esp_ble_gattc_cb_param_t::gattc_disconnect_evt_param' has no member named 'status'
     BLEUtils::gattStatusToString(evtParam->disconnect.status).c_str(),
                                                       ^
C:/esp/esp-idf/components/log/include/esp_log.h:242:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/BLEUtils.cpp: In static member function 'static void BLEUtils::dumpGattServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':
C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/BLEUtils.cpp:1592:23: error: 'struct esp_ble_gatts_cb_param_t::gatts_connect_evt_param' has no member named 'is_connected'
     evtParam->connect.is_connected);
                       ^
C:/esp/esp-idf/components/log/include/esp_log.h:242:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
C:/Users/Rick/Documents/EclipseProjects/orb1/components/cpp_utils/BLEUtils.cpp:1609:23: error: 'struct esp_ble_gatts_cb_param_t::gatts_connect_evt_param' has no member named 'is_connected'
     evtParam->connect.is_connected);
                       ^
C:/esp/esp-idf/components/log/include/esp_log.h:242:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
make[1]: *** [/c/esp/esp-idf/make/component_wrapper.mk:243: BLEUtils.o] Error 1
make: *** [C:/esp/esp-idf/make/project.mk:435: component-cpp_utils-build] Error 2

nkolban, 2017-11-02:
Howdy @rweller59 
This sure is a moving target.  Those good folks at Espressif changed the API structure that exposed the BLE APIs.  To be clear in my explanation ... at the lowest level, there is the ESP32 hardware.  Above that is a private implementation of Bluetooth that is super low level.  This is called "Bluedroid".   Above that is the ESP-IDF which is what ESP32 applications can program to.  Above that is the BLE C++ classes that you are using now.  The BLE C++ classes use the ESP-IDF that is exposed by Espressif.

About 2 days ago, that latest release of pre-3.0 ESP-IDF changed the data structures exposed by ESP-IDF that are used by BLE applications.  This  resulted in exactly the errors you encountered.  Thankfully, a chap called @Philius caught the problem and quickly made code changes to the BLE C++ library.  This was committed last night.

My hope/belief is that if you re-download the BLE C++ classes from this repository, all should be back in synch again.

How can we avoid these problems?  Not easily I'm afraid.  When ESP-IDF releases their 3.0 release, I'm also going to create a branch of the C++ classes that will always be guaranteed to work with 3.0.  As ESP-IDF then moves on to their next release, there will then be two development streams in this repository.  One which will be guaranteed to work with 3.0 and one that will keep in synch with the latest ESP-IDF.

If you or anyone else have better / alternate notions, I'm very open to hearing those about how you would like this library maintained.

Please let me know if a fresh download of the BLE C++ libraries found here resolves the issue.

rweller59, 2017-11-02:
Just got back to this and updated. Compiled fine. 

I'm not overly concerned with this type of issue personally - I'm not trying to stay on the cutting edge of espidf releases unless there is a have-to-have. So, I'll keep trying stuff out and let you know when I find issues. Adding mqtt and I2C now so I'm sure you'll hear from me.

fyi - I still have the original issue as stated above. I'll do more digging tomorrow and try to figure out how my code structure differs from your simple example and hopefully shed some light

nkolban, 2017-11-03:
@rweller59 Thank you sir.  I'm here to assist as best I can.  I'm not going away from this project any time soon.

rweller59, 2017-11-03:
so, here is what I have found.
I have a class call serviceWIFI with a method call wifiScan(). I created a serviceWIFI obect in main and call wifiScan.
in main:
  serviceWIFI serviceWifi;
  serviceWifi.wifiScan();

In the serviceWIFI class, I create WiFi wifi_debug and call wifi_debug.scan
 - if I declare wifi_debug INSIDE of wifiScan(), things work fine
 - if I declare wifi_debug OUTSIDE of wifiScan so that it is available class wide, program hangs at the statement listed in my original post up top

Philius, 2017-11-03:
In make menuconfig:
Did you set FreeRTOS->"Enable FreeRTOS static allocation API"?

rweller59, 2017-11-03:
I had not. Just tried it - clean, build. Didn't make a difference. Stopped at same spot

nkolban, 2017-11-03:
@Philius Thats a new one on me.  Do you know the semantics of that flag?

@rweller59 Weekend is coming.  I work 9-5 Mon/Fri but the weekends give me more time to work on ESP32.  Please believe that your ticket is first up.

I believe you have given me enough of a description that I can picture your code in my mind and will try and build a recreate from that.  Just to be clear, I believe you are **not** using Arduino but are instead using plain ESP-IDF.

rweller59, 2017-11-03:
That is correct. Eclipse c++ and ESPIDF. I appreciate you taking a look at it. I've got plenty of other things to work on in the mean time.

Philius, 2017-11-03:
I had to look it up just now myself :)
https://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#config-support-static-allocation

> Bear in mind that if this option is enabled you will need to implement the following function:
`void vPortCleanUpTCB ( void *pxTCB ) {
    // place clean up code here
}`
Maybe the cleanup is there for OTA updates?

Philius, 2017-11-03:
@nkolban, @rweller59 sorry I forgot to "at" you guys XP

rweller59, 2017-11-03:
Can you give me some info I why I would need to do this (FreeRTOS->"Enable FreeRTOS static allocation API"?) in the first place for this situation?

Philius, 2017-11-03:
@rweller59 the WiFi class has a FreeRTOS::Semaphore data member m_gotIpEvt.
Having a global WiFi instance means it's allocated in static ram, along with m_gotIpEvt.

nkolban, 2017-11-04:
@rweller59 

Again I tried to recreate the issue and I again I couldn't ... it still worked for me.

Here is the code I tried:

https://pastebin.com/DZ2WgLz7

Could you look over it and see if this is what you have?  If not, consider performing a pastebin of what you have and I'll give that a try.

rweller59, 2017-11-04:
Used your code in my existing project but I ended up with the same problem. Practically everything is commented out trying to get this to work but I'm assuming I've created a conflict somewhere.

I am starting from scratch with a new project to see what happens. Finally got basic project settings correct and compiling but I've got to run to something else and will get back to this in a few hours.

fyi - got this error when first compiling cpp_utils until I turned on c++ debug in menuconfig.
C:/Users/Rick/Documents/EclipseProjects/orb2/components/cpp_utils/Socket.cpp:70:9: error: exception handling disabled, use -fexceptions to enable
   throw se;

things take forever to recompile - time for a new computer...

chegewara, 2017-11-04:
@rweller59 To fi x last error run: make menuconfig->Compiler options->Enable C++ exceptions

rweller59, 2017-11-04:
new project. hangs at exact same spot with this message in console
0x4011c7c8: WiFiEventHandler::eventHandler(void*, system_event_t*) at C:/Users/R
ick/Documents/EclipseProjects/orb2/components/cpp_utils/WiFiEventHandler.cpp:26

the only changes from default I made with menuconfig were flash port, enable c++ exceptions, verbose

https://pastebin.com/D5060MK5

chegewara, 2017-11-04:
@rweller59  Just copied, pasted and compiled your code. This is output:
```
I (146) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (147) wifi: mode : sta (30:ae:a4:06:4f:20)
ssid: Orange-8F54, auth: WIFI_AUTH_WPA_WPA2_PSK, rssi: -34
ssid: NETIASPOT-8D1C60, auth: WIFI_AUTH_WPA_WPA2_PSK, rssi: -94
ssid: FON_NETIA_FREE_INTERNET, auth: WIFI_AUTH_OPEN, rssi: -95
```
Please try with this sdkconfig:
https://pastebin.com/bCSBnmi8

rweller59, 2017-11-04:
used that skdconfig - changed the name of com port through menuconfig. couldn't tell if it hung at the same point because debug messages are off.
https://pastebin.com/NHWChekg

chegewara, 2017-11-04:
Now we have little progress. I see you are working on windows. What toolchain you are using? 
`xtensa-esp32-elf-gcc-5.2.0 -v`

rweller59, 2017-11-04:
sorry, thought I had said that before. Windows 10, msys32, Eclipse Oxygen, toolchain = Cygwin GCC, builder = Gnu Maker Builder

rweller59, 2017-11-04:
guess I should be using xtensa...

chegewara, 2017-11-04:
Im not sure, but last pastebin message can be related to whitespaces during copy/paste of my sdkconfig:
```
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe0' in position 0:
ordinal not in range(128)
make: *** [/c/esp/esp-idf/components/esptool_py/Makefile.projbuild:96: monitor] Error 1
```
Lets try this gist:
https://gist.github.com/chegewara/df2c0f5ab55bd2cd19e9a81d54e3e6c2

You can change most options in menuconfig, just try to avoid to alter options in ESP32-specific and FreeRTOS, then if it works you can try to change more and more whatever you need

rweller59, 2017-11-04:
For the project I was working on originally, I followed the ESPIDF docs on setting up project for windows Eclipse. On this project I just created, I started with the steps on your youtube video 'programming in c++'. Had to make quite a few changes to get properties set so that I could compile.

I can't find any options to define toolchain as xtensa-esp32-elf-gcc-5.2.0 -v

Does your last comment mean start with the skdconfig in the link and start making changes until things work?

chegewara, 2017-11-04:
Mast last comment mean check with this sdkconfig isntead that from last pastebin. If it works do whatever you need in menuconfig. 
`xtensa-esp32-elf-gcc-5.2.0 -v` is a command to check toolchain version. Use it in MSYS/MING32 please

rweller59, 2017-11-04:
version of sdkconfig didn't seem to make a difference. function is called then it hangs on serviceWiFi.wifiScan();

results of xtensa-esp32-elf-gcc-5.2.0 -v:
https://pastebin.com/A0xDQ2aY

chegewara, 2017-11-04:
Now i have big problem. Ive build MSYS environment on windows 10. Then what i did in steps:
- git clone esp-idf --recursive,
- git clone esp32-snippets,
- copied template code from esp-idf/examples/hello-world,
- altered hello_world_main.c with main.cpp
- copied/pasted https://pastebin.com/D5060MK5 into main.cpp
- changed options in menuconfig->serial flasher and menuconfig->compiler options,
- make flash
- all works fine.

Im using this MSYS version, or older one https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170918.zip. It is configured with toolchain

nkolban, 2017-11-04:
I know this isn't going to win friends ... but I'm going to suggest building under Linux.  If we wanted to go that route, we have two choices:

1) Install Virtual Box and then Ubuntu ... now we have a Virtual machine running Linux and we can setup an ESP32 build environment there.

or

2)  On Windows 10 with the latest fixpacks, we can run Windows Subsystem for Linux (WSL) which gives you Ubuntu running inside Windows without a virtual machine.  That Ubuntu can access the normal Windows file system.

My vote would be option 1.

While I personally use Windows as my desktop, ALL my development is done in Ubuntu running on Virtual Box.  I have found the tools fast and easy to use.  While admitedly, I like to think I know Unix, the amount of Unix commands that need to be known are low.

rweller59, 2017-11-04:
The only steps I can see that I did differently was potentially use a different version of msys. I don't remember how I installed it but I thought that I was working through the espidf docs. Also, if you aren't using eclipse that would be a difference as well.

I am using Eclipse and trying to set up all of the environment so I could have messed up something there. Would it make sense if I tried without Eclipse? I haven't done that before so it might take me a bit to figure it out.
 
Also, I could delete my version of msys and re-install? 

Sure sounds like its something in the linking/compiling process

rweller59, 2017-11-04:
I've got a machine here with Ubuntu 16 installed. It's not very convenient to work on but if it solves some issues I can do that.

I do not know linux at all so its a bit daunting - I always end up with permissions issues and have never really taken the time to try and figure it out.

Do either of the steps in my last post sound like they've got a decent chance of potentially solving the issue? I feel like I'm running in circles and would love to take a positive step here. I just got some mqtt stuff compiling last night and thought I saw some similar issues on my first run there but shut down for the night and haven't gotten back to it.

chegewara, 2017-11-04:
My suggestion is to prepare new environment. Its quick and easy.

I dont see the reason why you cant work with eclipse. My first steps was on windows, and im still using this under linux sometimes, to write code with eclipse and then compile it from CLI with make flash. 
Now im working on vbox linux, but i know it requires rsources, especially a lot RAM. Working on linux makes life easier, but eclipse + msys on windows is not a big issue.

nkolban, 2017-11-04:
I work in Eclipse on Linux on Virtual Box on Windows 10.  A whole slew of permutations.

It might help if we spent a little time on Skype to show me the problem live?

My skype id is "neil.kolban".

rweller59, 2017-11-04:
What kind of resources do I need on the windows box to set up virtual?  I've got i7CPU, 2.8gHz, 8GB ram, running 64 bit mode. If that's not good enough, I'll switch to the other machine.

I'd love to skype if you think you might see something. Does, it make sense with the setup I've got or should I get the linux set up first?

chegewara, 2017-11-04:
Your rig is enough to run vbox linux and work with comfort. I have laptop with intel i3 and 8GB ram.
With my setup i can work with comfort on code with eclipse and have running onlive tv stream in chrome browser. Believe us, working with eclipse on linux makes life easier

rweller59, 2017-11-04:
ok, I'll setup Virtual Box, Ubuntu, Eclipse and ESPIDF. Might take awhile and I don't know how much longer my wife is going to let me work tonight. I'll let you know when I've got things running

chegewara, 2017-11-04:
My last 2 cents. Ive been trying ununtu 17, but it reqires more ram than ubuntu 14, so i personally prefer to work on linux ubuntu 14

rweller59, 2017-11-04:
Is 14 much different than 16? I've been using 16 for the past year or so and have a bit of experience

chegewara, 2017-11-04:
Then stick with 16

rweller59, 2017-11-05:
still working through the linux install. I can config with menuconfig but I can't flash because I get a 'permission denied accessing /dev/ttyUSB0 error after compiling when its trying to upload. Haven't had any luck finding any solutions other than adding user to dialog group which is done. I'm doing this from the console and haven't starting working through eclipse yet.
Any suggestions?

chegewara, 2017-11-05:
Only i can think of is dialog group, and you have to relogin when you add user to dialog group.
Or this port is occupied by some app but i dont see any reason why

rweller59, 2017-11-05:
I was thinking maybe I need to give permissions to python - trying to figure out to do that now

nkolban, 2017-11-05:
There are two solutions to this problem .... one is immediate and the other is long term.  When we add a serial port to Linux, it will show up as /dev/ttyUSB0.  If we run "ls -l /dev/ttyUSB0" we will see the permissions on that file.  On mine it shows up as:

```
crw-rw---- 1 root dialout
```

The way to read this is that the user root has read and write permissions.  Anyone who is a member of the group `dialout` will have read and write permissions ... and everyone else has no permissions.

Two fixes
1) sudo chmod 666 /dev/ttyUSB0

This fix will change the permissions to:
```
crw-rw-rw--
```
Which means that others have read and write permissions.  However this change will be lost when you next attach the USB and would have to be repeated.

2) Add your userid to the `dialout` group.

```
$ sudo useradd -G dialout yourUserId
```

After that, logout (or reboot) and login again.  Running the command:

```
$ id
```
should show that you are now part of the dialout group

eg.
```
$ id
uid=1000(kolban) gid=1000(kolban) groups=1000(kolban),20(dialout),27(sudo)
```

rweller59, 2017-11-05:
yep, got that far. this allows me to access with menuconfig. I can  issue sudo chmod 777 and that allows me to flash but the permission goes away when session ends as you stated.
Adding myself to dialout doesn't allow me to flash and I'm guessing there is a python user that needs to be added but haven't figured out who the python user is.

I'm going to try and run the wifi example tonight and see if the issue has gone away then figure out eclipse setup and hopefully will be back up and running and moving forward. As long as I can issue a temp permission at least  I can get work done.

Have you looked at the members of dialout? Is there a python user or something similar listed?

chegewara, 2017-11-05:
To confirm that you really got issues with python access permision you can download and start putty. But in situation when you didnt issue chmod 777

rweller59, 2017-11-05:
The issue was that I logged out and logged back in thinking I would then load my groups. That didn't happen. I had to shut down Ubuntu then start again before it recognized that I was part of dialout group.
I just flashed hello_world successfully. On to Eclipse

rweller59, 2017-11-06:
got Eclipse up and running. compiled wifi example with class obj. doesn't hang so I guess I'm back in business.
thanks for all of the help. the linux/eclipse environment is no big deal once set up

nkolban, 2017-11-06:
I really do believe you will be happier building in Linux in the long run.  If nothing else, its the same environment that the majority of ESP32 users use so when problems are encountered in future, Windows as a platform can be ignored.

rweller59, 2017-11-07:
I kind of like playing in linux, at least until something comes up that I can't figure out - like why std::string won't resolve. And yes, I did find your discussion regarding the issue (https://www.esp32.com/viewtopic.php?f=13&t=1284) although none of the suggestions seemed to fix it it on my machine... seems kind of funny that other libraries seem to be found.

At least I can compile at this point

nkolban, 2017-11-07:
If desired, I can offer some of my time via a Skype session to have a look at individual issues.  I am in US Central Time and available after 6:00pm (usually 10:00pm till Midnight) and at weekends.

rweller59, 2017-11-07:
well, it helps to type xtensa rather than extensa. makes a big difference - that's what I get for looking at instructions in windows and setup in virtual. can't cut and paste.
Things look good again so finally back to work. I'll save your skype offer for something non-trivial

nkolban, 2017-11-07:
Can we call this issue closed?  If so, go ahead and close it out.  You can open new issues for new puzzles.

rweller59, 2017-11-07:
solution:  error went away after moving from windows/eclipse to ubuntu/eclipse

closed 2017-11-07 by rweller59