nkolban/esp32-snippets #160: BLEScan

chegewara, 2017-11-02:
BLEScan::start() is causing crash when trying to scan second time:

```
0x4013c227: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112

Backtrace: 0x4008ab80:0x3ffee8e0 0x4008ac4b:0x3ffee900 0x4013c227:0x3ffee920 0x4013c26e:0x3ffee940 0x4012362f:0x3ffee960 0x40123736:0x3ffee980 0x4012d995:0x3ffee9a0 0x4010b17f:0x3ffee9c0 0x401571e9:0x3ffee9f0 0x40157493:0x3ffeeb10 0x4014c925:0x3ffeeb70 0x40119a92:0x3ffeebf0 0x401070dd:0x3ffeed30 0x401d4f35:0x3ffeedd0 0x4015c819:0x3ffeedf0 0x4015d5f9:0x3ffef230 0x4015ef84:0x3fff0390
0x4008ab80: invoke_abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:553

0x4008ac4b: abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:553

0x4013c227: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112

0x4013c26e: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112

0x4012362f: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87

0x40123736: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54

0x4012d995: __gnu_cxx::new_allocator<char>::allocate(unsigned int, void const*) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:2263
 (inlined by) std::allocator_traits<std::allocator<char> >::allocate(std::allocator<char>&, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/alloc_traits.h:360
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.tcc:157

0x4010b17f: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.tcc:223

0x401571e9: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char*>(char*, char*, std::__false_type) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:195
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:214
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<char*, void>(char*, char*, std::allocator<char> const&) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:538
 (inlined by) std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::str() const at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/sstream:173
 (inlined by) std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::str() const at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/sstream:766
 (inlined by) FreeRTOS::Semaphore::toString[abi:cxx11]() at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./FreeRTOS.cpp:200

0x40157493: FreeRTOS::Semaphore::take(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./FreeRTOS.cpp:169

0x4014c925: BLEScan::start(unsigned int) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./BLEScan.cpp:180

0x40119a92: BLEExplorer::scan() at /home/esp32/synchro/ESP32_Explorer/main/./BLEExplorer.cpp:39

0x401070dd: handle_REST_BLE_CLIENT_SCAN(HttpRequest*, HttpResponse*) at /home/esp32/synchro/ESP32_Explorer/main/./ESP32Explorer.cpp:407

0x401d4f35: PathHandler::invokePathHandler(HttpRequest*, HttpResponse*) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./HttpServer.cpp:429

0x4015c819: HttpServerTask::processRequest(HttpRequest&) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./HttpServer.cpp:294

0x4015d5f9: HttpServerTask::run(void*) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./HttpServer.cpp:294

0x4015ef84: Task::runTask(void*) at /home/esp32/synchro/ESP32_Explorer/components/cpp_utils/./Task.cpp:58
```
Some issues with semaphore?

nkolban, 2017-11-02:
When ever I see this in the trace:

`__gnu_cxx::new_allocator<char>::allocate(unsigned int, void const*)`

My mind immediately goes to the notion that the heap is under pressure and doesn't have memory to issue.  Can we determine how much heap is free prior to the call to BLE Scan Start?

The failure on a second run may be due to resources not being released from the first run?

nkolban, 2017-11-02:
I have a notion on releasing an additional 16K of RAM.  I believe you noticed that BLEUtils uses too much ... I believe I know how to reduce that by at least 16K.

chegewara, 2017-11-02:
We have enough heap to do another scan. Its from ESP32_Explorer, after 1st acan i can create BLE server etc.
System reports its still over 30kB free heap.

chegewara, 2017-11-02:
Its small chance it is heap related issue. Still investigating.

BTW, 16kB of RAM is nice piece of cake

chegewara, 2017-11-02:
`E (52829) ESP32Explorer: esp_get_free_heap_size: 29292, esp_get_minimum_free_heap_size: 16396`

chegewara, 2017-11-02:
Confirmed, its short of free heap. BTW Socket still require optimization. Is taking over 1400 bytes.

closed 2017-11-02 by chegewara