nkolban/esp32-snippets #161: Create service with 16bit UUID fails to start.

databeredning, 2017-11-02:
I'm trying to use the following line to create a 16 bit UUID service ( Device Info 0x180A )
pServer->createService(BLEUUID((uint16_t)0x180A));
This will not work and the service will not get started properly.

Characteristics created will work though. For example (manufacturer 0x2029):
pChar = pService1->createCharacteristic(BLEUUID((uint16_t)0x2A29), BLECharacteristic::PROPERTY_READ);
pChar->setValue("Company Name");

I am running Arduino-ESP32 as a component for my ESP-IDF.

chegewara, 2017-11-02:
After investigating situation i found that we have issue with creating 16 bit services. When you try to create service 0x180A but with 128 bit sting it will be working. 

Require further investigating to fix a bug.

chegewara, 2017-11-03:
@databeredning lets try it now, it works for me with latest esp32_snippets library update.

databeredning, 2017-11-03:
Thank you @chegewara for fast response. Yes 128bit works as mentioned. I'm not quite sure what you mean by esp32_snippets library could you please explain a bit further where to find the changes you did so I can merge them with my source?

chegewara, 2017-11-03:
@databeredning Its this library we are using, my bad with using _ in place of -. esp32-snippets got updated and all should now work as suppose to be

databeredning, 2017-11-03:
@chegewara Sorry to bother you in what might be a trivial question but I really don't know how to use or even compile esp32-snippets to be able to use your changes. As mentioned I use the latest Arduino-esp32 as a component in my project. Is there a better way than copying files from snippets to my project and recompile. Please advice.

databeredning, 2017-11-03:
@chegewara So I think I found out how to do it. I used the Makefile.arduino to create a zip-file and then just exchanged the corresponding arduino library folder. Still having problem with 16bit Service UUID. Please have a look at my code below which work when I use the commented out row. ( But obviously shown as a 128bit UUID to client. )

```
#define DEVICE_INFO_SRV_UUID       "0000180A-0000-1000-8000-00805F9B34FB"

extern "C" void app_main()
{
  initArduino();
  BLECharacteristic *pChar;
  BLEService *pService1;

  BLEDevice::init("BLE Timer");
  BLEServer *pServer = BLEDevice::createServer();

  pService1 = pServer->createService(BLEUUID((uint16_t)0x180A));
  //pService1 = pServer->createService(DEVICE_INFO_SRV_UUID);

  pChar = pService1->createCharacteristic(BLEUUID((uint16_t)0x2A29), BLECharacteristic::PROPERTY_READ);
  pChar->setValue("Company Name");

  pService1->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();

  while(1)
  {
    vTaskDelay(1000 / portTICK_RATE_MS);
  }

}

```

chegewara, 2017-11-03:
@databeredning Sorry for my ignorance. Im not working with Arduino and i just assumed that it will work if its working with esp-idf. Now i see ive been wrong, and i will try to fix it asap

chegewara, 2017-11-03:
Sadly its not working under esp-idf, still investigating, but it will take a while since its bed time. Sorry

databeredning, 2017-11-03:
No worries :) I'll use 128bit UUID style until it's sorted out. Appreciate all your work!

chegewara, 2017-11-03:
@databeredning You can kill me but i dont know how, its working for me with Arduino

databeredning, 2017-11-03:
I'm running some kind of hybrid solution as I'm not all in Arduino in my project. But maybe I can sort things out in my end if you could please share your working code?

chegewara, 2017-11-03:
Sure. Its Arduino code - example BLEserver with 1 change to create service fom 16bit hex value:
```

/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("MyESP32");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x180A));
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

chegewara, 2017-11-03:
And this is esp-idf code. Its mutlipurpose test server code, so it may look weird, but idea to create service from 16 bit value is the same:

```
/**
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include <esp_log.h>
#include <string>
#include <Task.h>

#include "sdkconfig.h"

static char LOG_TAG[] = "SampleServer";

class MainBLEServer: public Task {
	void run(void *data) {
		ESP_LOGD(LOG_TAG, "Starting BLE work!");

		BLEDevice::init("ESP32");
		BLEServer* pServer = BLEDevice::createServer();

		BLEService* pService = pServer->createService("91bad492-b950-4226-aa2b-4ede9fa42f59");

		BLECharacteristic* pCharacteristic = pService->createCharacteristic(
			BLEUUID("0d563a58-196a-48ce-ace2-dfec78acc814"),
			BLECharacteristic::PROPERTY_BROADCAST | BLECharacteristic::PROPERTY_READ  |
			BLECharacteristic::PROPERTY_NOTIFY    | BLECharacteristic::PROPERTY_WRITE |
			BLECharacteristic::PROPERTY_INDICATE
		);

		pCharacteristic->setValue("Hello World!");

		BLE2902* p2902Descriptor = new BLE2902();
		p2902Descriptor->setNotifications(true);
		pCharacteristic->addDescriptor(p2902Descriptor);

		pService->start();
		pService = pServer->createService(BLEUUID((uint16_t)0x180A));
		pService->start();
		

		BLEAdvertising* pAdvertising = pServer->getAdvertising();
		pAdvertising->addServiceUUID(BLEUUID(pService->getUUID()));
		pAdvertising->start();

		ESP_LOGD(LOG_TAG, "Advertising started!");
		delay(1000000);
	}
};

void SampleServer(void)
{
	//esp_log_level_set("*", ESP_LOG_DEBUG);
	MainBLEServer* pMainBleServer = new MainBLEServer();
	pMainBleServer->setStackSize(20000);
	pMainBleServer->start();

} // app_main

```

chegewara, 2017-11-03:
My last advise: do a clean compilation, this mean, first `make clean`, then delete build folder and then compile with make.

databeredning, 2017-11-03:
I can make it work in Arduino IDE environment like you, question is where the differences are because I softlinked the BLE source so both my ESP-IDF projects component folder and Arduino IDE use the same code base and it still doesn't work when I run my ESP-IDF code. Strange indeed so I will continue developing this specific project under Arduino IDE just to get it working in time. Please post any progress so I can participate in testing.

chegewara, 2017-11-05:
@databeredning Im investigating this problem, but as for now no result. Until we find source of this issue this can help:
```
#define SERVICE_UUID_16 (uint16_t)0x180A
pService = pServer->createService((BLEUUID(SERVICE_UUID_16)).toString());
```

databeredning, 2017-11-07:
My conclusion so far is that the only way for me to success in having the 0x180A 16 bitt UUID service to show up like "180A" in any client I have to stick to my installed Arduino IDE + Arduino-esp32. I have no success using esp-idf environment either by latest cpp_utils or by softlinking source from Arduino-esp32 release. As said I'll stick to Arduino IDE even if it's not my favourite cup of tea.

Anyway, a great thank you for a really great job with developing (in my case primary) BLE library. I hope to be able to participate more later as I already extended functionallity by implementing a very modest class for handling 0x2901 descriptor.

nkolban, 2017-11-28:
Closing for just now.  Re-open as needed.

closed 2017-11-28 by nkolban

giobauermeister, 2018-03-14:
I'm having the same issue. Any updates on this?