nkolban/esp32-snippets #162: Save ram in BLE Utils

nkolban, 2017-11-02:
Within BLE Utils, we create tables of human readable descriptions of services, characteristics and BLE device providers.  There is about 20K of text.  The current implementation references these in structures.  An example of a structure would be:

```
struct entry {
   std::string description;
   uint16_t id;
};
```

These would be initialized similar to:

```
const struct entry myData[] = {
   {"text1", 1}, {"text2", 2}, {"text3", 3} .... {"", 0}
};
```

The assumption had been that because `myData` was flagged as a constant,  the data would be readonly.  Readonly data doesn't have to live in RAM but can instead live in flash and hence had no burden on our RAM usage.  However my implementation was horrible ... can **you** see the problem?  I didn't for the longest time.

Look again at the definition of `struct entry`:

```
struct entry {
   std::string description;
   uint16_t id;
};
```

The problem is `std::string` ... while this is of course a representation of a text/binary buffer ... it is **actually** a C++ class object.  What that means is that for every entry in the array, a C++ class instance of type std::string was being created and the text written into the class ... i.e. it was being carved out of the heap!!!

The solution was trivial.  Change the data from "std::string" to "const char *".   The fix for that is coming next....

nkolban, 2017-11-02:
Changes made ... we now leave this defect open for 14 days to make sure that all is good.

chegewara, 2017-11-03:
Great work, it saves me about 25kB.

nkolban, 2017-11-03:
Digging further, we find that coding:

```
const char* myVar = "My data";
```
causes the data to be created in initialized RAM
while
```
const char myVar[] = "My data";
```
causes the data to be created in read only flash.

We found we could save 500bytes (1/2K) in HttpRequest using this technique.

copercini, 2017-11-03:
Howdy @nkolban 
Where can I find more details about `const char* myVar` vs `const char myVar[]` ?

nkolban, 2017-11-03:
Howdy my friend.   Not sure what you mean about "more details".   What Ive been doing recently is studying the Linux "nm" command.  When we run this over a compiled source file (an object file) we get a dump of the "nature" of that compiled code.  Have a run of nm over an object file and have a look.  There is a wealth of information "coded" in there.  While working with a chap who needs BLE and WiFi running at the same time, he was running out of RAM so I though I'd take a look at how the cpp_utils classes were using RAM and see if there was any immediate and gross wasteage. 

That's where I found that strings defined as:

```
char *mydata = "hello world";
```
were mapped to initialized RAM data

while data mapped to
```
char mydata[] = "hello world";
```
are mapped to read-only flash.  I'm pretty darn certain this isn't rocket science to those who work in this level of knowledge at all ... and there may be write-ups on it all over the place (if there is, i'd love to hear about them too).

nkolban, 2017-11-18:
Closing for just now.

closed 2017-11-18 by nkolban

robpelu, 2018-01-18:
Hi Neil, I don't know if you are aware of this trick to lower BLE consumption after initialization. It releases some 30KB of RAM, but I have to test that all BLE functions are still working after this.
https://www.esp32.com/viewtopic.php?t=3139

Perhaps this could be included in the class after the init code ?

chegewara, 2018-01-18:
@robpelu This line is prepared to free heap, but its commented because some time ago it was some bug in esp-idf causing heap crash. Its not tested for some time, but im using it in my apps in app_main():
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L330

robpelu, 2018-01-20:
OK thanks chegewara, actually it works releasing the memory, but when I want to use BLE functions it crashes with some messages related to memory allocation, so probably there are some issues with the IDF.

Thanks.

chegewara, 2018-01-20:
https://github.com/espressif/esp-idf/blob/5401a75badcfc9c20f7818fdf3c683e3da8b2547/components/bt/include/esp_bt.h#L215

robpelu, 2018-01-21:
Thanks chegewara, I've read the comments but I have some doubts, since it says that esp_bt_controller_mem_release() should be called only before esp_bt_controller_init(), but here is after the init call (although it's commented out)
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L330

Could you show me some example of when it's possible to call it to release mem ?
Because I'm only using BLE and BT classic is not selected in menuconfig, but it doesn't matter where I use the releasemem call, it fails after that.

Thanks in advance.
Robert.

chegewara, 2018-01-21:
Its commented because it caused errors. It caused error because its called after esp_bt_controller_init(). Its called after esp_bt_controller_init() because i didnt read documentation thorough enough.

This is example how i used it and it works:
https://github.com/nkolban/ESP32_Explorer/blob/master/main/main.cpp#L75

robpelu, 2018-01-21:
Thanks that worked like a charm !!, the strange thing is that I modified the init() class function and put the release call before esp_bt_controller_init(), but still crashed the BT functions. Anyway, outside the class in app_main() as you said works so problem solved.

Thanks a lot !.
BR, Robert.