nkolban/esp32-snippets #163: service advertising not working

monteslu, 2017-11-03:
Just modified the BLE_notify example to add a line to advertise the service:
```
  // Start advertising
  pServer->getAdvertising()->addServiceUUID(SERVICE_UUID);  // <- added this
  pServer->getAdvertising()->start();
```

In your video demo of this example sketch with web bluetooth you had to filter by name and specify `optionalServices` with the service uuid.

When a BLE peripheral advertises a service, you can just filter by `services` without a name or the optionalServices property.

I have a sketch for the Arduino 101/ TinyTile that has a working service advertisement:  https://github.com/monteslu/ble-io/blob/master/arduino/curie_plus/curie_plus.ino#L127
(This sketch exposes a generic GPIO service I've been connecting to for johnny-five/nodebots)

Here's the two peripherals (curie chip device and esp32) viewed using the android NRF Connect app from android:
[image]
As you can see, the curie device is advertising a service uuid, and the esp32 isn't.  Is this a bug, or perhaps am I missing a step?

btw, thanks for creating ESP32_BLE_Arduino !

chegewara, 2017-11-03:
@monteslu see this https://github.com/nkolban/esp32-snippets/issues/144#issuecomment-340185708

Short story, your device name is too long. You have to make it max 5 characters long. If it wont help please get back and we will work on it.

monteslu, 2017-11-03:
fast response :) 
thanks, @chegewara  I'll give that a shot

monteslu, 2017-11-06:
that was it.  thanks!

closed 2017-11-06 by monteslu

devinsewell, 2018-10-13:
Try adding pAdvertising->addServiceUUID(pService->getUUID()); before pAdvertising->start();