nkolban/esp32-snippets #164: Clarification on using snippets example code

rweller59, 2017-11-03:
I am going to use pahomqtt library but cannot figure out how to add it into the esp dev environment - I am using Eclipse c++ with ESPIDF and windows 10.

Thought it would be fairly straightforward, especially since examples exist in snippets but after hours of trying to figure it out I thought I'd finally ask for some help

nkolban, 2017-11-03:
Apparently it was January when I last looked at this code ... sorry ... I'm going to be sluggish.  It looks like one takes the directory from this repository called 

networking/mqtt/paho_mqtt_embedded_c/

and places that in your ESP32 project components folder.

<yourproject>/components/paho_mqtt_embedded_c

By doing that, it will cause the code to be compiled and linked.

There appears to be a sample here.

https://github.com/nkolban/esp32-snippets/tree/master/networking/mqtt/paho_mqtt_embedded_c/ESP32_Sample

of a client caller.

rweller59, 2017-11-03:
thanks, that was certainly easy. I had tried something similar but copied the 'MQTTClient-C' folder instead.

nkolban, 2017-11-03:
No problems.  Feel free to join in any time as a collaborator on these projects.  If you think the README or other documentation could be improved ... have a bash.  And of course, if you choose not to, that's also not a problem.  I just want to make sure you feel welcome to come join the fun and games of being a repository contributor :-)

rweller59, 2017-11-03:
I'll be happy to join the fun. I realize that I asked this before but, what is the best method for adding to the code and/or documentation. I'm not always fully confident in my code so not sure you want me to add that directly.

nkolban, 2017-11-03:
I'll be happy to assist in any way I can.  Read about Git and Git Hub.  At a high level, you will create a git hub account, clone the esp32-snippets project into your own github account.  Clone/pull YOUR copy onto your PC ... make changes, commit the changes back to your Github clone and then issue a pull request.

Alternatively, if all of that is too much ... feel free to just write the changes you want made as text posts in an issue and I'll transcribe.

rweller59, 2017-11-04:
Had some trouble getting paho mqtt compiled so thought I'd start out with something I hoped was a bit more straightforward and imported this library instead (https://github.com/tuanpmt/esp32-mqtt). I've got most of it ready to use with c++ but ran into something I wasn't expecting... When I try to call the library functions (from c++) defined in the header and c files, I get undefined reference errors.

Is there a simple step I am missing to call c functions from a component library? The library is loaded under the components directory. I have access to the structure and variable definitions. I looked through your cpp_utils cpp code but didn't see that you were doing anything other than calling a function.

chegewara, 2017-11-04:
@rweller59 If i understand, you are trying to use this library https://github.com/tuanpmt/espmqtt/tree/2967332b95454d4b53068a0d5484ae60e312eb12
This is plain C library, so need to include header files inside extern guard:
```
extern "C" {
#include some_header.h
}
```

rweller59, 2017-11-04:
I could swear that I had tried that but what I had done instead was put the functions in the declaration
extern "c" {
c_func a();
c_func b();
}
Everything now compiles - thanks once again

rweller59, 2017-11-08:
went back to paho_mqtt library. copied library from snippets into project components directory. Tried to compile, got error (undefined reference to `mbedtls_debug_set_threshold'). Turned mbedTLS debug ON in menuconfig. Compiled, ran and works.

Looks like there are quite a few changes in embedded library since you pulled it so thought I'd update. Downloaded latest version and replaced previous version in components directory with this latest version. I am now getting a bunch of unknown type name errors like this:
_components/paho.mqtt.embedded-c/MQTTClient-C/src/MQTTClient.h:127:5: error: unknown type name 'Network' Network* ipstack;_

Are there steps that I did not take for configuring a components library. I made sure that component.mk exists but haven't done anything else

rweller59, 2017-11-21:
figured it out. most of configuration was inside MQTTLinux.h. Still having some other issues with the library but not related to this

closed 2017-11-21 by rweller59