nkolban/esp32-snippets #165: Bootwifi not recognized for the program

katianemedeiros, 2017-11-03:
Hello, I'm trying using the library bootwifi but i have some problems because the library is not recognized for the program. I already checked the location of the library and I can't find the solution.

chegewara, 2017-11-03:
Hi, can you provide some more info. Is it Arduino IDE or eclipse? Log message?

mtrohman, 2017-11-06:
hi, could you please how to use bootwifi in esp32, thankyou.

chegewara, 2017-11-06:
```
#include "BootWiFi.h"
extern "C"{
void app_main(void);
}
void app_main(void)
{
    BootWiFi *boot = new BootWiFi();
    boot->boot();
    while(1){
         vTaskDelay(10000);
    }
}
```

When you flash and run esp32 then at first esp32 will start AP. You can find open wifi AP with name esp32. Connect to it with any device, then from browser connect to address 192.168.4.1 and fill form with setting from your home router(assuming you will run esp32 at home). SSID and passwrd are mandatory (assuming your home network is not open) and ip, gateway, netmask are optional. When you submit form, esp32 will store values you issued in nvs and will reboot. Then every time when esp32 will restart and find credentials in nvs will try to connect to wifi network with those credentials.

katianemedeiros, 2017-11-06:
@chegewara Thank's!! Now is working. 
But after I connect the adress 192.168.4.1 the ESP doesn't enter in the STA mode.

nkolban, 2017-11-06:
Howdy folks,
When using BootWiFi and the ESP32 doesn't know which Access Point to connect to, it will start itself as an access point.  At this point, you can open your cell phone and connect to the ESP32 through WiFi.  The default is that the ESP32 will show up as SSID: ESP32 and password "password".  If you then open a browser to "http://192.168.4.1" you will see a web page where you can enter your own access point information.  This is the information that the ESP32 will subsequently use to connect to the supplied access point.  Once you have submit that information, the ESP32 will reboot and attempt to connect to that access point.

If you wish to see additional diagnostics, then switch logging to "Verbose" in "make menuconfig" and the log/traces of the BootWiFi library will log to the console.   We will be delighted to make any changes that you would like to see on BootWiFi so if anything is not as desired, or you find problems, keep coming back and lets us know.

chegewara, 2017-11-06:
The only i can think of is to ask, did you update esp32-snippets? That second screenshot is good, it should be blank. In this moment esp32 should have to restart and join you local network. 
Try to use `make erase_flash` to get rid of any informations from nvs and try again to flash and input ssid and password

katianemedeiros, 2017-11-06:
Every time the program starts the variable rc starts with -1 because the version is not incompatible, how can I fix it?

chegewara, 2017-11-06:
Can you pastebin logs?

katianemedeiros, 2017-11-06:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5584
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:12476
entry 0x40078f74
W (73) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 40
 MHz.
I (39) boot: ESP-IDF v3.0-dev-806-gde750e99-dirty 2nd stage bootloader
I (39) boot: compile time 13:56:38
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2a100 (172288)
 map
I (164) esp_image: segment 1: paddr=0x0003a128 vaddr=0x3ffb0000 size=0x032ac ( 12972
) load
I (169) esp_image: segment 2: paddr=0x0003d3dc vaddr=0x40080000 size=0x00400 (  1024
) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Katiane/esp/esp-idf/components/freert
os/xtensa_vectors.S:1675

I (171) esp_image: segment 3: paddr=0x0003d7e4 vaddr=0x40080400 size=0x0282c ( 10284
) load
I (184) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x993f8 (627704
) map
0x400d0018: _flash_cache_start at ??:?

I (406) esp_image: segment 5: paddr=0x000d9418 vaddr=0x40082c2c size=0x0cc7c ( 52348
) load
0x40082c2c: lock_release_generic at C:/msys32/home/Katiane/esp/esp-idf/components/ne
wlib/locks.c:205

I (428) esp_image: segment 6: paddr=0x000e609c vaddr=0x400c0000 size=0x00000 (     0
) load
I (437) boot: Loaded app from partition at offset 0x10000
I (437) boot: Disabling RNG early entropy source...
I (439) cpu_start: Pro cpu up.
I (443) cpu_start: Starting app cpu, entry point is 0x40080e88
0x40080e88: call_start_cpu1 at C:/msys32/home/Katiane/esp/esp-idf/components/esp32/c
pu_start.c:219

I (0) cpu_start: App cpu up.
I (454) heap_init: Initializing. RAM available for dynamic allocation:
I (460) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (466) heap_init: At 3FFBA650 len 000259B0 (150 KiB): DRAM
I (473) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (479) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (485) heap_init: At 4008F8A8 len 00010758 (65 KiB): IRAM
I (492) cpu_start: Pro cpu start user code
I (177) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
1
2
3
I (179) bootwifi: >> boot
I (179) bootwifi:  +----------+
I (179) bootwifi:  | BootWiFi |
I (189) bootwifi:  +----------+
I (189) bootwifi:  Access point credentials: teste_AP/12345678
4
I (199) bootwifi: >> bootWiFi2
5
13
6
I (199) bootwifi: >> getConnectionInfo
**I (209) bootwifi: Incompatible versions ... current is 3ffc79c0, found is 100**
**************************************************
RC -1
**************************************************
I (299) wifi: wifi firmware version: ee52423
I (299) wifi: config NVS flash: enabled
I (299) wifi: config nano formating: disabled
I (299) system_api: Base MAC address is not set, read default base MAC address from
BLK0 of EFUSE
I (309) system_api: Base MAC address is not set, read default base MAC address from
BLK0 of EFUSE
I (329) wifi: Init dynamic tx buffer num: 32
I (329) wifi: Init data frame dynamic rx buffer num: 32
I (329) wifi: Init management frame dynamic rx buffer num: 32
I (329) wifi: wifi driver task: 3ffc29e4, prio:23, stack:4096
I (339) wifi: Init static rx buffer num: 10
I (339) wifi: Init dynamic rx buffer num: 32
I (339) wifi: Init rx ampdu len mblock:7
I (349) wifi: Init lldesc rx ampdu entry mblock:4
I (349) wifi: wifi power manager task: 0x3ffd09f4 prio: 21 stack: 2560
I (419) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 0
I (419) wifi: mode : softAP (30:ae:a4:0b:50:4d)
I (429) bootwifi: << bootWiFi2
7
I (429) BootWifiEventHandler: >> apStart
I (429) BootWifiEventHandler: << apStart
I (123869) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (123869) wifi: station: 14:ab:c5:a8:5d:bd join, AID=1, n, 40U
8

chegewara, 2017-11-06:
There is some problem with data in nvs. What you have to do is to `make erase_flash` first, then make flash app one more time. If the issue wont be fixed then something is been messed in this line https://github.com/nkolban/esp32-snippets/blob/master/networking/bootwifi/BootWiFi.cpp#L106 or this line https://github.com/nkolban/esp32-snippets/blob/master/networking/bootwifi/BootWiFi.cpp#L79 in your code.

closed 2017-11-09 by katianemedeiros