nkolban/esp32-snippets #166: BootWiFi example issue

chegewara, 2017-11-04:
@nkolban Hi, im trying to get it running, and i dont understand why WiFi::init() is private. Maybe im doing something wrong, so im asking about clarification. This is my start:

```
void app_main(void)
{
    BootWiFi *boot = new BootWiFi();
    boot->boot();
    while(1){
vTaskDelay(10000);
    }
}
```
Everything seems to work, except one thing. After setting up AP ssid and password thru website http://192.168.4.1/, esp32 restarts and is trying to retrieve data from. Till this point all works fine. Now we have this line: https://github.com/nkolban/esp32-snippets/blob/master/networking/bootwifi/BootWiFi.cpp#L233 and esp32 cant retrive data from nvs. What did i found its because nvs is not initialized yet. 
Part of code which makes initialization of nvs is in WiFi::init(). We can add `m_wifi.scan()`, and this fix issue with reading AP data from nvs, data is retrived and esp32 is connecting to AP, but gives very strange log output. 

```
D (6350) event: SYSTEM_EVENT_STA_GOT_IP, ip:36.56.19.128, mask:32.248.252.63, gw:68.217.252.63
I (6360) event: sta ip: 36.56.19.128, mask: 32.248.252.63, gw: 68.217.252.63

```

nkolban, 2017-11-04:
A whole slew of ugly errors (made by me of course).  The fixes here include:

* Mandatory call to nvs_flash_init() to initialize flash memory
* Bad handling/checking for no over-ridden IP, Netmask and GW.  If they weren't specified in the form returned from the browser, random values were used.
* Some cleanup of the README

Hopefully all should be well now.

chegewara, 2017-11-07:
There is one more issue. When app starts first time it will try to start in AP mode and all looks good. But when is trying to start in STA mode then stall because some issues with semaphore here >> https://github.com/nkolban/esp32-snippets/blob/master/networking/bootwifi/BootWiFi.cpp#L308
Commenting semaphores in BootWiFi::boot() https://github.com/nkolban/esp32-snippets/blob/master/networking/bootwifi/BootWiFi.cpp#L300, let run app normaly.
Commenting semaphores before first start, when there is no credentials in nvs dont let start app in AP mode.

nkolban, 2017-11-09:
Great catch.  Again I messed up.  Now we detect that IF we boot and attempt to connect as a station to an access point THEN IF we are allocated an IP address ... that indicates that the boot has been succesful and we release the semaphore and continue.  I haven't tested the code but all compiles cleanly and "reads ok".

nkolban, 2017-11-28:
Closing for just now.  Re-open as needed.

closed 2017-11-28 by nkolban