nkolban/esp32-snippets #167: ESP32_BLE_Arduino - It's possible to create a BLEServer with an specific GUID?

feliceferri, 2017-11-04:
Thank you for sharing this amazing library!

I'm trying to connect from a client using a function called ConnectToKnownDeviceAsync() that would take the DeviceID (GUID) as a parameter.

Wondering if there is any way to specify that GUID in the same way as you do with the name 
BLEDevice::init("MyESP32");

Thanks!

nkolban, 2017-11-04:
Howdy my friend.  Do you have a link to this API called "ConnectToKnownDeviceAsync()"?  What platform are you writing your client application upon?

In BLE we have the concept of Bluetooth Addresses (6 byte identifier of a device), service ids (16 bit, 32bit or 128bit identifiers of a service) and characteristic ids (16bit, 32bit or 128bit identifiers of characteristics exposed by a service).

Not being familiar with "ConnectToKnownDeviceAsync()", I'm afraid I don't know if this is a device name, a bluetooth address, a service id ... or something completely different.

I find this a good intro to BLE:

https://learn.adafruit.com/introduction-to-bluetooth-low-energy

feliceferri, 2017-11-04:
Dear Neil Kolban,

Thank you for your response, its a truly honor for me.

ConnectToKnownDeviceAsync(GUID) is part of this Xamarin Plugin called
*xamarin-bluetooth-le
<https://github.com/xabre/xamarin-bluetooth-le>*

https://github.com/xabre/xamarin-bluetooth-le

I already tried the followings:

connectedDevice.GetServiceAsync(Guid.Parse("
ffe0ecd2-3d16-4f8d-90de-e89e7fc396a5"));
service.GetCharacteristicAsync(Guid.Parse("
d8de624e-140f-4a22-8594-e2216b84a5f2"));
characteristic.ReadAsync();

They all work beautifully with your library!!!

But, It would be lovely if in the same way we can give the Device GUID to
the ConnectToKnownDeviceAsync(GUID), to avoid iterating trough all
Bluetooth Devices looking for (in this example) for the Name "MyESP32".

By the way, after that loop, I found that in my case the ESP32 Device GUID
was

00000000-0000-0000-0000-240ac40866a6

Of course, once I got it, I can use the ConnectToKnownDeviceAsync(GUID)
successfully.

Thanks again!

Sincerely,

Felice Ferri MD

chegewara, 2017-11-04:
@feliceferri My knowledge of Xamarin is not impressive, but you should have option to pair device. When you do that you can easy try to connect to that device without scanning and if that device is in range shuld be possible to connect.

feliceferri, 2017-11-04:
Yea, I already did it successfully!

And got the characteristic value "Hello World says Neil" in Xamarin
perfectly.

I was just wondering if I could look for the Bluetooth Server directly
using a "Pre-known" GUID that I could specify in Arduino, the same way as I
can specify the SERVICE_UUID and CHARACTERISTIC_UUID.

feliceferri, 2017-11-04:
I mean, that 00000000-0000-0000-0000-240ac40866a6 has to come from
somewhere, wonder if it is a Manufacturer GUID.

2017-11-04 0:56 GMT-04:00 Felice Ferri <feliceferri@gmail.com>:

chegewara, 2017-11-04:
Yes, you can. You can do this in two ways. One way is to add service to advertising with command `pserver->getAdvertising()->addService(UUID)`, and the other is by setting up appearance. In my opinion in this case 1st method is better.

feliceferri, 2017-11-04:
Thanks!

I'll try that

feliceferri, 2017-11-04:
I'm sorry, but it didn't work.

This is the Arduino Code:

  BLEDevice::init("MyESP32v2");
  BLEServer *pServer = BLEDevice::createServer();

pServer->getAdvertising()->addServiceUUID("00110011-0011-0011-0011-001100110011");

  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

I changed the name to "MyESP32v2" to be sure the chip was running this code.

I'm still getting the same GUID 00000000-0000-0000-0000-240ac40866a6  (It
remained the same after uploading the new Arduino Sketch)

I took a second ESP32 Dev Board => Uploaded the Arduino Sketch => I got
this GUID 00000000-0000-0000-0000-240ac40cb49e

First 3 bytes remained the same, probably as the Manufacturer Identificator
convention.

My Guess is that this GUID is the 6 byte identifier of the device exposed
by the Xamarin Plugin as an 16 byte GUID, and in the same fashion the
ConnectToKnownDeviceAsync()  instead of asking for a 6 bytes ID, ask for a
16 bytes GUID and then parses it.

Would you agree?

I really thank you for being so kind, and help me through

Thanks!

2017-11-04 1:00 GMT-04:00 Felice Ferri <feliceferri@gmail.com>:

chegewara, 2017-11-04:
Since i dont know what you are trying to achieve this is how i see this. 
1. When you first time start client application then app will enumerate all BT devices.
2. User will pair your esp32 device and app will remember his GUID, ( 00000000-0000-0000-0000-240ac40cb49e - its most likely mac address, which is unique)
3. Next time when user start application the app can check if there is any device paired and will try to connect to that device.

Now its second approach:
1. You can setup esp32 device to advertise 128 bit service.
2. Your client app can search for that particular service and when find it can connect to it without user knowledge and action. In this case you have to use different function than ConnectToKnownDeviceAsync().

Second approach is used here in sample code https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleClient_Notify.cpp#L86

nkolban, 2017-11-28:
Closing for just now.

closed 2017-11-28 by nkolban

chegewara, 2021-10-10:
@Marcus10110 You are mixing 2 things:
1) service UUID is something that every BLE peripheral can advertise and every central can recognize the same way, no matter android or iOS
2) android and most other BLE devices scan and can display BLE peripheral MAC address, just iOS as far as i know is making things harder and is hashing MAC address and you can see only hash instead of MAC, and they call it UUID

they are not the same and related.