nkolban/esp32-snippets #169: BLE + WiFi, WiFi get's disconnected

Cypher87, 2017-11-05:
The issue is also described here: [https://github.com/nkolban/esp32-snippets/issues/110#issuecomment-341973084](url)

I'm using the Arduino BLE package, with latest IDF. 

My code: [https://pastebin.com/zsfBBrde](url)

My output:

> Connecting...
.......
Connected!
E (34999) BT: btc_search_callback  BLE observe complete. Num Resp 2
5
e6:33:09:67:68:15
Sending data

Where it's first connecting to my WiFi network. Then starts a scan and sends it using udp. "5" is the WiFi state, which is WL_CONNECTION_LOST.

Sometimes this happens after a while (maybe 2 or 3 scans). WiFi does not recover after the connection has been lost.

@chegewara 
Your output [https://github.com/nkolban/esp32-snippets/issues/110#issuecomment-341975003](url) is indeed what I am expecting. Can you keep the scan running for maybe 2 or 3 minutes to see if you're losing your WiFi connection as well? It's not too important, so get some sleep before breaking your head over this issue ;)

chegewara, 2017-11-05:
I am running scan all the time since then. I dont know if there was any disconnections but im still getting the same output. 
```
E (1649515) BT: btc_search_callback  BLE observe complete. Num Resp 1
[D][BLEScan.cpp:205] start(): << start()
3
24:0a:c4:03:b1:c2
Sending data
3
Sending done
```

chegewara, 2017-11-05:
What we can do? Few questions:
- what Arduino ide number are you using? (mine is 1.8.5)
- when last time did you update esp-idf and esp32-snippets?

Cypher87, 2017-11-05:
Updated both last friday (3rd of November). Arduino version is 1.8.5 as well.

I've tried 2 different dev boards. Both show the same result. Which board are you using?

chegewara, 2017-11-05:
Im using devKitC, but in my opinion board is not important.
Can you try to update esp32-snippets? It was updated yesterday.
I dont remember if i did something some changes in sdkconfig (most likely i dont), so my next step would be to try this version. At first make some backup your old sdkconfig, you can also compare both files with some online compare files website. But first, just update library and try to compile.

https://gist.github.com/chegewara/6864f271004e30a919260d25cbc1a825

chegewara, 2017-11-05:
One important thing. You should have use this instruction to update library, not git pull from Arduino_BLE repo:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

Cypher87, 2017-11-05:
I've been using the Arduino_BLE version indeed. I've now updated it to the cpp_utils version. But getting the same results.

>>Connecting...
I (407) wifi: wifi firmware version: 111e74d
I (407) wifi: config NVS flash: enabled
I (407) wifi: config nano formating: disabled
I (422) wifi: Init dynamic tx buffer num: 32
I (458) wifi: Init data frame dynamic rx buffer num: 64
I (517) wifi: Init management frame dynamic rx buffer num: 64
I (583) wifi: wifi driver task: 3ffe3a94, prio:23, stack:4096
I (648) wifi: Init static rx buffer num: 10
I (695) wifi: Init dynamic rx buffer num: 0
I (742) wifi: Init rx ampdu len mblock:7
I (786) wifi: Init lldesc rx ampdu entry mblock:4
I (839) wifi: wifi power manager task: 0x3ffeaacc prio: 21 stack: 2560
I (2002) wifi: mode : null
I (2024) wifi: mode : sta (30:ae:a4:38:17:4c)
....I (4481) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
..I (5138) wifi: state: init -> auth (b0)
..I (6138) wifi: state: auth -> init (2)
I (6139) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
....I (8546) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (8547) wifi: state: init -> auth (b0)
I (8549) wifi: state: auth -> assoc (0)
I (8567) wifi: s.tate: assoc -> run (10)
I (8640) wifi: connected with [deleted], channel 9
...
Connected!
D (10575) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffeb374), owner: <N/A> for start
D (10576) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffeb374), owner: start
D (10620) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffeb374), owner: start for start
D (10621) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
D (10800) BLEUtils: [status: 0]
D (10834) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
D (10916) BLEUtils: [status: 0]
I (11602) wifi: pm start, type:0
D (13050) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (13051) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (13050) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -36, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (20024) wifi: bcn_timout,ap_probe_send_start
D (20773) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (20773) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (20773) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -44, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (22526) wifi: ap_probe_send over, resett wifi status to disassoc
I (22526) wifi: state: run -> init (1)
I (22526) wifi: pm stop, total sleep time: 0/10923819
I (22526) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
D (28497) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (28498) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (28497) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -39, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (32353) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (32353) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (32353) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -39, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (36213) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (36214) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (36214) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -36, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (40073) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (40073) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (40073) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -36, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
E (40621) BT: btc_search_callback  BLE observe complete. Num Resp 2
D (40622) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (40636) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_CMPL_EVT
D (40699) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffeb374), owner: start
V (40709) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffeb374), owner: start
V (40792) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffeb374), owner: start
5
e6:33:09:67:68:15
Sending data
5
Sending done

chegewara, 2017-11-05:
I see we have different logging outputs. Maybe its nothing important and you just have turned on higher level of logging. But...
Now is time to try with my sdkconfig. It is in ./hardware/espressif/esp32/tools/sdk/

https://github.com/nkolban/esp32-snippets/issues/169#issuecomment-341978070

Cypher87, 2017-11-05:
Yes, I've upped the log level to 5 (CONFIG_LOG_DEFAULT_LEVEL)

Now I replaced the sdkconfig file with your sdkconfig file. This is the output (looks the same):

> Connecting...
I (399) wifi: wifi firmware version: 111e74d
I (399) wifi: config NVS flash: enabled
I (399) wifi: config nano formating: disabled
I (412) wifi: Init dynamic tx buffer num: 32
I (449) wifi: Init data frame dynamic rx buffer num: 64
I (508) wifi: Init management frame dynamic rx buffer num: 64
I (574) wifi: wifi driver task: 3ffe3b1c, prio:23, stack:4096
I (639) wifi: Init static rx buffer num: 10
I (686) wifi: Init dynamic rx buffer num: 0
I (733) wifi: Init rx ampdu len mblock:7
I (777) wifi: Init lldesc rx ampdu entry mblock:4
I (830) wifi: wifi power manager task: 0x3ffeab3c prio: 21 stack: 2560
I (1984) wifi: mode : null
I (2006) wifi: mode : sta (30:ae:a4:38:17:4c)
....I (4463) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
..I (5120) wifi: state: init -> auth (b0)
I (5123) wifi: state: auth -> assoc (0)
I (5132) wifi: state: assoc -> run (10)
I (5162) wifi: connected with [deleted], channel 9
.
Connected!
D (6057) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffeb304), owner: <N/A> for start
D (6058) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffeb304), owner: start
D (6100) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffeb304), owner: start for start
D (6101) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
D (6278) BLEUtils: [status: 0]
D (6311) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
D (6392) BLEUtils: [status: 0]
D (6493) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (6499) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (6498) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (8132) wifi: pm start, type:0
D (10353) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (10354) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (10353) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (12013) wifi: bcn_timout,ap_probe_send_start
D (14213) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (14213) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (14213) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (14515) wifi: ap_probe_send over, resett wifi status to disassoc
I (14566) wifi: state: run -> init (1)
I (14607) wifi: pm stop, total sleep time: 0/6475479
I (14608) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
D (18075) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (18075) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (18075) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -47, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (21938) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (21939) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (21938) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -46, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (25799) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (25800) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (25800) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (29660) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (29660) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (29660) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
D (33524) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (33524) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (33524) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -41, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
E (36101) BT: btc_search_callback  BLE observe complete. Num Resp 2
D (36102) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (36115) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_CMPL_EVT
D (36179) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffeb304), owner: start
V (36189) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffeb304), owner: start
V (36272) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffeb304), owner: start
5
e6:33:09:67:68:15
Sending data
5
Sending done

Looking at the log, it looks like wifi is disconnecting right after I start the BLE scan? 

>I (14515) wifi: ap_probe_send over, resett wifi status to disassoc
I (14566) wifi: state: run -> init (1)
I (14607) wifi: pm stop, total sleep time: 0/6475479
I (14608) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1

chegewara, 2017-11-05:
If we have the same sdkconfig, and we have so different outputs then i have to study your logging very thorough to understand it. Now i will be heading to work, so next hour i am out.

Cypher87, 2017-11-05:
No problem. Thank you for your help thus far.

chegewara, 2017-11-05:
@Cypher87 Please dont take me wrong, whatever you did it did not change esp32-snippets with latest version. If you look closer on those lines:
```
D (6498) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (8132) wifi: pm start, type:0
D (10353) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
D (10354) BLEUtils: Unknown esp_ble_evt_type_t: 173 (0xad)
D (10353) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: e6:33:09:67:68:15, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_RANDOM, ble_evt_type: *** Unknown ***, rssi: -42, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 28, scan_rsp_len: 0
I (12013) wifi: bcn_timout,ap_probe_send_start
```
program does scan the same device over and over. But in latest library, dont remember since when, there is loop which is cheching if that device has been scaned and mapped. You can see it here:
```
[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x12 (ESP_BLE_AD_TYPE_INT_RANGE), length: 4, data: 20004000
[D][BLEAdvertisedDevice.cpp:278] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 12, data: 554152542053657276696365
[D][BLEAdvertisedDevice.cpp:342] setName(): - setName(): name: UART Service
[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.
[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 24:0a:c4:03:b1:c2, already seen it.
```

And there has been made very important changes in WIFI library. My advise: compare wifi.cpp from repo and from arduino library.

Cypher87, 2017-11-05:
I'm not sure what to say. I see the code that should log what you're seeing in my BTEUtils.cpp, but it's just not logging it that way.

I've replaced sdkconfig, but `tools\sdk\include\sdkconfig.h` didn't change. It seems that's the actual file that holds the config?

Also, I'm using the latest IDF from [https://github.com/espressif/arduino-esp32](url). So the WiFi.cpp is the same as in that repo.

Am I missing something here?

chegewara, 2017-11-05:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md,
and 
https://github.com/espressif/arduino-esp32
are not the same. Like i mentioned earlier, you nedd to git pull this repo
https://github.com/nkolban/esp32-snippets.git
and follow those steps

> Should you wish to build your own instance of the ZIP from the source, here is the recipe.
> 
> 1. Create a new directory called build
> 2. Enter that directory and run git clone https://github.com/nkolban/esp32-snippets.git
> 3. Change into the directory called ./esp32_snippets/cpp_utils
> 4. Run make -f Makefile.arduino build_ble
> 5. Change into the directory called ./Arduino
> 
> And here you will find the ESP32_BLE.zip that is build from the latest source. You can then install this into your Arduino IDE environment are you are ready to go.

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

Cypher87, 2017-11-05:
I understand. I don't have the BLE library installed from https://github.com/espressif/arduino-esp32

The library I'm using is from https://github.com/nkolban/esp32-snippets.git

Though it's still not working as it should. I'm getting a bit frustrated and wondering if I should just go another way. The ESP32 is just not stable enough. And all these differences in versions is not helping either.

Don't get me wrong, I do appreciate your help.

chegewara, 2017-11-05:
Is it possible you have installed 2 BLE libraries in your system and arduino is confused and is using wrong one?
There is one more option. Arduino is using WiFi library from other package, not esp32-snippets. Because there is a lot logs which dont even exists in wifi library:
`wifi: pm stop, total sleep time`
`wifi: ap_probe_send over, resett wifi status to disassoc`

If its possible, you can ask mr @nkolban to make skype session to solve this issue. Only i can say is that the library used to compile this program is not the same which is on this repo.

I totally understand your frustration. Been there many times. Got situation when everything been working, then i did clean git clone and nothing worked, just because i forgot to change options in menuconfig. Other time i compiled app and flashed it and app worked just fine, next i did the same with exact the same code and got crash all the time. I found that option which should help me with debugging caused that.
From my experience i can say one last thing. Its very important to prepare development environment in proper way. Ive been doing this few times until i decided it is the moment when everything works as it should. If its not big problem, then maybe just remove arduino, and all libraries, and start from beggining. Follow this istructions
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

chegewara, 2017-11-05:
At the end its with profit to all of us. When you and other users will have good prepared environment we can find and solve any issue a lot faster. And this im our/ my objective.

Cypher87, 2017-11-05:
You're totally right. I've started with a clean install. And followed:
[https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md](url)
to install the ESP32 lib and 
[https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md](url)
to install the BLE lib.

Then I changed sdkconfig.h to enable debugging. Still the same problem.

Because of this message in my logging:

`I (12013) wifi: bcn_timout,ap_probe_send_start`

I wondered if the scanning takes to long and breaks WiFi. So I changed the timeout to 5 seconds.
`BLEScanResults res = pBLEScan->start(5);`
 Now everything is working just fine. Somehow scanning for a longer period of time breaks WiFi.

chegewara, 2017-11-05:
It is possible. But this is good information that you eventually found source of this issue. I hope you will figure it how to make this program to work as you wish. 
Good luck

chegewara, 2017-11-05:
What you can do now is to try to split BLE scan and wifi to 2 different tasks. This may resolve issue with disconnecting wifi whan ble scan takes too long

Cypher87, 2017-11-06:
Created a task with:

`xTaskCreate(
    handleBT,
    "HandleBT",
    10000,
    NULL,
    1,
    NULL);`

Set the scan time to 30 again, still WiFi drops out.

chegewara, 2017-11-06:
Then for the moment you have to set time to scan short. Sorry, if this collide with your app functionality i can try to find reason, but this may be not so easy

Cypher87, 2017-11-06:
I also found this thread [https://github.com/espressif/esp-idf/issues/675](url) which sounds like the same issue.

Thanks for all your help. I will try to work my way around this issue.

ManniAT, 2017-11-08:
How about creating a pinned task - you created a task with priority 1 (same as loop) so I guess when you are blocking the cpu this will block the main task(s).
A pinned task (on core 0) could possibly help
Just my 2 cents

Cypher87, 2017-11-08:
Tried that as well. Did not resolve the issue either.

chegewara, 2017-11-08:
As you can see there is few BT tasks with very high priority. I dont think you can change any of them from system. Sure, there are freertos methods which lets you to change priority of any task, but its low level api and i would not risk to use it with ble or wifi.
[image]

Cypher87, 2017-11-09:
I'm wondering if anyone else is able to reproduce this issue. Just to make sure it's not my dev setup/ my dev-boards/power issue/wifi router or anything.

chegewara, 2017-11-09:
Just checked and i cant recreate it. Ive set scan time to 30 sec and is scanning prettynice, and no problem with wifi disconnect.
```
[D][BLEScan.cpp:205] start(): << start()
3
00:00:00:00:00:00
Sending data
3
Sending done
[D][BLEScan.cpp:178] start(): >> start(duration=30)
[D][BLEAdvertisedDevice.cpp:353] setRSSI(): - setRSSI(): rssi: -46
[D][BLEAdvertisedDevice.cpp:219] parseAdvertisement(): Type: 0xff (ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE), length: 29, data: 06000109200034a1144a17752bcbed69fe68532abe062ec44483b3cc67
[D][BLEAdvertisedDevice.cpp:330] setManufacturerData(): - manufacturer data: 06000109200034a1144a17752bcbed69fe68532abe062ec44483b3cc67
[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 00:00:00:00:00:00, already seen it.
[D][BLEScan.cpp:92] gapEventHandler(): Ignoring 00:00:00:00:00:00, already seen it.
```
```
[D][BLEScan.cpp:205] start(): << start()
[D][BLEScan.cpp:178] start(): >> start(duration=30)
E (164022) BT: btc_search_callback  BLE observe complete. Num Resp 2
[D][BLEScan.cpp:205] start(): << start()
[D][BLEScan.cpp:178] start(): >> start(duration=30)
E (196036) BT: btc_search_callback  BLE observe complete. Num Resp 2
[D][BLEScan.cpp:205] start(): << start()
[D][BLEScan.cpp:178] start(): >> start(duration=30)
E (228050) BT: btc_search_callback  BLE observe complete. Num Resp 2
```

If you wish i can send you my BLE zip library for arduio and sdkconfig. Then you can chack and corfim its stack fault or something else. (My BLE library has been created nov 3rd)

Cypher87, 2017-11-09:
Sure that would be nice. My anonymized address is: cru7f [at] notsharingmy.info

anhvu100, 2017-12-22:
I got the disconnection when using https://github.com/espressif/arduino-esp32 and installing BLE using Sketch -> Include Library -> Manage Libraries. In the library filter, enter "esp32 ble arduino".
The sketch is simple, keep running BLE scan for 2 seconds and if device is found, submit data to the web using HTTPClient.
                  BLEScanResults foundDevices = g_pBLEScan->start(2);
                  if (foundDevices.getCount() > 0) code-to-submit-data.

The wifi disconnect happen after running for 10-15 minutes.
Another test is: while the code is running, from another PC, trying to PING the ip of esp32 and the ping does not response all the time. Just type: ping -t 172.16.16.11 where 172.16.16.11 is the IP of the esp32.
Any idea why? Thank you!

chegewara, 2018-01-05:
@anhvu100 I dont think you can ping esp32 at all. You have to ask on forum.

KryvosheyaIvan, 2018-06-15:
@chegewara, where is the source of that interesting table with system tasks?

chegewara, 2018-06-15:
https://github.com/nkolban/ESP32_Explorer

hanadjerbi1, 2018-08-14:
https://github.com/espressif/arduino-esp32/issues/1756