nkolban/esp32-snippets #171: undefined reference to `app_main

vishnunaik, 2017-11-06:
when I am running make all in command window I am getting this error as:
[image]

I tried to change the code from extern void app_main(void) as  extern "C" void app_main(void); in cpu_start.c file. Still id didn't work.

please help out with this.
Thank you.

chegewara, 2017-11-06:
Please show output of: ll main

vishnunaik, 2017-11-06:
@chegewara  when I am typing in command window: II main, it giving as command not found.

chegewara, 2017-11-06:
then ls -al main

vishnunaik, 2017-11-06:
[image]

chegewara, 2017-11-06:
Im guessing you just copied SampleClient.cpp from example folder to main folder. Esp-idf application requires entry point to start, by default it is void app_main(). You have few options now. 

1. create main.cpp with this code:
```
extern void SampleClient();
extern "C" {
void app_main();
}
void app_main(){
SampleClient();
}
```
2. at the end of SampleClient.cpp add:
```
extern "C" {
void app_main();
}
void app_main(){
SampleClient();
}
```
There is few other options but those 2 are simplest.

vishnunaik, 2017-11-06:
Wow @chegewara your awesome it is working now. I am able to run the program.
Thank you for your support.

closed 2017-11-06 by vishnunaik