nkolban/esp32-snippets #172: Not able to connect the external Server with ESP32 (sampleclient.cpp program)

vishnunaik, 2017-11-06:
We entered our service UUID and Characteristic UUID of external server in sampleclient.cpp program. After flashing the program to ESP32 we are unable to search or connect the external server, we are getting error in our debug window as follows.  
[image]
Thank you very much.

chegewara, 2017-11-06:
From this screen i cat tell you only one thing. This is not error. Yes, it looks like error, because someone in espressif dev team set log to loge, but its just info that BT ended scanning and found 1 server nearby.

vishnunaik, 2017-11-06:
Oho Is it, but it is not able to proceed further to connect the server having services UUID mentioned in the program.
To connect the specific server what should we do?

chegewara, 2017-11-06:
You need to know what service is advertised by this server. In most cases, i think, server dont advertise any services. This example is prepared to connect to other esp32 with flashed SampleServer on it.
Its a lot to explain how its work, if you dont know bluetooth basic functionality.

vishnunaik, 2017-11-06:
Yes I know the all services and characteristics provided by my external server. The services and characteristics I can able to see by using "nrfconnect" app from Nordic and I can able to read the live sensor data with the nrfconnect app.
The same thing I want to do with the ESP32, by mentioning  service UUID and along with one of the characteristic in the sampleclient.cpp program. 
Could it be possible with the ESP32?

chegewara, 2017-11-06:
Like i said, it depends on what your external server is advertising. Your external server should have to advertise UUIDs similat to this on screenshot from this comment:
https://github.com/nkolban/esp32-snippets/issues/144#issuecomment-340185708

On this screenshot it is 128bit UUID, but it can be any 16bit UUID. If your server is advertising any UUIDs then yes, you can, if dont, then you can try to search this server by name and then join.

vishnunaik, 2017-11-06:
Yes @chegewara  our server advertising 128 bit UUID like which you shown in #144 (comment). 
Our 
service UUID is : 2ea78970-7d44-44bb-b097-26183f402400
Characteristic UUID : 2ea78970-7d44-44bb-b097-26183f402400
The picture shows in nrfconnect app as follows.
[image]  
If I want to connect it by advertising name what extra functions I need add in sampleclient.cpp program.
Thank you.

chegewara, 2017-11-06:
Your screenshot shows something different than mine. On my screenshot you can see informations provided by server before you even connect to it. You can see it in nRF when you scan and click on device name next to connect button.

vishnunaik, 2017-11-06:
@chegewara  before connection.
[image]

chegewara, 2017-11-06:
Thats it. As you can see there is one 16bit UUID, you can modify SampleClient to scan for device with that uuid. Its 0x2ea7. Im gussing you will have to try few, maybe even many times to change program to get required result. But if you will try to do it alone then you will learn how ble and this library works.

vishnunaik, 2017-11-06:
Thank you @chegewara  I will try.

chegewara, 2017-11-06:
This little snippet may be helpful:
```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
	/**
	 * Called for each advertising BLE server.
	 */
	void onResult(BLEAdvertisedDevice advertisedDevice) {
		ESP_LOGD(LOG_TAG, "Advertised Device: %s", advertisedDevice.toString().c_str());

//		if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
		if (advertisedDevice.getName() == "UART Service") {
			advertisedDevice.getScan()->stop();

			ESP_LOGD(LOG_TAG, "Found our device!  address: %s", advertisedDevice.getAddress().toString().c_str());
			MyClient* pMyClient = new MyClient();
			pMyClient->setStackSize(18000);
			pMyClient->start(new BLEAddress(*advertisedDevice.getAddress().getNative()));
		} // Found our server
	} // onResult
}; // MyAdvertisedDeviceCallbacks
```

vishnunaik, 2017-11-07:
Hi

I am also doing my project in the similar flow. I tried to include the UART services line in my code which you mentioned above  but there is no change in output. It is scanning and just displaying the devices nearby but is not getting connected to the BLE server.
How to write 16 bit UUID in the code.? Should I write it in the hexadecimal format or normal representation? Kindly help me with this issue.

nkolban, 2017-11-18:
Looking at the messages ... it isn't 100% clear to me the nature of the puzzle.  Lets start at the beginning.  I am assuming you are using your ESP32 as a BLE client and that you have an external BLE server.  So our first question is ... are you able to locate your BLE server using a BLE scan from the ESP32?

chegewara, 2017-11-18:
I think we have the same issue in here #198. The problem is that dongle is advertising service in 16 bit form and then you need to use 128 bit uuid to connect to it.

nkolban, 2017-11-18:
I chatted extensively with the raiser.  We saw that the ESP32 is indeed receiving an indication that the UUID is 16bit however when looking at the advertisement through nRFConnect and through Linux, we see that it appears to be advertising a 128bit UUID.  The discrepancy is odd to say the least.  Our next plan is to have the user install the latest ESP-IDF and the latest version of the BLE libraries.  Then, with debugging on, run it again.  Assuming that the problem is still present, we will add some code to the library to dump (in hex) the advertising message and see what it looks like from a raw data perspective.  From there we should be able to manually walk through it and see what we find.

chegewara, 2017-11-18:
Thats odd, because this screenshot shows that nRF sees 16 bit advertised service https://github.com/nkolban/esp32-snippets/issues/172#issuecomment-342122960 also i did speak with raiser #198 issue and he confirmed that nrfconnect sees 16 bit advertised service. And as we discovered earlier its the only way, because dongle name is too long to advertise 128 bit service.

nkolban, 2017-11-18:
ARGHHH!!!   I'm an idiot (and lots of people across the world are now nodding).  I have managed to confuse myself between what services are available within a BLE device and what services are advertised by the BLE device.  Now that you point it out it seems obvious so a HUGE thank you.

When we run a BLE scan from the ESP32, we are receiving advertising packets from the BLE partners.  The ESP32 logs/displays/returns what ever the BLE partner chooses to advertise.  Now, put that to one side.  When we CONNECT to a BLE partner, we can ask it "What services and characteristics do you offer?".   These two interactions are NOT the same (and that's where I messed up).  An advertisement is what ever a BLE device broadcasts for others to find.  A listing of exposed services and characteristics is what a BLE partner claims to be able to support/offer WHEN the partner ACTUALLY connects to it.

When running the Linux scan/examination tool ... we "scanned" for devices and the tool showed the device.  We then ran "info" against the device and say the services and their characteristics.  Where I goofed was thinking that the return from "info" was the result from the scan ... it wasn't.  It was the result from ACTUALLY connecting to the device and asking it what it offered.

The working hypothesis is that the BLE target device is "advertising" a 16 bit UUID but offering up a 128 bit UUID which can't be deduced from the 16 bit value.  Obviously I don't like these hypothesis because it pushes the issue onto a third party device (the BLE partner) and I always assume that if something is not working then its because of MY failure to understand/code and not something inherent in the partner.  However, now that mr @chegewara has pointed it out clearly, nRFConnect is saying "I found a 16 bit UUID" (which is what ESP32 also claims) HOWEVER, the actual partner device is advertising a 128bit UUID.

chegewara, 2017-11-18:
Like i said, this dongle is using this workaround to bypass restrictions in ble specification. In best case we can advertise one 128 bit service and in worst case we cant advertise even one. But we can advertise 16 bit service to let central to find us. But 16 bit services are not such unique like 128 bit, then we need to connect to device and check if this is for sure device we are looking for by serching for 128 bit service, which is the real one that peripharal is serving. 

It sounds complicated but it is not. Just force us to use one more additional variable.

chegewara, 2017-11-19:
This is example which will works with that dongle. Its not full support, only one characteristic to show how to do it. 
https://github.com/chegewara/esp32-tests-library/blob/master/arduino/BLE_client_uuid_advertising_172_198/BLE_client_uuid_advertising_172_198.ino

Since i dont have that dongle i have had to create emulator and this is my emulator:
https://github.com/chegewara/esp32-tests-library/blob/master/arduino/BLE_server_uuid_advertising_172_198/BLE_server_uuid_advertising_172_198.ino

vishnunaik, 2017-11-20:
I understand that our dongle is advertising one 16bit-UUID and after connected servicing 128-bit UUID. Thanks for the clarification.
@chegewara I have tried the code which is mentioned in the above comment. 
[](https://github.com/chegewara/esp32-tests-library/blob/master/arduino/BLE_client_uuid_advertising_172_198/BLE_client_uuid_advertising_172_198.ino)

The output log shows here: 
[image]

nkolban, 2017-12-14:
Closing for just now.

closed 2017-12-14 by nkolban