nkolban/esp32-snippets #174: Find Bluetooth devices around me ( ESP32 ) with RSSI

franzel61, 2017-11-07:
Dear Neil,

Actually I’m facing  an issue where I can’t find a way thru.

My intention is to use the ESP32 to get RSSI information and device information from any Bluetooth device which is in the range and act on the result ..
For a example:
Walking thru my homeoffice on RSSI  50 of my phone the room light switches on 70 the light o the desk switch’s on ..... if I leave my phone (Apple) on the desk and move out the system will recognize the RSSI of my watch is changing and the the system will act on this
 By using your BLE Example I can see a lot of devices around me, however my IOS devices are not visible and I get the information only ones after it’s found

Do you have a idea how I can solve to pass this part of my project?

Thx Franzel

nkolban, 2017-11-09:
OOh ... good stuff.  In my household I have tons of devices but not a single apple product and no iPhone.  I'm not rabidly against them ... just don't own one and have chosen Android as my cell phone device.  What this means is that I have personally never tested against an IOS device.  I haven't heard others yell out about this problem but the absence of knowledge doesn't mean that it isn't broke.

The first thing I'd check is find yourself someone with an Android device and install the Android "nrfConnect" tool.  That is a great BLE utility that I use all the time.  One of its capabilities is to display found BLE devices.   Run that and see if IT shows your IOS devices.   The answer to that will guide us on our next steps.

chegewara, 2017-11-09:
Ive been looking on it. I cant see my android phone either. Ive been trying every combination in scan parameters but nothing. I can see laptop and esp32 but i cant android phone. Maybe its question to espressif devs or some deep digging on google?

franzel61, 2017-11-09:
Hi Neil,
I ask a friende to perform a search with nrfConnect and he could see IOS devices with the MAC and the corresponding RSSI in scan mode without a BT Connect, exactly what I want to discover with the ESP32
[image]

nkolban, 2017-11-10:
Next step for me:  Going to run up an ESP32 and run the scanning tests to make sure that there hasn't been a regression.

nkolban, 2017-11-10:
Hmmm ... no problems finding devices on my end.  Here is the program I am using:

https://pastebin.com/gGXrvg6z

When I run it, after 10 seconds I get a list of found devices.  I started my Samsung Android phone and ran nRFConnect on it and asked it to advertise ... that was found as well:

```
D (10903) SampleScan: We found 4 devices
D (10907) BLEScan: >> Dump scan results:
D (10911) BLEScan: - Name: , Address: 39:61:d7:75:4c:78, manufacturer data: 06000109200007338a5492f313df14b45a80dcd5993fc365dbc3a6d148
D (10923) BLEScan: - Name: SAMSUNG-SM-G950U, Address: 73:b4:e0:c6:82:c6, manufacturer data: 11991234, serviceUUID: 91bad492-b950-4226-aa2b-4ede9fa42f59, txPower: -7
D (10938) BLEScan: - Name: , Address: 12:3b:6a:1a:e2:fd, manufacturer data: 4c000215fda50693a4e24fb1afcfc6eb076478250001fde2c5
D (10950) BLEScan: - Name: , Address: 12:3b:6a:1a:ee:93, manufacturer data: 4c000215b5b182c7eab14988aa99b5c1517008d9000193eec5
D (10961) SampleScan: Scanning sample ended

```

Now we need to study and see what is different.  I'm at a loss to explain.  Theory says that your iPhone will be broadcasting BLE packets and everyone in range should see them.  We see that through nRFConnect.  Now we need to ask "What is distinct in these iOS broadcast packets as compared to those broadcast by others?".   One more check would be to attempt to use a Linux machine with a BLE dongle on it and use some of the Linux BLE tools to dump what they see.   If that were to work, we could then install Wireshark and grab a super low level trace of the packets from a device that is detected as compared to iOS.

chegewara, 2017-11-10:
[image]
Most likely  its my phone fault i cant see it

nkolban, 2017-11-10:
And here's my found Samsung S8
[image]

chegewara, 2017-11-10:
Peripheral mode makes difference?

nkolban, 2017-11-10:
Don't know yet ... but its a clue for consideration.  Only by gathering information can we determine causes.  Obviously one Android phone shows up on ESP32 and the other doesn't so the notion is to compare what the phones tells us and think about the distinctions and see which might be useful and which can be excluded.

franzel61, 2017-11-10:
May found a idea, please check if that could be right
A.) Apple Manufacture are not recognised by name - Manufactor list is 004C, however some of my devices show up with 4C  ( missing the leading 00)
B.) i guess the MAC will not shown-up correct. My IOS test -device has the MAC 28:A0:2B:42:44:08 in the scan list it shows 4c:f5:5A:1b:61:f6

nkolban, 2017-11-10:
I'm far from an expert on BLE (sadly) ... however somewhere in the back of my mind was the notion that Bluetooth Device Addresses can change/randomize.  The idea behind this is that your cell phone may be pinging the world saying "I'm here, tell me something that I am interested in receiving".  If they used a constant Bluetooth Device Address then there are those who say that your privacy is being intruded upon.  If there was a constant address then you could be "tracked" over time and space.  Instead, if your BLE address changes over periods in a random fashion, then you become anonymous.

franzel61, 2017-11-11:
How can a BLE device uniiquely identified, to make sure my system will react only on this spevific device?

nkolban, 2017-11-12:
Apparently there is a whole science on this topic :-)

Here is the result of a quick search:

https://devzone.nordicsemi.com/question/43670/how-to-distinguish-between-random-and-public-gap-addresses/

I don't know how this applies to the ESP32 exposed APIs nor what (if anything) we need to do at the C++ BLE class level.

franzel61, 2017-11-16:
I did run a similar tool to recognice BLE information on ios and there is under the section device information (2A24) which seem to be constant and manufacture information (2A29). How can I make the visible under the Arduino Development for ESP32 Wroom?

franzel61, 2017-11-21:
Neil, how can I support the study?

nkolban, 2017-11-23:
First thing we'll need to see is if the problem of not finding iOS devices is limited to just one user or if it is endemic across multiple users with iOS devices.  Sadly, I don't have access to an iOS device to test against.  If it is endemic, that would be great because then we can look for a systemic cure.  If it is localized, that too would be great as it would mean that one user is suffering but the majority are not and we can try and find distinction between working and not.

Mr @franzel61 ... do you have an iOS device?  Are you able to see the iOS device on an ESP32 scan?

Anyone else?  A yes/no post on "I have an iOS device and I can see it on an ESP32 BLE scan" ... including the version of the iOS devices and software level would also be useful.

If we find that iOS devices aren't showing up for ESP32 BLE users, then things get interesting.  What I'd like to then do is find some users who have a Linux machine and a BLE dongle they can plugin to their USB port on their PC.  We would then run Wireshark and gather some traces of adverts arriving from iOS BLE broadcasts.  We would then run similar broadcasts from Android (which would be our test baseline that works) and compare the advertisement packets.  In theory, they should be the same (or very similar) but what we would be looking for would be a distinction between the two that ESP32 is handling/failing by case.  Then we would try with low level ESP-IDF APIs to see if the problem is at the ESP-IDF level.  It is, we create an issue against ESP-IDF.  If it is at the C++ BLE class level, then we fix that in this repository.  I'll be delighted to partner with folks via skype/IRC/screen sharing to run tests, gather data and participate in the analysis.

chegewara, 2017-11-23:
Maybe this is our answer, or at least some clue?
https://github.com/espressif/esp-idf/issues/1172#issuecomment-341877961

franzel61, 2017-12-08:
Hello Neil,
I did run 2 bluetooth wireshark session with a moving iphone, can I send them direct to your email?

franzel61, 2017-12-08:
Sorry wrong

mahdikan, 2017-12-27:
hi mr neil you are perfect reasercher in esp i want to find bluetooth mac in esp32 using arduino but i cant,
please help me , if you have .ino file to continusley search macs i will be glad to here from you
thanks a lot
mahdi

nkolban, 2017-12-28:
By "mac" I am assuming you mean the Bluetooth Device Address of advertising BLE servers.  In which case, a sample such as this is likely what you seek.

https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_scan

drschlaumeier, 2018-02-25:
Hi,
I used the BLE_scan example to find nearby Bluetooth devices.

> https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/examples/BLE_scan

Arduino IDE 1.8.5, followed instructions to install `https://github.com/espressif/arduino-esp32.git`.

However, I have same problem as mentioned in that issue. I can see my win10 BT device with random address and I also see my TV BT. Seems to be all BLE.

I cannot see my Nexus 5, Samsung S7E or the Iphone 7. Is it Classic BT?

Can you please help? The issue is still open ... why?  

Thanks in advance

beats0126, 2018-05-24:
@chegewara  thanks for replying, my problem solved. i found the my problem happened with the function "esp_event_loop_init". this is my senior using to make ESP32 can keep checking the wifi connection with esp32. i removed it then the miss scanning problem solved. i think it is because keep updating make ESP32 scanning too heavy already.
but still have to say thanks for all!
good job haha

chegewara, 2018-05-24:
@beats0126 bluetooth and wifi are sharing radio, you can try to set different interval and window times and use wifi. this should solve issue with ble scanning + wifi

chegewara, 2018-05-24:
I just found new bluetooth options in menuconfig(a lot new options), there is also option that can decrease amount of scanned devices and a lot more options, worth to study

beats0126, 2018-05-24:
@chegewara  wow, i will try with set different interval and window times. thanks for suggestion!

bootrino, 2018-09-19:
I believe Apple has gone to considerable effort specifically to prevent iOS devices from being identified wirelessly.  I read an article on it somewhere can't remember where but if it's getting hard, it may be because Apple wants it to be hard.

closed 2018-11-13 by chegewara

descosa, 2019-01-16:
Have you find any solution?

I can't see me iphone XR running iOS 12.1.2

Lawris, 2019-02-18:
Same here for my iPhone if anyone has a solution...

marcoavaccaro, 2019-03-27:
Hello, i'm making a project where the ESP32 scans every 5sec for ble beacons on range and send their mac address and rssi to the clould. For me is better that the scan is made more frequently, but if i down the scan time the ble cannot found every ble beacon on range.

My question is:
I'm programming my esp32 in arduino ide and using the lib "ESP32_BLE_Arduino". Does this lib have some configuring for me to find more devices in less time?

Any tips are welcome also.

I did a test with different scan times, this is the result.
[image]

Thanks

descosa, 2019-03-28:
Marco, are you seeing ios devices ?

marcoavaccaro, 2019-04-01:
@descosa No, i'm trying just to find BLE beacons.

schmidtb89, 2019-05-18:
@marcoavaccaro do you speak spanish? about your problem try setting pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster

PeteRobo, 2023-10-12:
I'm not sure how relevant this is but I'll put it out there incase it helps.
I'm using BLE to scan the BlueTooth spectrum looking for my iPhone and at 1st I couldn't see my iPhone but could see my laptop.
The problem was the my mac address was AC0345064023, but when received by the ESP32, it received AC34564023.
It turned out, if the Mac Address had a 0 in the high nibble it would drop the digit making the Mac Address shorter. If the lower nibble was 0 then it would show a 0. once I figured this out and changed the code to allow shorter Mac Address and changed my lookup table to remove the 0's from the upper nibble it all worked fine.