nkolban/esp32-snippets #175: can't add BLE characteristic with same UUID

ManniAT, 2017-11-08:
I try to add two Humidity characteristics to a service - when adding the second one I get:
E (579) BLEService: << Attempt to add a characteristic but we already have one with this UUID

This should be possible when I add a descriptor to the characteristic.
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.environmental_sensing.xml

Name: ES Measurement
Type: org.bluetooth.descriptor.es_measurement
Requirement: if_multiple_instances_of_same_characteristic_optional_otherwise

Requirement - I have a Meteo Station using NodeMCU esp32 which has values for indoor and outdoor.
So I have the same measurement two times - humidity indoor and humidity outdoor.
From my point of view adding more characteristics with the same UUID should be possible - as long as they provide a different ES Measurement description.

By the way - I just check the BLE_Notify sample.
This uses a lot of memory
Sketch uses 913158 bytes (69%) of program storage space. Maximum is 1310720 bytes.
Global variables use 63888 bytes (21%) of dynamic memory, leaving 231024 bytes for local variables. Maximum is 294912 bytes.
Other examples (WiFi or so) just use a fraction of memory (10-15%) - is this normal?

chegewara, 2017-11-08:
Honestly, ive been convinced it is prohibited to have multiple characteristics with the same uuid in the same service, but looks like it is legit. 

Yes, it is normal that BLE_notify example takes more space than examples which does not use ble, just wifi. Esp stack, not BLE_snippets, allocates about 70kB when you are using bluetooth. Also all libraries are pretty 'expensive' in memory allocation (ram and flash).

nkolban, 2017-11-09:
Oh my ...  During the design, I had wrongly assumed that the characteristics associated with a service would all have unique IDs.  Within the logic, when we are creating a server, the user code creates a server, creates one or more services and for each service creates 1 or more characteristics.  When all this is "defined", the last thing the user code does is call BLEService::start() which causes the "model" built in the ESP32 to be "realized".  It is at this point we run through all the services and create them 1 at a time and for each service, create each characteristic associated with the service.  The characteristics are stored in a data structure called a hash map keyed by their UUID value.  This means that the UUID value has to be unique and hence the test and fail if we attempt to create a second characteristics with the same UUID.

My hope had been to expose characteristics by their UUID alone.  However, there is a 2nd story that I haven't yet surfaced and that is the notion of a characteristic being indentified by its handle.  A handle is a small integer that acts as a "synonym" for the characteristics (or service or descriptor).   These are generated at runtime.  For example, when we create a characteristic, we know its UUID identity ... but we don't yet know the "handle" that will be assigned to it at runtime.  However, once we create a characteristic in the runtime, we are indeed told its handle.  I can just about see a way we can create multiple characteristics with the same UUID ... however ... after we do that .. a couple of things will change.

First ... today, we can ask a service for the BLECharacteristic object with a given UUID ... we can ask:

```
myBLEService.getCharacteristic(uuid)
```

If we change the design to allow multiple characteristics for the same UUID then this function won't always work.  I think I can detect that and throw a "NonUniqueCharacteristic" exception.

We can then add a new method that would be:

```
BLECharacteristic* BLEService::getCharacteristicByHandle(handle)
```
which would return a reference to the characteristic if you know its handle.

Alternatively, since your own code is creating the characteristic in the first place using:

```
BLECharacteristic* BLEService::createCharacteristic()
```

You might not need to retrieve the BLECharacteristic from the service again as you were given it right up front and you could be responsible for remembering it.

All in all ... supporting multiple characteristics is do-able ... it just isn't a quick fix.

chegewara, 2017-11-09:
@ManniAT Hi, last update has been addressed this issue. Could you please check if this meet you requirements and functionality?

ManniAT, 2017-11-09:
Wow - blazing fast.
I hope I can test it this weekend - or at least on monday!

ManniAT, 2017-11-10:
So I gave it a try - the code posted below can't add a characteristic with the new version.
The same code with the old version shows the warning that I add a characteristic with exisiting UUID (of course).
Output using new library:
added callbacks
created service
set desc values
added descriptors
added 2902

The Code:

`#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

bool deviceConnected = false;
uint8_t value = 0;

//https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.es_measurement.xml
//#define ES_MEASUREMENT_DESC 0x290C
byte aDescOutdoor[] = {
  0x00, 0x00, //Reserved for Future Use (RFU) bits in the Flags fields shall be set to 0
  0x00,   //sampling function Unspecified
  0x00, 0x00, 0x00, //mesurement period - not in used
  0x00, 0x00, 0x00, //internal upadate interval (could be used to inform client how often value updates)
  0x13, //Application 0x13 outdoor, 0x14 Indoor, 0x1B inside, 0x1C Outside, 0x1f Internal....
  0xFF  //Meas. unvertainity - Info. not available - else percentage of the CURRENT!!! value       
};
byte aDescIndoor[] = {
  0x00, 0x00, //Reserved for Future Use (RFU) bits in the Flags fields shall be set to 0
  0x00,   //sampling function Unspecified
  0x00, 0x00, 0x00, //mesurement period - not in used
  0x00, 0x00, 0x00, //internal upadate interval (could be used to inform client how often value updates)
  0x14, //Application 0x13 outdoor, 0x14 Indoor, 0x1B inside, 0x1C Outside, 0x1f Internal....
  0xFF  //Meas. unvertainity - Info. not available - else percentage of the CURRENT!!! value       
};

BLEDescriptor indoorTemperatureDescriptor(BLEUUID((uint16_t)0x2901));
BLEDescriptor indoorTemperatureDescriptor2(BLEUUID((uint16_t)0x290C));
BLEDescriptor outdoorTemperatureDescriptor(BLEUUID((uint16_t)0x2901));
BLEDescriptor outdoorTemperatureDescriptor2(BLEUUID((uint16_t)0x290C));

BLECharacteristic indoorTemperatureCharacteristic(BLEUUID((uint16_t)0x2A6E),  // standard 16-bit characteristic UUID
BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
BLECharacteristic outdoorTemperatureCharacteristic(BLEUUID((uint16_t)0x2A6E),  // standard 16-bit characteristic UUID
BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void InitBLE() {
  BLEDevice::init("ESP32-EE060");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  Serial.println("added callbacks");
  // Create the BLE Service
  BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x181A));
  Serial.println("created service");
  outdoorTemperatureDescriptor.setValue("Temperature -40-60°C");
  outdoorTemperatureDescriptor2.setValue((uint8_t*)aDescOutdoor,11);
  Serial.println("set desc values");
  outdoorTemperatureCharacteristic.addDescriptor(&outdoorTemperatureDescriptor);
  outdoorTemperatureCharacteristic.addDescriptor(&outdoorTemperatureDescriptor2);
  Serial.println("added descriptors");
  outdoorTemperatureCharacteristic.addDescriptor(new BLE2902());
  Serial.println("added 2902");
  //LAST OUTPUT FROM NEW VERSION
  pService->addCharacteristic(&outdoorTemperatureCharacteristic);
  Serial.println("added characteristic");
  
  indoorTemperatureDescriptor.setValue("Indoor temperature 0-50 °C");
  indoorTemperatureDescriptor2.setValue((uint8_t*)aDescIndoor,11);
  Serial.println("set IN desc values");
  indoorTemperatureCharacteristic.addDescriptor(&indoorTemperatureDescriptor);
  indoorTemperatureCharacteristic.addDescriptor(&indoorTemperatureDescriptor2);
  Serial.println("added IN descriptors");
  indoorTemperatureCharacteristic.addDescriptor(new BLE2902());
  Serial.println("added IN 2902");
  
  pService->addCharacteristic(&indoorTemperatureCharacteristic);
  Serial.println("added IN characteristic");
  pService->start();
  Serial.println("Service started");
  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("advertising started");
}

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  InitBLE();
  // Start advertising
  
  Serial.println("Waiting a client connection to notify...");
}
float fTemp=0;
float fTempOut=-40;

void loop() {

  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    fTemp+=1.1;
    if(fTemp>50){
      fTemp=0;
    }
    fTempOut+=1.1;
    if(fTempOut>60){
      fTempOut=-40;
    }
    int16_t nTemp=(int16_t)(fTemp*100);
    int16_t nTempOut=(int16_t)(fTempOut*100);

  outdoorTemperatureCharacteristic.setValue((uint8_t*)&nTempOut, 2);
  outdoorTemperatureCharacteristic.notify();

  //indoorTemperatureCharacteristic.setValue((uint8_t*)&nTemp, 2);
  //indoorTemperatureCharacteristic.notify();
  }
  delay(2000);
  value++;
}`

chegewara, 2017-11-10:
I will try with arduino and your code in a moment, just one question. Did you update arduino with latest code from this library or from here >> https://github.com/espressif/arduino-esp32 ?

ManniAT, 2017-11-10:
I downloaded your code using git clone https://github.com/nkolban/esp32-snippets.git
Then I added it as "zip lib" - adding service stopped working.
I removed the lib folder from (windows environment) ...\Documents\Arduino\libraries; restarted Arduino - and after a new build the code worked again (of course arguing about already added UUID).

chegewara, 2017-11-10:
@ManniAT I see some issues with your code. You have to addCharacteristic before you try to add descriptor and set value. In you code you add characteristic at the end. When you fix that it will work, confirmed:

```
[D][BLECharacteristic.cpp:351] handleGATTServerEvent(): Sending a response (esp_ble_gatts_send_response)
[D][BLECharacteristic.cpp:386] handleGATTServerEvent():  - Data: length=0, data=, offset=0
[D][BLEDescriptor.cpp:208] handleGATTServerEvent(): - Testing: Sought handle: 0x2c == descriptor handle: 0xffff ?
[D][BLEDescriptor.cpp:208] handleGATTServerEvent(): - Testing: Sought handle: 0x2c == descriptor handle: 0xffff ?
[D][BLEDescriptor.cpp:208] handleGATTServerEvent(): - Testing: Sought handle: 0x2c == descriptor handle: 0xffff ?
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
*** NOTIFY: 21 ***
[D][BLECharacteristic.cpp:629] setValue(): >> setValue: length=2, data=18f2, characteristic UUID=00002a6e-0000-1000-8000-00805f9b34fb
[D][BLECharacteristic.cpp:636] setValue(): << setValue
[D][BLECharacteristic.cpp:492] notify(): >> notify: length: 2
[D][BLECharacteristic.cpp:511] notify(): << notifications disabled; ignoring
*** NOTIFY: 22 ***
[D][BLECharacteristic.cpp:629] setValue(): >> setValue: length=2, data=86f2, characteristic UUID=00002a6e-0000-1000-8000-00805f9b34fb
[D][BLECharacteristic.cpp:636] setValue(): << setValue
[D][BLECharacteristic.cpp:492] notify(): >> notify: length: 2
[D][BLECharacteristic.cpp:511] notify(): << notifications disabled; ignoring
```

I have to check and fix the issue with descriptor handle 0xffff, but you can add multiple characteristics with the same uuid.

chegewara, 2017-11-10:
@ManniAT Code works with current library, but i did wrong assumption and made mistake in library. Will fix it today and PR. Sorry for that

chegewara, 2017-11-11:
@ManniAT With new update, when @nkolban will merge it, all should works fine. Tested with your code.
To make it happened some changes were made in library. Now we have to keep this order:
- create server,
- create service,
- add characteristic to service, because in next step descriptor require characteristic already created, before all was "automated",
- add descriptor and values to characteristic.
Thanks for code, was much of help and patience.

ManniAT, 2017-11-11:
@chegewara - thank you for your efforts. I also work with a different BT system and there I have to assign the description (0x290C) prior adding the characteristic to the service (at least for the second one) else I get an error - same UUID requires 0x290C :-)
I guess the reason for this is that I can add the characteristic to a running (advertising) service which would (without the 0x290C) enter an invalid state in such a situation.
By the way - in the other system to get a characteristic from a service I have a method which returns a collection of services and an optional filter parameter.

Anyhow - knowing that I have to add the characteristic prior setting the descriptors solves the problem for me - thank you again!!!

chegewara, 2017-11-11:
@ManniAT In that way it will run, but it wont work yet. There is still issue with descriptors which i have to resolve. When you try to read from multiple characteristics all is good, but when you try to read from or write to descriptor then things goes wrong.

chegewara, 2017-11-12:
@ManniAT I have it working on esp-idf, but some suues with arduino. When i fix it i will PR it. I hope it will be today

chegewara, 2017-11-14:
Hi @ManniAT i think im ready with this functionality. Sorry its been so long. Before i will make PR, can you try this library? It will work with code you posted here >> https://github.com/nkolban/esp32-snippets/issues/175#issuecomment-343581092 with just one small change. String in this line seems to be too long `indoorTemperatureDescriptor.setValue("Indoor temperature 0-50 °C");`. When i delete `Indoor` in it then all is ok in nRF. Thanks

ManniAT, 2017-11-14:
I tried it but unfortunately it doesn't work.
Moving the characteristic adding up I get it running.
My Code:

```
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

bool deviceConnected = false;
uint8_t value = 0;

//https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.es_measurement.xml
//#define ES_MEASUREMENT_DESC 0x290C
byte aDescOutdoor[] = {
  0x00, 0x00, //Reserved for Future Use (RFU) bits in the Flags fields shall be set to 0
  0x00,   //sampling function Unspecified
  0x00, 0x00, 0x00, //mesurement period - not in used
  0x00, 0x00, 0x00, //internal upadate interval (could be used to inform client how often value updates)
  0x13, //Application 0x13 outdoor, 0x14 Indoor, 0x1B inside, 0x1C Outside, 0x1f Internal....
  0xFF  //Meas. unvertainity - Info. not available - else percentage of the CURRENT!!! value       
};
byte aDescIndoor[] = {
  0x00, 0x00, //Reserved for Future Use (RFU) bits in the Flags fields shall be set to 0
  0x00,   //sampling function Unspecified
  0x00, 0x00, 0x00, //mesurement period - not in used
  0x00, 0x00, 0x00, //internal upadate interval (could be used to inform client how often value updates)
  0x14, //Application 0x13 outdoor, 0x14 Indoor, 0x1B inside, 0x1C Outside, 0x1f Internal....
  0xFF  //Meas. unvertainity - Info. not available - else percentage of the CURRENT!!! value       
};

BLEDescriptor indoorTemperatureDescriptor(BLEUUID((uint16_t)0x2901));
BLEDescriptor indoorTemperatureDescriptor2(BLEUUID((uint16_t)0x290C));
BLEDescriptor outdoorTemperatureDescriptor(BLEUUID((uint16_t)0x2901));
BLEDescriptor outdoorTemperatureDescriptor2(BLEUUID((uint16_t)0x290C));

BLECharacteristic indoorTemperatureCharacteristic(BLEUUID((uint16_t)0x2A6E),  // standard 16-bit characteristic UUID
BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
BLECharacteristic outdoorTemperatureCharacteristic(BLEUUID((uint16_t)0x2A6E),  // standard 16-bit characteristic UUID
BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void InitBLE() {
  BLEDevice::init("ESP32-EE060");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  Serial.println("added callbacks");
  // Create the BLE Service
  BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x181A));
  Serial.println("created service");
  pService->addCharacteristic(&outdoorTemperatureCharacteristic);
  Serial.println("added characteristic");
//comment next two lines for old lib
  pService->addCharacteristic(&indoorTemperatureCharacteristic);
  Serial.println("added IN characteristic");
  
  outdoorTemperatureDescriptor.setValue("Temperature -40-60°C");
  outdoorTemperatureDescriptor2.setValue((uint8_t*)aDescOutdoor,11);
  Serial.println("set desc values");
  outdoorTemperatureCharacteristic.addDescriptor(&outdoorTemperatureDescriptor);
  outdoorTemperatureCharacteristic.addDescriptor(&outdoorTemperatureDescriptor2);
  Serial.println("added descriptors");
  outdoorTemperatureCharacteristic.addDescriptor(new BLE2902());
  Serial.println("added 2902");

  
  indoorTemperatureDescriptor.setValue("Temp 0-50 °C");
  indoorTemperatureDescriptor2.setValue((uint8_t*)aDescIndoor,11);
  Serial.println("set IN desc values");
  indoorTemperatureCharacteristic.addDescriptor(&indoorTemperatureDescriptor);
  indoorTemperatureCharacteristic.addDescriptor(&indoorTemperatureDescriptor2);
  Serial.println("added IN descriptors");
  indoorTemperatureCharacteristic.addDescriptor(new BLE2902());
  Serial.println("added IN 2902");
  
  pService->start();
  Serial.println("Service started");
  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("advertising started");
}

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  InitBLE();
  // Start advertising
  
  Serial.println("Waiting a client connection to notify...");
}
float fTemp=0;
float fTempOut=-40;

void loop() {

  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    fTemp+=1.1;
    if(fTemp>50){
      fTemp=0;
    }
    fTempOut+=1.1;
    if(fTempOut>60){
      fTempOut=-40;
    }
    int16_t nTemp=(int16_t)(fTemp*100);
    int16_t nTempOut=(int16_t)(fTempOut*100);

  outdoorTemperatureCharacteristic.setValue((uint8_t*)&nTempOut, 2);
  outdoorTemperatureCharacteristic.notify();
//comment next two lines for old lib
  indoorTemperatureCharacteristic.setValue((uint8_t*)&nTemp, 2);
  indoorTemperatureCharacteristic.notify();
  }
  delay(2000);
  value++;
}
```
I try the configuration using https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=de as client.
I can see both Temperature characteristics - but I can't get values and they don't offer subscription.
When I use the old library it (commenting out the marked 4 lines) everything works as expected.
But only with the old version - with the current one I get (after commenting) one characteristic - but it doesn't really work. The difference - I can read the value once (which is not possible with both temperatures).

I attached two screenshots to show the difference.
[image]
So it looks if the things work like they should:
[image]

chegewara, 2017-11-14:
I dont know how its possible but with your code (just had to add missing #include  <BLEDevice. h>) this is my output from nRF:
[image]
[image]

ManniAT, 2017-11-15:
I tried it again using a new clean install of Arduino and a new pull from here - and noticed that you reverted the support for multiple characteristic.
I then tried it with the Arduino clean install and your version from yesterday - same problems as yesterday.

chegewara, 2017-11-15:
@ManniAT Just did new PR, its full working. Now you can dl library from here >> https://dsm.chege.eu.org/fbsharing/VqWmJAqk

Im sorry for any inconvenience, ive been trying to make it work and then i did so mess in my library, but now all should be ok. In this reworked library there is no need to create or add characteristic prior to descriptions.

ManniAT, 2017-11-15:
I installed this version and get
E (420) BLEService: << Attempt to add a characteristic but we already have one with this UUID
Unfortunately I have to visit a customer now - I'll give it a closer look in the evening (it's 9 AM here)
Thank you for your great support fixing this!

chegewara, 2017-11-15:
@ManniAT Dont worry about this message, this `E (420) BLEService: << Attempt to add a characteristic but we already have one with this UUID`. This message left from old code and i left it on purpose, but i should change it from error message to debug, my bad.

ManniAT, 2017-11-15:
Hi @chegewara - just that message - UUID already... :-)
This morning I tried yesterday's and the "reverted" version - yesterday's failed - and the "current" brought exactly that message. Being a bit stressed (had an appointment) I gave your version a try and got exactly the same output as with the "current" (reverted / original) version. I didn't try it with the client.
Now I did and I can confirm that it finally works as expected.
I also moved (more common to me) the characteristic registration down (added description before) and this too works!!!!
I use this lib in a solution I wrote (will further write) about at https://www.hackster.io/ManniAT/esp32-ble-environmental-gatt-server-14cceb
For sure I'll mention your great efforts to help me with this "problem" (some might call it an enhancement)!

Two final comments:
I thought description should be able to hold 80 characters - but found no documentation about a length limit at all.
Includes in library.properties should be fixed / adapted. (BLE.h doesn't exist)

Thank you again.

chegewara, 2017-11-15:
Im glad that it finally works as expected.

chegewara, 2018-01-01:
As we have thi working we can close issue, but i would like to thank @ManniAT for openning it. Thanks to that feature i am able now to work on HID library.