nkolban/esp32-snippets #177: WifiEventHandler crashes when using EventGroupHandle_t

rweller59, 2017-11-08:
I have the following in my wifi class. I crash when xEventGroupSetBits(... is called. I'm programming in C++ using snippets components WiFi.h and WiFiEventHandler.h

EventGroupHandle_t wifi_event_group;
const int CONNECTED_BIT = BIT0;
class wifiEventHandler: public WiFiEventHandler {

	esp_err_t staGotIp(system_event_sta_got_ip_t event_sta_got_ip) {
		// apConnect successful
		ESP_LOGI(TAG, "WIFIHANDLER: staGotIp");
		xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);

nkolban, 2017-11-09:
Howdy,
Here on this repository, we will never turn anyone away ... but please realize that the issues logged here are supposed to be issues against the code that is available here.  For general ESP-IDF / ESP32 programming, we recommend using the ESP-IDF forum found here.

https://esp32.com/index.php

Could the problem that you report here be caused by not calling xEventGroupCreate() to create an event group that is then saved in the "EventGroupHandle_t"?  In your code snippets, you don't show that being done and, if not, that could certainly be an issue?

rweller59, 2017-11-09:
sorry Neil, trying not to post anything unrelated to the repository. Was certain I had done that but you are correct, I had not created the group.

closed 2017-11-09 by rweller59

nkolban, 2017-11-09:
No harm no foul ... its all good ... and as I say, we'll never turn anyone away ... but responses may be delayed or responded to with "sorry ... no idea" ...  and I'd hate for you to wait if you could have gotten an answer so much quicker on the forum.