nkolban/esp32-snippets #178: Bootwifi, cannot locate #include <GeneralUtils.h>

versamodule, 2017-11-09:
@nkolban, First I would like to say thank you so much for all your videos, and your kolban-ESP32.pdf which I bought. I am shocked at the amount of detail it has. 

Today I found the Bootwifi, and in BootWiFi.cpp, it has #include <GeneralUtils.h>. 
I cannot find this anywhere, and I downloaded the latest ESP-IDF From here. 
https://github.com/espressif/esp-idf

Did I forget something else ? 

Thanks for your time. 
James

chegewara, 2017-11-09:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/GeneralUtils.cpp

What IDE you are working on? Arduino or eclipse with esp-idf?

nkolban, 2017-11-09:
Howdy @versamodule 
Great find my friend.  I was recently doing some work in BootWiFi and realized that I could add better debugging and that the routine I wanted already exists in the `cpp_utils` component.  Without thinking, I added the usage of GeneralUtils class but neglected to document that it was a dependency.  I have updated the docs.  The bottom line is that in order to now use BootWiFi, one must also leverage the "cpp_utils" component.

nkolban, 2017-11-28:
Closing for just now.  Re-open as needed.

closed 2017-11-28 by nkolban