nkolban/esp32-snippets #179: wifi.connectAP hangs when invalid password is sent

rweller59, 2017-11-09:
I am testing connectAP with invalid passwords. The relevant part of the code is:
		ESP_LOGI(TAG, "calling connectAP");
		wifi_t.connectAP(ssid, password, true);
		ESP_LOGI(TAG, "returned from connectAP");
The second msg is never printed. I have a wifieventhandler defined and it catches esp_err_t staDisconnected() event.

Also tried wifi_t.connectAP(ssid, password); 
but that didn't change anything since it looks like you did not implement waitForConnection logic

nkolban, 2017-11-09:
Ouch ... that's ugly.   Sincere apologies for that.  A fix for this is now number 1 on my list.

nkolban, 2017-11-09:
Phew ... thankfully that seemed easy.  I didn't test but it compiles cleanly and my "thinking" says it should now work.  Can you pull and test for me?

rweller59, 2017-11-09:
worked for me. thanks

closed 2017-11-09 by rweller59