nkolban/esp32-snippets #182: BLEClient callback for onDisconnect()?

savageautomate, 2017-11-09:
How do you determine if a BLEClient has disconnected?  

I have successfully connected to a BLE client, acquire its services and subscribed to a NOTIFY characteristic.   However, if the physical BLE device times out or is turned off, then I do see a debug message  `E (41502) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
` printed in the console, but I could not find a callback in the API or a connection state property to check to see if the BLEClient instance is still connected.    

```
/**
 * @brief Callbacks associated with a %BLE client.
 */
class BLEClientCallbacks {
public:
	virtual ~BLEClientCallbacks() {};
	virtual void onConnect(BLEClient *pClient) = 0;
};
```

Any help would be greatly appreciated!   Thank You

nkolban, 2017-11-09:
What ESP32 environment are you using?  Is it pure ESP-IDF or is it Arduino ESP32?  Is your ESP32 running as a BLEClient connecting to a remote server and you want to detect that your client has been disconnected or is your ESP32 being a BLE Server that wants to know if a remote client has disconnected?

savageautomate, 2017-11-09:
@nkolban 

_Sorry for the missing details._

> What ESP32 environment are you using? Is it pure ESP-IDF or is it Arduino ESP32? 

Arduino ESP32 on a ESP-WROOM-32

> Is your ESP32 running as a BLEClient connecting to a remote server and you want to detect that your client has been disconnected or is your ESP32 being a BLE Server that wants to know if a remote client has disconnected?

I am writing a client (central) on the ESP32 platform that connects to another BLE device (server/peripheral) to collect sensor and measurement data and print it on the ESP32's serial port.

nkolban, 2017-11-28:
Howdy ... sorry for the delays.  Weve made some changes to the code based as described here #228.  We think these may address the desires for detecting disconnects.  When you get a moment, can you review these changes and see if they are acceptable?

savageautomate, 2017-11-30:
@nkolban 

I tool a look at the change set in the commit and it looks exactly like what I was after.  
How do I suck in this change to test it.  I'm currently linked against the Arduino library here:  https://github.com/nkolban/ESP32_BLE_Arduino 

Thanks, Robert

nkolban, 2017-11-30:
Howdy Robert,
If we read #228 we will see the great words of mr @me-no-dev 

That gentleman is the owner/designer/developer of the Arduino-ESP32 libraries.  Study in depth the conversation relating to #228.

What you will find is that the additions that were made for you/the community resulted in leveraging C++ exception handling.   Specifically, if we try and read or write a remote characteristic or descriptor after a connection that previously existed has now been lost, we throw an exception.  This is all goodness as in a relatively recent ESP-IDF update, Espressif implemented C++ exception handling which wasn't previously present.  However, in order to use this, we need to compile the ESP-IDF with a flag that declares we do indeed wish C++ exception handling enabled.  The Arduino-ESP32 libraries (that are present for Arduino IDE usage) are precompiled by the Arduino-ESP32 distribution.  That means that we need to wait for a re-release from Arduino-ESP32 which includes the C++ exception flag.

In the interim, you can use Arduino-ESP32 in its mode of an "ESP-IDF" component.  If that make sense to you, all is well.  If not, then it is likely that you are exclusively using the Arduino IDE and you have two choices.

1) Study up of ESP-IDF programming and learn how to build a native ESP-IDF application that can use the Arduino-ESP32 library as a component.
2) We wait for a notification that there is a build of Arduino-ESP32 that includes C++ exception handling enabled in ESP-IDF.

me-no-dev, 2017-11-30:
@nkolban please make sure that the Arduino lib repo is updated if any changes are required so I can pull it when I am ready for rebuilding :)

nkolban, 2017-11-30:
Maybe we can synch up.  The BLE Arduino Lib repository does NOT yet require the C++ exceptions because the Arduino-ESP32 lib doesn't enable them yet.  I didn't want to put something in the BLE Arduino Lib that couldn't be compiled with the current available Arduino-ESP32 Arduino IDE.   So they are back-level.  How often are you prepared to repackage the Arduino-ESP32 package?   My thinking would be:

1) You enable the C++ Exception support in Arduino-ESP32
2) You publish a release of Arduino-ESP32 with that enabled
3) I install a virgin Arduino-ESP32 with C++ exception support and test fully the BLE C++ library
4) I update the BLE C++ library into the Arduino BLE repo

Job done ...

Alternative work flows as you choose.

me-no-dev, 2017-11-30:
Sounds good to me :)

savageautomate, 2017-11-30:
@nkolban 

Thanks for the comments, I think I understand the chain a bit better now.  Presently, I am just using the Arduino IDE for a hobby project.  So I can wait for the updates/releases to take place and everything gets synced up.  

Thanks again, Robert

closed 2018-11-13 by chegewara