nkolban/esp32-snippets #185: Create dump() for BLEScanResults

nkolban, 2017-11-10:
When we perform a BLE Scan to find devices in range, we get back an object called BLEScanResults() which is a holder for found BLE devices.  We can iterate over the content.  However, what keeps coming up is the idea that we simply want to dump() the contents to the console.  A new method called dump() has been added to BLEScanResults to dump the found devices to the log.

closed 2017-11-10 by nkolban

zoobab, 2018-07-28:
I am trying to call dump(), any example besides getCount()?

I tried the following:

```
void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.println("Dumping: ");
  Serial.println(foundDevices.dump());
  Serial.println("Scan done!");
}
```
But it does not compile.

chegewara, 2018-07-29:
When you run your code you should get this code running after foundDevices.dump():
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L490-L508